nxhack/openwrt-node-packages #546: some packages do not build

p3x-robot, 2019-07-10:
hello, 
i have some packages that never build, eg:
```text
ERROR: package/feeds/node/node-authenticate-pam failed to build.
ERROR: package/feeds/node/node-bignum failed to build.
ERROR: package/feeds/node/node-bluetooth-hci-socket failed to build.
ERROR: package/feeds/node/node-ejdb failed to build.
ERROR: package/feeds/node/node-fuse-bindings failed to build.
ERROR: package/feeds/node/node-hashtable failed to build.
ERROR: package/feeds/node/node-mdns failed to build.
ERROR: package/feeds/node/node-mknod failed to build.
ERROR: package/feeds/node/node-openzwave-shared failed to build.
ERROR: package/feeds/node/node-pi-spi failed to build.
ERROR: package/feeds/node/node-pty failed to build.
ERROR: package/feeds/node/node-rpio failed to build.
ERROR: package/feeds/node/node-statvfs failed to build.
```

do you know if you could know that would help me to build a bit more packages?
i use a debian testing repo docker container and i use these packages (via apt):
```text
asciidoc
bash
bc
bcc
bin86
binutils
build-essential
bzip2
bzr
cmake
curl
default-jdk
ecj
fastjar
file
flex
g++-multilib
gawk
gcc
genisoimage
gettext
git
git-core
help2man
intltool
java-wrappers
jikespg
libboost-dev
libelf-dev
libgtk2.0-dev
libncurses5-dev
libpcap-dev
libssl-dev
libusb-dev
libxml-parser-perl
make
mc
mercurial
nano
nasm
ncdu
patch
perl-modules
python
python-dev
python3
python3-dev
quilt
rsync
ruby
sdcc
sharutils
software-properties-common
sshpass
subversion
sudo
swig
time
unzip
util-linux
wget
xsltproc
xz-utils
zlib1g-dev
```

if you know any packages that i am missing please let me know, thanks very much!

nxhack, 2019-07-10:
Are you using node v12?
I'm waiting for the upstream to respond.

I'm a little busy lately and I'm sorry for the late reply.

https://github.com/nxhack/openwrt-node-packages/issues/464#issuecomment-486926142

p3x-robot, 2019-07-10:
linksys wrt, nodejs v12, openwrt 18.06.4

nxhack, 2019-11-27:
I will close the issue. Don't hesitate to re-open it if you still have the issue.

closed 2019-11-27 by nxhack