nxhack/openwrt-node-packages #568: openwrt mips serialport bindings binary

heiterkiter, 2019-08-01:
Hi,

I am trying to install zigbee2mqtt on a mips ar7xxx device with CC15.05. 
I got nodejs 8.16 15.05 compiled fine for the device and github cloned zigbee2mqtt

while doing "npm install" in the zigbee2mqtt folder, it comes up with an 404 error.
https://github.com/node-serialport/node-serialport/releases/download/@serialport/bindings@2.0.8/bindings-v2.0.8-node-v57-linux-mips.tar.gz 

because it does not exist.
can you guide me into the right direction to cross compile the file. i got it compiled to have a release folder with a .node file but how can i get this into the npm install process/be recognized by npm

sorry for my noob knowledge. this is the first time i kinda go into OSs. I am more a microprocessor designer/programmer

or do i understand something completely wrong here?

puzzled greets
heiterkiter
hamburg, germany

nxhack, 2019-08-02:
Hi @heiterkiter 

First of all
chaos_calmer is completely obsolete. Security vulnerabilities have not been addressed at all.
If possible, I recommend using the latest version of OpenWrt.

The "@serialport/bindings" module is a binary module, and you need to prepare the binary for the target machine.
Build a binary from source. 
(The target MCU does not have enough resources., so cross build is necessary)

If you don't know about OpenWrt SDK, please read the following document.

https://openwrt.org/docs/guide-developer/using_the_sdk
and https://github.com/nxhack/openwrt-node-packages/blob/master/README.md

/*
There is no information on chaos_calmer here, the SDK for chaos_calmer is here.
https://archive.openwrt.org/chaos_calmer/15.05.1/ar71xx/generic/
*/

Please refer to this when making OpenWrt package of "@serialport/bindings" module.
(chaos_calmer is not supported)

https://github.com/openwrt/packages/tree/master/lang/node-serialport-bindings

I made a (trial) package of zigbee2mqtt for OpenWrt.
(chaos_calmer is not supported)

https://github.com/nxhack/openwrt-node-packages/tree/master/node-zigbee2mqtt
It seems to work in my environment.

FYI
```
daemon.info npm[2505]: 		Zigbee2mqtt requires node version 10, you are running v8.16.0!
```
Requires node v10.x  to run

heiterkiter, 2019-08-02:
Hi Hirokazu,

thanks for your detailed answer.
unluckily i am stuck with cc since the device is kinda special and replacing the cc is too big of a task for me for now.  
with your information i was able to install zigbee2mqtt which runs now wonderful,
Even with v8. (crosscompiled the nessecary modules)

Thanks a lot,
heiterkiter

closed 2019-08-02 by heiterkiter