nxhack/openwrt-node-packages #571: lede-17.01 and Node 10.x: unrecognized command line option '-m64'

deaddy64, 2019-08-05:
Hello.
I use "lede-17.01" branch and getting a error trying to compile Node 10.x. It seems there is a problem compiling Openssl as Node depency. My target platform is 32 bit, so whаt is the option '-m64' for?
How can i fix this error? Thank you.
Here is tail of Make log:
```
 mips-openwrt-linux-musl-gcc -Os -pipe -mno-branch-likely -mips32r2 -mtune=24kc -o /home/openwrt/openwrt-imagebuilder/openwrt-17.01.7/build_dir/target-mips_24kc_musl-1.1.16/node-v10.16.0/out/Release/obj.target/openssl/deps/openssl/openssl/ssl/bio_ssl.o ../deps/openssl/openssl/ssl/bio_ssl.c '-DOPENSSL_NO_HW' '-DNDEBUG' '-DOPENSSL_USE_NODELETE' '-DL_ENDIAN' '-DOPENSSL_PIC' '-DOPENSSL_CPUID_OBJ' '-DOPENSSL_IA32_SSE2' '-DOPENSSL_BN_ASM_MONT' '-DOPENSSL_BN_ASM_MONT5' '-DOPENSSL_BN_ASM_GF2m' '-DSHA1_ASM' '-DSHA256_ASM' '-DSHA512_ASM' '-DKECCAK1600_ASM' '-DRC4_ASM' '-DMD5_ASM' '-DAES_ASM' '-DVPAES_ASM' '-DBSAES_ASM' '-DGHASH_ASM' '-DECP_NISTZ256_ASM' '-DX25519_ASM' '-DPADLOCK_ASM' '-DPOLY1305_ASM' '-DOPENSSLDIR="/etc/ssl"' '-DENGINESDIR="/dev/null"' '-DTERMIOS' -I/home/openwrt/openwrt-imagebuilder/openwrt-17.01.7/staging_dir/target-mips_24kc_musl-1.1.16/usr/include -I../deps/openssl/openssl -I../deps/openssl/openssl/include -I../deps/openssl/openssl/crypto -I../deps/openssl/openssl/crypto/include -I../deps/openssl/openssl/crypto/modes -I../deps/openssl/openssl/crypto/ec/curve448 -I../deps/openssl/openssl/crypto/ec/curve448/arch_32 -I../deps/openssl/config -I../deps/openssl/config/archs/linux-x86_64/asm_avx2 -I../deps/openssl/config/archs/linux-x86_64/asm_avx2/include -I../deps/openssl/config/archs/linux-x86_64/asm_avx2/crypto -I../deps/openssl/config/archs/linux-x86_64/asm_avx2/crypto/include/internal  -pthread -Wall -Wextra -Wno-unused-parameter -Wa,--noexecstack -Wall -O3 -pthread -m64 -Wall -O3 -Wno-missing-field-initializers -Wno-old-style-declaration -O3 -fno-omit-frame-pointer  -MMD -MF /home/openwrt/openwrt-imagebuilder/openwrt-17.01.7/build_dir/target-mips_24kc_musl-1.1.16/node-v10.16.0/out/Release/.deps//home/openwrt/openwrt-imagebuilder/openwrt-17.01.7/build_dir/target-mips_24kc_musl-1.1.16/node-v10.16.0/out/Release/obj.target/openssl/deps/openssl/openssl/ssl/bio_ssl.o.d.raw  -Os -pipe -mno-branch-likely -mips32r2 -mtune=24kc -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -msoft-float -iremap/home/openwrt/openwrt-imagebuilder/openwrt-17.01.7/build_dir/target-mips_24kc_musl-1.1.16/node-v10.16.0:node-v10.16.0 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro  -I/home/openwrt/openwrt-imagebuilder/openwrt-17.01.7/staging_dir/target-mips_24kc_musl-1.1.16/usr/include -I/home/openwrt/openwrt-imagebuilder/openwrt-17.01.7/staging_dir/target-mips_24kc_musl-1.1.16/include -I/home/openwrt/openwrt-imagebuilder/openwrt-17.01.7/staging_dir/toolchain-mips_24kc_gcc-5.4.0_musl-1.1.16/usr/include -I/home/openwrt/openwrt-imagebuilder/openwrt-17.01.7/staging_dir/toolchain-mips_24kc_gcc-5.4.0_musl-1.1.16/include/fortify -I/home/openwrt/openwrt-imagebuilder/openwrt-17.01.7/staging_dir/toolchain-mips_24kc_gcc-5.4.0_musl-1.1.16/include  -c
  mips-openwrt-linux-musl-gcc -Os -pipe -mno-branch-likely -mips32r2 -mtune=24kc -o /home/openwrt/openwrt-imagebuilder/openwrt-17.01.7/build_dir/target-mips_24kc_musl-1.1.16/node-v10.16.0/out/Release/obj.target/openssl/deps/openssl/openssl/ssl/d1_lib.o ../deps/openssl/openssl/ssl/d1_lib.c '-DOPENSSL_NO_HW' '-DNDEBUG' '-DOPENSSL_USE_NODELETE' '-DL_ENDIAN' '-DOPENSSL_PIC' '-DOPENSSL_CPUID_OBJ' '-DOPENSSL_IA32_SSE2' '-DOPENSSL_BN_ASM_MONT' '-DOPENSSL_BN_ASM_MONT5' '-DOPENSSL_BN_ASM_GF2m' '-DSHA1_ASM' '-DSHA256_ASM' '-DSHA512_ASM' '-DKECCAK1600_ASM' '-DRC4_ASM' '-DMD5_ASM' '-DAES_ASM' '-DVPAES_ASM' '-DBSAES_ASM' '-DGHASH_ASM' '-DECP_NISTZ256_ASM' '-DX25519_ASM' '-DPADLOCK_ASM' '-DPOLY1305_ASM' '-DOPENSSLDIR="/etc/ssl"' '-DENGINESDIR="/dev/null"' '-DTERMIOS' -I/home/openwrt/openwrt-imagebuilder/openwrt-17.01.7/staging_dir/target-mips_24kc_musl-1.1.16/usr/include -I../deps/openssl/openssl -I../deps/openssl/openssl/include -I../deps/openssl/openssl/crypto -I../deps/openssl/openssl/crypto/include -I../deps/openssl/openssl/crypto/modes -I../deps/openssl/openssl/crypto/ec/curve448 -I../deps/openssl/openssl/crypto/ec/curve448/arch_32 -I../deps/openssl/config -I../deps/openssl/config/archs/linux-x86_64/asm_avx2 -I../deps/openssl/config/archs/linux-x86_64/asm_avx2/include -I../deps/openssl/config/archs/linux-x86_64/asm_avx2/crypto -I../deps/openssl/config/archs/linux-x86_64/asm_avx2/crypto/include/internal  -pthread -Wall -Wextra -Wno-unused-parameter -Wa,--noexecstack -Wall -O3 -pthread -m64 -Wall -O3 -Wno-missing-field-initializers -Wno-old-style-declaration -O3 -fno-omit-frame-pointer  -MMD -MF /home/openwrt/openwrt-imagebuilder/openwrt-17.01.7/build_dir/target-mips_24kc_musl-1.1.16/node-v10.16.0/out/Release/.deps//home/openwrt/openwrt-imagebuilder/openwrt-17.01.7/build_dir/target-mips_24kc_musl-1.1.16/node-v10.16.0/out/Release/obj.target/openssl/deps/openssl/openssl/ssl/d1_lib.o.d.raw  -Os -pipe -mno-branch-likely -mips32r2 -mtune=24kc -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -msoft-float -iremap/home/openwrt/openwrt-imagebuilder/openwrt-17.01.7/build_dir/target-mips_24kc_musl-1.1.16/node-v10.16.0:node-v10.16.0 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro  -I/home/openwrt/openwrt-imagebuilder/openwrt-17.01.7/staging_dir/target-mips_24kc_musl-1.1.16/usr/include -I/home/openwrt/openwrt-imagebuilder/openwrt-17.01.7/staging_dir/target-mips_24kc_musl-1.1.16/include -I/home/openwrt/openwrt-imagebuilder/openwrt-17.01.7/staging_dir/toolchain-mips_24kc_gcc-5.4.0_musl-1.1.16/usr/include -I/home/openwrt/openwrt-imagebuilder/openwrt-17.01.7/staging_dir/toolchain-mips_24kc_gcc-5.4.0_musl-1.1.16/include/fortify -I/home/openwrt/openwrt-imagebuilder/openwrt-17.01.7/staging_dir/toolchain-mips_24kc_gcc-5.4.0_musl-1.1.16/include  -c
  LD_LIBRARY_PATH=/home/openwrt/openwrt-imagebuilder/openwrt-17.01.7/build_dir/target-mips_24kc_musl-1.1.16/node-v10.16.0/out/Release/lib.host:/home/openwrt/openwrt-imagebuilder/openwrt-17.01.7/build_dir/target-mips_24kc_musl-1.1.16/node-v10.16.0/out/Release/lib.target:$LD_LIBRARY_PATH; export LD_LIBRARY_PATH; cd ../deps/v8/gypfiles; mkdir -p /home/openwrt/openwrt-imagebuilder/openwrt-17.01.7/build_dir/target-mips_24kc_musl-1.1.16/node-v10.16.0/out/Release/obj/gen/torque-generated; "/home/openwrt/openwrt-imagebuilder/openwrt-17.01.7/build_dir/target-mips_24kc_musl-1.1.16/node-v10.16.0/out/Release/torque" ../src/builtins/base.tq ../src/builtins/array.tq ../src/builtins/typed-array.tq -o "/home/openwrt/openwrt-imagebuilder/openwrt-17.01.7/build_dir/target-mips_24kc_musl-1.1.16/node-v10.16.0/out/Release/obj/gen/torque-generated"
  mips-openwrt-linux-musl-gcc -Os -pipe -mno-branch-likely -mips32r2 -mtune=24kc -o /home/openwrt/openwrt-imagebuilder/openwrt-17.01.7/build_dir/target-mips_24kc_musl-1.1.16/node-v10.16.0/out/Release/obj.target/openssl/deps/openssl/openssl/ssl/d1_msg.o ../deps/openssl/openssl/ssl/d1_msg.c '-DOPENSSL_NO_HW' '-DNDEBUG' '-DOPENSSL_USE_NODELETE' '-DL_ENDIAN' '-DOPENSSL_PIC' '-DOPENSSL_CPUID_OBJ' '-DOPENSSL_IA32_SSE2' '-DOPENSSL_BN_ASM_MONT' '-DOPENSSL_BN_ASM_MONT5' '-DOPENSSL_BN_ASM_GF2m' '-DSHA1_ASM' '-DSHA256_ASM' '-DSHA512_ASM' '-DKECCAK1600_ASM' '-DRC4_ASM' '-DMD5_ASM' '-DAES_ASM' '-DVPAES_ASM' '-DBSAES_ASM' '-DGHASH_ASM' '-DECP_NISTZ256_ASM' '-DX25519_ASM' '-DPADLOCK_ASM' '-DPOLY1305_ASM' '-DOPENSSLDIR="/etc/ssl"' '-DENGINESDIR="/dev/null"' '-DTERMIOS' -I/home/openwrt/openwrt-imagebuilder/openwrt-17.01.7/staging_dir/target-mips_24kc_musl-1.1.16/usr/include -I../deps/openssl/openssl -I../deps/openssl/openssl/include -I../deps/openssl/openssl/crypto -I../deps/openssl/openssl/crypto/include -I../deps/openssl/openssl/crypto/modes -I../deps/openssl/openssl/crypto/ec/curve448 -I../deps/openssl/openssl/crypto/ec/curve448/arch_32 -I../deps/openssl/config -I../deps/openssl/config/archs/linux-x86_64/asm_avx2 -I../deps/openssl/config/archs/linux-x86_64/asm_avx2/include -I../deps/openssl/config/archs/linux-x86_64/asm_avx2/crypto -I../deps/openssl/config/archs/linux-x86_64/asm_avx2/crypto/include/internal  -pthread -Wall -Wextra -Wno-unused-parameter -Wa,--noexecstack -Wall -O3 -pthread -m64 -Wall -O3 -Wno-missing-field-initializers -Wno-old-style-declaration -O3 -fno-omit-frame-pointer  -MMD -MF /home/openwrt/openwrt-imagebuilder/openw [... cut ...]
make[2]: Leaving directory '/home/openwrt/openwrt-imagebuilder/openwrt-17.01.7/feeds/node/node'
package/Makefile:105: recipe for target 'package/feeds/node/node/compile' failed
make[1]: *** [package/feeds/node/node/compile] Error 2
make[1]: Leaving directory '/home/openwrt/openwrt-imagebuilder/openwrt-17.01.7'
Build failed - please re-run with -j1 to see the real error message
/home/openwrt/openwrt-imagebuilder/openwrt-17.01.7/include/toplevel.mk:205: recipe for target 'package/node/compile' failed
make: *** [package/node/compile] Error 1
```

nxhack, 2019-08-05:
Hi @deaddy64 

We apologize for the inconvenience. Old branches are not maintained often.
I will backport the latest changes from now on. Please give me some time.

nxhack, 2019-08-06:
@deaddy64 

Addressed openssl build error. (lede-17.01 branch)
Please test.

nxhack, 2019-08-23:
Oh!!!!!

lede-17.01 did not have that setting.
I applied the patch myself, so I made a mistake.

I will change it as soon as possible.

deaddy64, 2019-08-23:
Now it compiles without errors. thank you.

closed 2019-08-23 by deaddy64