nxhack/openwrt-node-packages #572: bus error

p3x-robot, 2019-08-05:
hello,
how are you?
do you have an idea on a Linksys WRT3200ACM / WRT32X i started to get this error:
```text
root@home:/opt/router-scripts-openwrt# npm install
Bus error
root@home:/opt/router-scripts-openwrt# 
```

p3x-robot, 2019-08-05:
https://forum.openwrt.org/t/getting-bus-error/42067/2

nxhack, 2019-08-05:
Hi @p3x-robot 

please tell me.

OpenWrt version:
node.js version:
npm version:
packages config options (node.js):

If it is node.js v12.x, it can not be helped what happens. It's still unstable.

p3x-robot, 2019-08-05:
```text
root@hawk:/opt/router-scripts-openwrt# node -v
v12.7.0
root@hawk:/opt/router-scripts-openwrt# npm -v
Bus error
root@hawk:/opt/router-scripts-openwrt# cat package.json 
{
  "name": "router-scripts-openwrt",
  "version": "1.0.0",
  "description": "",
  "main": "index.js",
  "directories": {
    "lib": "src"
  },
  "scripts": {
    "test": "echo \"Error: no test specified\" && exit 1"
  },
  "repository": {
    "type": "git",
    "url": "https://git.patrikx3.com/router-scripts-openwrt.git"
  },
  "author": "patrikx3",
  "license": "ISC",
  "private": true,
  "dependencies": {
    "corifeus-utils": "^2019.10.140",
    "globby": "^10.0.1",
    "is-port-reachable": "^2.0.1",
    "nodemailer": "^6.3.0"
  },
  "devDependencies": {
    "grunt": "^1.0.4",
    "time-grunt": "^2.0.0"
  }
}
root@hawk:/opt/router-scripts-openwrt# 
```

p3x-robot, 2019-08-06:
do you know if there is a newer npm version that fixes this `bus error` in `npm`? (although same with `yarn`)?

p3x-robot, 2019-08-06:
there is a new npm version though, just so you know...

p3x-robot, 2019-08-07:
ping: ok, tested, with node 12.8.0, this default npm is still giving `bus error`

p3x-robot, 2019-08-08:
if you have anything, i am open to build it and test it on router...

p3x-robot, 2019-08-09:
ping:  i have built the latest commit, now npm is working!

p3x-robot, 2019-08-09:
i guess i close this

closed 2019-08-09 by p3x-robot