nxhack/openwrt-node-packages #585: Failed to build node v12.x on OpenWrt 18.04 for x86_64

1715173329, 2019-08-20:
Please forgive my poor Englist :)

Error info is here: https://paste.ubuntu.com/p/Yjkgrk5mb3

I tried to build it on my machine, but was failed.
It seems that there was something wrong with chacha, but I have no idea to solve it, could you please help me?

And, when I built it, the console printed the warning, is my old c++/g++ version causing it?
```
WARNING: C++ compiler too old, need g++ 6.3.0 or clang++ 8.0.0 (CXX=g++)
WARNING: warnings were emitted in the configure phase
INFO: configure completed successfully
```

Node.JS version: v12.8.1
OpenWrt version: 18.04 SNAPSHOT for x86_64
It works when I choose node.js v8.x and v10.x.

nxhack, 2019-08-21:
Hi @1715173329 

OpenWrt 18.06's openssl package version is 1.0.x, so we are building openssl 1.1.x included in the node.js source. I tested with only mips and arm.
I will check x86_64 as well. 
Please wait for a while.

1715173329, 2019-08-22:
OS: Ubuntu 14.04.6 x86_64 LTS
System gcc version: 4.8.4
OpenWrt toolchain gcc version: 7.4.0

nxhack, 2019-08-22:
Oh, it ’s too old.
I highly recommend upgrade.

nxhack, 2019-08-22:
> Error info is here: https://paste.ubuntu.com/p/Yjkgrk5mb3

At this log, it is a "hostpkg" build error.
What is your build environment?

OS
OS version
gcc version

1715173329, 2019-08-22:
So, is the old ubuntu version causing it? or system gcc version?

1715173329, 2019-08-22:
All right, I will update Ubuntu version and retry.
Thank you very much~

1715173329, 2019-08-22:
It works on Ubuntu 18.04.3 LTS, thank you again!

closed 2019-08-22 by 1715173329