nxhack/openwrt-node-packages #586: package configuration menu is empty

deaddy64, 2019-08-21:
Hello. 
I'm trying to check your fix #571 , but after updating feed i have an empty list in "Node.js" dir of menuconfig. So can't choose node version. Nothing changes after feeds update and install.
Has anything changed in feed index? Or i do something wrong?

nxhack, 2019-08-22:
Hi @deaddy64 

Check if MIPS_FPU_EMULATOR is enabled.
https://github.com/nxhack/openwrt-node-packages/tree/lede-17.01#illegal-instruction-issue

deaddy64, 2019-08-23:
In my make menuconfig i have no "MIPS_FPU_EMULATOR"
[image]
I have this option only in kernel_menuconfig. I've selected it.
[image]
But there is still empty list in Node dir of menuconfig. Then i edited DEPENDS of Makefile: add "||y" to the first expression. Finally it's ok now!
But the question remained. It seems that KERNEL_MIPS_FPU_EMULATOR option is not defined even if it was selected in kernel_menuconfig.
[image]
Thnx.

nxhack, 2019-08-23:
Oh!!!

lede-17.01 did not have that setting.
I applied the patch myself, so I made a mistake.

I will change it as soon as possible.

nxhack, 2019-08-23:
@deaddy64 

Fixed.
Thank you for the report.

closed 2019-08-23 by deaddy64