nxhack/openwrt-node-packages #609: cannot build nodejs on openwrt 18.06.5

p3x-robot, 2019-11-09:
giving error :
```text
docker@d66828a60bac:/build/source$ make menu_config
WARNING: Makefile 'package/feeds/node/node/Makefile' has a dependency on 'icu-full-data', which does not exist
 make[1] menu_config
make -r menu_config: build failed. Please re-run make with -j1 V=s or V=sc for a higher verbosity level to see what's going on
make: *** [/build/source/include/toplevel.mk:218: menu_config] Error 1
```

p3x-robot, 2019-11-09:
do i have to select since for sometime small icu?

p3x-robot, 2019-11-09:
i am trying on nodejs v12

p3x-robot, 2019-11-10:
in olrder version in v12 was working, but since v12.13.0 LTS i checked the log and there is an error, so it does not build to node.

p3x-robot, 2019-11-10:
it is not the icu problem, that is just a warning...

nxhack, 2019-11-10:
Oh! 
I did not test with 18.06. Fixed.

18.06 Support will be discontinued soon.

closed 2019-11-19 by p3x-robot