nxhack/openwrt-node-packages #617: I can't compile node-spi-device from MT7688

arno608rw, 2019-11-21:
I can't compile node-spi-device from MT7688

nxhack, 2019-11-21:
please show me

OpenWrt version
node.js version

arno608rw, 2019-11-21:
openwrt 18.06.4
node v10.17

---- err -----
npm info it worked if it ends with ok
npm info using npm@6.11.3
npm info using node@v10.17.0
npm timing stage:loadCurrentTree Completed in 103ms
npm timing stage:loadIdealTree:cloneCurrentTree Completed in 0ms
npm timing stage:loadIdealTree:loadShrinkwrap Completed in 3ms
npm http fetch GET 200 https://registry.npmjs.org/bindings 233ms
npm http fetch GET 200 https://registry.npmjs.org/nan 241ms
npm http fetch GET 200 https://registry.npmjs.org/file-uri-to-path 52ms
npm http fetch GET 200 https://registry.npmjs.org/file-uri-to-path/-/file-uri-to-path-1.0.0.tgz 61ms
npm timing stage:loadIdealTree:loadAllDepsIntoIdealTree Completed in 416ms
npm timing stage:loadIdealTree Completed in 431ms
npm timing stage:generateActionsToTake Completed in 10ms
npm http fetch GET 200 https://registry.npmjs.org/bindings/-/bindings-1.5.0.tgz 84ms
npm http fetch GET 200 https://registry.npmjs.org/nan/-/nan-2.14.0.tgz 102ms
npm timing action:extract Completed in 163ms
npm timing action:finalize Completed in 18ms
npm timing action:refresh-package-json Completed in 32ms
npm info lifecycle file-uri-to-path@1.0.0~preinstall: file-uri-to-path@1.0.0
npm info lifecycle bindings@1.5.0~preinstall: bindings@1.5.0
npm info lifecycle nan@2.14.0~preinstall: nan@2.14.0
npm info lifecycle spi-device@3.0.0~preinstall: spi-device@3.0.0
npm timing action:preinstall Completed in 7ms
npm info linkStuff file-uri-to-path@1.0.0
npm info linkStuff bindings@1.5.0
npm info linkStuff nan@2.14.0
npm info linkStuff spi-device@3.0.0
npm timing action:build Completed in 12ms
npm info lifecycle file-uri-to-path@1.0.0~install: file-uri-to-path@1.0.0
npm info lifecycle bindings@1.5.0~install: bindings@1.5.0
npm info lifecycle nan@2.14.0~install: nan@2.14.0
npm info lifecycle spi-device@3.0.0~install: spi-device@3.0.0

> spi-device@3.0.0 install /home/arno/openwrt-18/build_dir/target-mipsel_24kc_musl/node-spi-device-3.0.0/ipkg-install/usr/lib/node_modules/spi-device
> node-gyp rebuild

gyp info it worked if it ends with ok
gyp info using node-gyp@5.0.3
gyp info using node@10.17.0 | linux | x64
gyp info find Python using Python version 2.7.17 found at "/home/arno/openwrt-18/staging_dir/hostpkg/bin/python"
gyp ERR! configure error 
gyp ERR! stack Error: EACCES: permission denied, mkdir '/home/arno/openwrt-18/build_dir/target-mipsel_24kc_musl/node-spi-device-3.0.0/ipkg-install/usr/lib/node_modules/spi-device/build'
gyp ERR! System Linux 4.9.184-linuxkit
gyp ERR! command "/home/arno/openwrt-18/staging_dir/hostpkg/bin/node" "/home/arno/openwrt-18/staging_dir/hostpkg/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "rebuild"
gyp ERR! cwd /home/arno/openwrt-18/build_dir/target-mipsel_24kc_musl/node-spi-device-3.0.0/ipkg-install/usr/lib/node_modules/spi-device
gyp ERR! node -v v10.17.0
gyp ERR! node-gyp -v v5.0.3
gyp ERR! not ok 
npm info lifecycle spi-device@3.0.0~install: Failed to exec install script
npm timing action:install Completed in 481ms
npm timing stage:rollbackFailedOptional Completed in 54ms
npm timing stage:runTopLevelLifecycles Completed in 1355ms
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! spi-device@3.0.0 install: `node-gyp rebuild`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the spi-device@3.0.0 install script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.
npm timing npm Completed in 2279ms

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/arno/openwrt-18/tmp/npm-cache-ZVZijRhdjl/_logs/2019-11-20T09_48_02_763Z-debug.log
Makefile:69: recipe for target '/home/arno/openwrt-18/build_dir/target-mipsel_24kc_musl/node-spi-device-3.0.0/.built' failed
make[3]: *** [/home/arno/openwrt-18/build_dir/target-mipsel_24kc_musl/node-spi-device-3.0.0/.built] Error 1
make[3]: Leaving directory '/home/arno/openwrt-18/feeds/node/node-spi-device'
time: package/feeds/node/node-spi-device/compile#1.49#0.51#3.46
package/Makefile:111: recipe for target 'package/feeds/node/node-spi-device/compile' failed
make[2]: *** [package/feeds/node/node-spi-device/compile] Error 2
make[2]: Leaving directory '/home/arno/openwrt-18'
package/Makefile:107: recipe for target '/home/arno/openwrt-18/staging_dir/target-mipsel_24kc_musl/stamp/.package_compile' failed
make[1]: *** [/home/arno/openwrt-18/staging_dir/target-mipsel_24kc_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/arno/openwrt-18'
/home/arno/openwrt-18/include/toplevel.mk:216: recipe for target 'world' failed
make: *** [world] Error 2

arno608rw, 2019-11-22:
gyp ERR! stack Error: EACCES: permission denied, mkdir xxx/xxx//build_dir/target-mipsel_24kc_musl/node-spi-device-3.0.0/ipkg-install/usr/lib/node_modules/spi-device/build

nxhack, 2019-11-22:
hmmm.. It works fine in my environment.
It's an x86_64 environment, but mipsel is probably fine too.
```
npm info it worked if it ends with ok
npm info using npm@6.11.3
npm info using node@v10.17.0
npm timing stage:loadCurrentTree Completed in 71ms
npm timing stage:loadIdealTree:cloneCurrentTree Completed in 0ms
npm timing stage:loadIdealTree:loadShrinkwrap Completed in 2ms
npm http fetch GET 200 https://registry.npmjs.org/bindings 73ms
npm http fetch GET 200 https://registry.npmjs.org/nan 74ms
npm http fetch GET 200 https://registry.npmjs.org/file-uri-to-path 18ms
npm http fetch GET 200 https://registry.npmjs.org/file-uri-to-path/-/file-uri-to-path-1.0.0.tgz 20ms
npm timing stage:loadIdealTree:loadAllDepsIntoIdealTree Completed in 141ms
npm timing stage:loadIdealTree Completed in 146ms
npm timing stage:generateActionsToTake Completed in 4ms
npm http fetch GET 200 https://registry.npmjs.org/bindings/-/bindings-1.5.0.tgz 63ms
npm http fetch GET 200 https://registry.npmjs.org/nan/-/nan-2.14.0.tgz 70ms
npm timing action:extract Completed in 107ms
npm timing action:finalize Completed in 10ms
npm timing action:refresh-package-json Completed in 22ms
npm info lifecycle file-uri-to-path@1.0.0~preinstall: file-uri-to-path@1.0.0
npm info lifecycle bindings@1.5.0~preinstall: bindings@1.5.0
npm info lifecycle nan@2.14.0~preinstall: nan@2.14.0
npm info lifecycle spi-device@3.0.0~preinstall: spi-device@3.0.0
npm timing action:preinstall Completed in 3ms
npm info linkStuff file-uri-to-path@1.0.0
npm info linkStuff bindings@1.5.0
npm info linkStuff nan@2.14.0
npm info linkStuff spi-device@3.0.0
npm timing action:build Completed in 3ms
npm info lifecycle file-uri-to-path@1.0.0~install: file-uri-to-path@1.0.0
npm info lifecycle bindings@1.5.0~install: bindings@1.5.0
npm info lifecycle nan@2.14.0~install: nan@2.14.0
npm info lifecycle spi-device@3.0.0~install: spi-device@3.0.0

> spi-device@3.0.0 install /mnt/openwrt-18.06/openwrt/build_dir/target-x86_64_musl/node-spi-device-3.0.0/ipkg-install/usr/lib/node_modules/spi-device
> node-gyp rebuild

gyp info it worked if it ends with ok
gyp info using node-gyp@5.0.3
gyp info using node@10.17.0 | linux | x64
gyp info find Python using Python version 2.7.17 found at "/mnt/openwrt-18.06/openwrt/staging_dir/hostpkg/bin/python"
gyp info spawn /mnt/openwrt-18.06/openwrt/staging_dir/hostpkg/bin/python
gyp info spawn args [ '/mnt/openwrt-18.06/openwrt/staging_dir/hostpkg/lib/node_modules/npm/node_modules/node-gyp/gyp/gyp_main.py',
gyp info spawn args   'binding.gyp',
gyp info spawn args   '-f',
gyp info spawn args   'make',

<SNIP>

gyp info ok 
npm timing action:install Completed in 5842ms
npm info lifecycle file-uri-to-path@1.0.0~postinstall: file-uri-to-path@1.0.0
npm info lifecycle bindings@1.5.0~postinstall: bindings@1.5.0
npm info lifecycle nan@2.14.0~postinstall: nan@2.14.0
npm info lifecycle spi-device@3.0.0~postinstall: spi-device@3.0.0
npm timing action:postinstall Completed in 2ms
npm timing stage:executeActions Completed in 6002ms
npm timing stage:rollbackFailedOptional Completed in 0ms
npm timing stage:runTopLevelLifecycles Completed in 6227ms
+ spi-device@3.0.0
added 4 packages from 9 contributors in 6.231s
npm timing npm Completed in 6546ms
npm info ok 
```

Check the directory permissions carefully.

nxhack, 2019-11-22:
I tested with mipsel, it was fine.

arno608rw, 2019-11-25:
> I tested with mipsel, it was fine.

I try chmod -R 777 /home/arno/openwrt-18/
It gyp ERR! stack Error: EACCES: permission denied, mkdir ......

[image]

nxhack, 2019-11-25:
Are you using the root user or sudo?
Please run with general user.

arno608rw, 2019-12-06:
OK it' work in other user. Thank @nxhack

closed 2019-12-06 by arno608rw