nxhack/openwrt-node-packages #637: Failed to build nodejs 12.14.1 with icu on OpenWrt 18.06 for x86_64

1715173329, 2020-01-21:
OS: Ubuntu
GCC Ver: 8.3.0
OpenWrt Ver: 18.06
Target: x86_64
Nodejs Ver: 12.14.1 with small-icu

Error info:
```bash
touch /home/ctcgfw/openwrt/build_dir/target-x86_64_musl/node-v12.14.1/out/Release/obj.target/tools/v8_gypfiles/run_torque.stamp
rm -f /home/ctcgfw/openwrt/build_dir/target-x86_64_musl/node-v12.14.1/out/Release/obj.target/tools/icu/libicui18n.a ar-file-list; mkdir -p `dirname /home/ctcgfw/openwrt/build_dir/target-x86_64_musl/node-v12.14.1/out/Release/obj.target/tools/icu/libicui18n.a`
x86_64-openwrt-linux-musl-gcc-ar crsT /home/ctcgfw/openwrt/build_dir/target-x86_64_musl/node-v12.14.1/out/Release/obj.target/tools/icu/libicui18n.a @/home/ctcgfw/openwrt/build_dir/target-x86_64_musl/node-v12.14.1/out/Release/obj.target/tools/icu/libicui18n.a.ar-file-list
rm -f /home/ctcgfw/openwrt/build_dir/target-x86_64_musl/node-v12.14.1/out/Release/obj.target/tools/icu/libicuucx.a ar-file-list; mkdir -p `dirname /home/ctcgfw/openwrt/build_dir/target-x86_64_musl/node-v12.14.1/out/Release/obj.target/tools/icu/libicuucx.a`
x86_64-openwrt-linux-musl-gcc-ar crsT /home/ctcgfw/openwrt/build_dir/target-x86_64_musl/node-v12.14.1/out/Release/obj.target/tools/icu/libicuucx.a @/home/ctcgfw/openwrt/build_dir/target-x86_64_musl/node-v12.14.1/out/Release/obj.target/tools/icu/libicuucx.a.ar-file-list
  g++  -o /home/ctcgfw/openwrt/build_dir/target-x86_64_musl/node-v12.14.1/out/Release/obj.host/icutools/deps/icu-small/source/tools/toolutil/filetools.o ../deps/icu-small/source/tools/toolutil/filetools.cpp '-DV8_DEPRECATION_WARNINGS' '-DV8_IMMINENT_DEPRECATION_WARNINGS' '-DU_COMMON_IMPLEMENTATION=1' '-DU_I18N_IMPLEMENTATION=1' '-DU_IO_IMPLEMENTATION=1' '-DU_TOOLUTIL_IMPLEMENTATION=1' '-DU_ATTRIBUTE_DEPRECATED=' '-D_CRT_SECURE_NO_DEPRECATE=' '-DU_STATIC_IMPLEMENTATION=1' '-DUCONFIG_NO_SERVICE=1' '-DU_ENABLE_DYLOAD=0' '-DU_HAVE_STD_STRING=1' '-DUCONFIG_NO_BREAK_ITERATION=0' -I/home/ctcgfw/openwrt/staging_dir/target-x86_64_musl/usr/include -I../deps/icu-small/source/common -I../deps/icu-small/source/i18n -I../deps/icu-small/source/tools/toolutil  -pthread -Wall -Wextra -Wno-unused-parameter -m64 -Wno-deprecated-declarations -Wno-strict-aliasing -O3 -fno-omit-frame-pointer -fno-exceptions -std=gnu++1y -frtti -MMD -MF /home/ctcgfw/openwrt/build_dir/target-x86_64_musl/node-v12.14.1/out/Release/.deps//home/ctcgfw/openwrt/build_dir/target-x86_64_musl/node-v12.14.1/out/Release/obj.host/icutools/deps/icu-small/source/tools/toolutil/filetools.o.d.raw   -c
In file included from ../deps/icu-small/source/tools/toolutil/filetools.cpp:22:
../deps/icu-small/source/common/putilimp.h:226:20: error: missing binary operator before token "("
 #elif __has_builtin(__c11_atomic_load) && \
                    ^
tools/icu/icutools.host.mk:556: recipe for target '/home/ctcgfw/openwrt/build_dir/target-x86_64_musl/node-v12.14.1/out/Release/obj.host/icutools/deps/icu-small/source/tools/toolutil/filetools.o' failed
make[5]: *** [/home/ctcgfw/openwrt/build_dir/target-x86_64_musl/node-v12.14.1/out/Release/obj.host/icutools/deps/icu-small/source/tools/toolutil/filetools.o] Error 1
rm cc87e600b79047e202ad4f1f031be19305aa61a6.intermediate 3b5e7f5440316a08a9855aefe97325981757b04d.intermediate
Makefile:101: recipe for target 'node' failed
make[4]: *** [node] Error 2
make[4]: Leaving directory '/home/ctcgfw/openwrt/build_dir/target-x86_64_musl/node-v12.14.1'
Makefile:174: recipe for target '/home/ctcgfw/openwrt/build_dir/target-x86_64_musl/node-v12.14.1/.built' failed
make[3]: *** [/home/ctcgfw/openwrt/build_dir/target-x86_64_musl/node-v12.14.1/.built] Error 2
```

nxhack, 2020-01-22:
Hi @1715173329 

I understand this issue.

This is a conflict between icu package and small-icu of node.js.

There are two solutions.

(1)
If your firmware requires icu packages, specify the i18n option in node.js to system-icu.

(2)
If the firmware does not require the icu package, disable the icu package and remove it from the build environment.

1715173329, 2020-01-22:
All right, thank you so much.
Build passed.

closed 2020-01-22 by 1715173329