nxhack/openwrt-node-packages #639: Package node-hashtable is not compiled

BKPepe, 2020-01-27:
It seems that node-hashtable can be used just with Node v8 as it does not support v10. 

Faillog, which I get:
```
make[3]: Entering directory '/mypath/build/build_dir/target-arm_cortex-a9+vfpv3_musl_eabi/node-hashtable-2.0.2/ipkg-install/usr/lib/node_modules/hashtable/build'
  CXX(target) Release/obj.target/native/src/hashtable.o
In file included from ../src/hashtable.h:15:0,
                 from ../src/hashtable.cpp:1:
../src/v8_value_hasher.h: In member function 'bool v8_value_equal_to::operator()(CopyablePersistent*, CopyablePersistent*) const':
../src/v8_value_hasher.h:43:24: error: no matching function for call to 'v8::Value::Equals(v8::Local<v8::Value>&)'
         if (a->Equals(b)) {          /* same as JS == */
                        ^
In file included from /mypath/build/staging_dir/target-arm_cortex-a9+vfpv3_musl_eabi/usr/include/node/node.h:63:0,
                 from ../src/hashtable.h:13,
                 from ../src/hashtable.cpp:1:
/mypath/build/staging_dir/target-arm_cortex-a9+vfpv3_musl_eabi/usr/include/node/v8.h:2616:37: note: candidate: v8::Maybe<bool> v8::Value::Equals(v8::Local<v8::Context>, v8::Local<v8::Value>) const
   V8_WARN_UNUSED_RESULT Maybe<bool> Equals(Local<Context> context,
                                     ^~~~~~
/mypath/build/staging_dir/target-arm_cortex-a9+vfpv3_musl_eabi/usr/include/node/v8.h:2616:37: note:   candidate expects 2 arguments, 1 provided
../src/hashtable.cpp: In static member function 'static void HashTable::init(v8::Local<v8::Object>)':
../src/hashtable.cpp:26:82: error: no matching function for call to 'v8::FunctionTemplate::GetFunction()'
     target->Set(Nan::New("HashTable").ToLocalChecked(), constructor->GetFunction());
                                                                                  ^
In file included from /mypath/build/staging_dir/target-arm_cortex-a9+vfpv3_musl_eabi/usr/include/node/node.h:63:0,
                 from ../src/hashtable.h:13,
                 from ../src/hashtable.cpp:1:
/mypath/build/staging_dir/target-arm_cortex-a9+vfpv3_musl_eabi/usr/include/node/v8.h:5995:46: note: candidate: v8::MaybeLocal<v8::Function> v8::FunctionTemplate::GetFunction(v8::Local<v8::Context>)
   V8_WARN_UNUSED_RESULT MaybeLocal<Function> GetFunction(
                                              ^~~~~~~~~~~
/mypath/build/staging_dir/target-arm_cortex-a9+vfpv3_musl_eabi/usr/include/node/v8.h:5995:46: note:   candidate expects 1 argument, 0 provided
../src/hashtable.cpp: In static member function 'static Nan::NAN_METHOD_RETURN_TYPE HashTable::Constructor(Nan::NAN_METHOD_ARGS_TYPE)':
../src/hashtable.cpp:50:43: error: no matching function for call to 'v8::Value::Int32Value()'
         int buckets = info[0]->Int32Value();
                                           ^
In file included from /mypath/build/staging_dir/target-arm_cortex-a9+vfpv3_musl_eabi/usr/include/node/node.h:63:0,
                 from ../src/hashtable.h:13,
                 from ../src/hashtable.cpp:1:
/mypath/build/staging_dir/target-arm_cortex-a9+vfpv3_musl_eabi/usr/include/node/v8.h:2613:40: note: candidate: v8::Maybe<int> v8::Value::Int32Value(v8::Local<v8::Context>) const
   V8_WARN_UNUSED_RESULT Maybe<int32_t> Int32Value(Local<Context> context) const;
                                        ^~~~~~~~~~
/mypath/build/staging_dir/target-arm_cortex-a9+vfpv3_musl_eabi/usr/include/node/v8.h:2613:40: note:   candidate expects 1 argument, 0 provided
../src/hashtable.cpp: In static member function 'static Nan::NAN_METHOD_RETURN_TYPE HashTable::Rehash(Nan::NAN_METHOD_ARGS_TYPE)':
../src/hashtable.cpp:231:42: error: no matching function for call to 'v8::Value::Int32Value()'
     size_t buckets = info[0]->Int32Value();
                                          ^
In file included from /mypath/build/staging_dir/target-arm_cortex-a9+vfpv3_musl_eabi/usr/include/node/node.h:63:0,
                 from ../src/hashtable.h:13,
                 from ../src/hashtable.cpp:1:
/mypath/build/staging_dir/target-arm_cortex-a9+vfpv3_musl_eabi/usr/include/node/v8.h:2613:40: note: candidate: v8::Maybe<int> v8::Value::Int32Value(v8::Local<v8::Context>) const
   V8_WARN_UNUSED_RESULT Maybe<int32_t> Int32Value(Local<Context> context) const;
                                        ^~~~~~~~~~
/mypath/build/staging_dir/target-arm_cortex-a9+vfpv3_musl_eabi/usr/include/node/v8.h:2613:40: note:   candidate expects 1 argument, 0 provided
../src/hashtable.cpp: In static member function 'static Nan::NAN_METHOD_RETURN_TYPE HashTable::Reserve(Nan::NAN_METHOD_ARGS_TYPE)':
../src/hashtable.cpp:249:43: error: no matching function for call to 'v8::Value::Int32Value()'
     size_t elements = info[0]->Int32Value();
                                           ^
In file included from /mypath/build/staging_dir/target-arm_cortex-a9+vfpv3_musl_eabi/usr/include/node/node.h:63:0,
                 from ../src/hashtable.h:13,
                 from ../src/hashtable.cpp:1:
/mypath/build/staging_dir/target-arm_cortex-a9+vfpv3_musl_eabi/usr/include/node/v8.h:2613:40: note: candidate: v8::Maybe<int> v8::Value::Int32Value(v8::Local<v8::Context>) const
   V8_WARN_UNUSED_RESULT Maybe<int32_t> Int32Value(Local<Context> context) const;
                                        ^~~~~~~~~~
/mypath/build/staging_dir/target-arm_cortex-a9+vfpv3_musl_eabi/usr/include/node/v8.h:2613:40: note:   candidate expects 1 argument, 0 provided
../src/hashtable.cpp: In static member function 'static Nan::NAN_METHOD_RETURN_TYPE HashTable::MaxLoadFactor(Nan::NAN_METHOD_ARGS_TYPE)':
../src/hashtable.cpp:266:39: error: no matching function for call to 'v8::Value::NumberValue()'
         factor = info[0]->NumberValue();
                                       ^
In file included from /mypath/build/staging_dir/target-arm_cortex-a9+vfpv3_musl_eabi/usr/include/node/node.h:63:0,
                 from ../src/hashtable.h:13,
                 from ../src/hashtable.cpp:1:
[... 20 lines cut ...]
/mypath/build/staging_dir/target-arm_cortex-a9+vfpv3_musl_eabi/usr/include/node/v8.h:2590:31: note: candidate: v8::Local<v8::Object> v8::Value::ToObject(v8::Isolate*) const
                 Local<Object> ToObject(Isolate* isolate) const);
                               ^
/mypath/build/staging_dir/target-arm_cortex-a9+vfpv3_musl_eabi/usr/include/node/v8config.h:311:3: note: in definition of macro 'V8_DEPRECATED'
   declarator __attribute__((deprecated(message)))
   ^~~~~~~~~~
/mypath/build/staging_dir/target-arm_cortex-a9+vfpv3_musl_eabi/usr/include/node/v8.h:2590:31: note:   candidate expects 1 argument, 0 provided
                 Local<Object> ToObject(Isolate* isolate) const);
                               ^
/mypath/build/staging_dir/target-arm_cortex-a9+vfpv3_musl_eabi/usr/include/node/v8config.h:311:3: note: in definition of macro 'V8_DEPRECATED'
   declarator __attribute__((deprecated(message)))
   ^~~~~~~~~~
../src/hashtable.cpp:303:33: error: no matching function for call to 'v8::Function::Call(v8::Local<v8::Object>&, const unsigned int&, v8::Local<v8::Value> [3])'
         cb->Call(ctx, argc, argv);
                                 ^
In file included from /mypath/build/staging_dir/target-arm_cortex-a9+vfpv3_musl_eabi/usr/include/node/node.h:63:0,
                 from ../src/hashtable.h:13,
                 from ../src/hashtable.cpp:1:
/mypath/build/staging_dir/target-arm_cortex-a9+vfpv3_musl_eabi/usr/include/node/v8.h:4193:43: note: candidate: v8::MaybeLocal<v8::Value> v8::Function::Call(v8::Local<v8::Context>, v8::Local<v8::Value>, int, v8::Local<v8::Value>*)
   V8_WARN_UNUSED_RESULT MaybeLocal<Value> Call(Local<Context> context,
                                           ^~~~
/mypath/build/staging_dir/target-arm_cortex-a9+vfpv3_musl_eabi/usr/include/node/v8.h:4193:43: note:   candidate expects 4 arguments, 3 provided
native.target.mk:111: recipe for target 'Release/obj.target/native/src/hashtable.o' failed
make[3]: *** [Release/obj.target/native/src/hashtable.o] Error 1
```

nxhack, 2020-01-28:
Sadly, upstream doesn't support v12 yet.

https://github.com/chad3814/node-hashtable/issues/47

nxhack, 2020-01-29:
A new package for hash operations was added.

https://github.com/jhuckaby/megahash

https://github.com/nxhack/openwrt-node-packages/tree/master/node-megahash

closed 2020-04-25 by BKPepe