nxhack/openwrt-node-packages #640: Node can not be compiled with enabled ASLR PIE on aarch64_cortex-a53

BKPepe, 2020-01-27:
When I enabled `PKG_ASLR_PIE` (which is not default) for the build, I can not compile Node on **aarch64_cortex-a53**, but it works for **arm_cortex-a9+vfpv3_musl_eabi**.

Output:

```
/path/build/build_dir/target-aarch64_cortex-a53_musl/node-v12.14.1/out/Release/obj.target/tools/v8_gypfiles/libv8_snapshot.a /path/build/build_dir/target-aarch64_cortex-a53_musl/node-v12.14.1/out/Release/obj.target/tools/v8_gypfiles/libv8_initializers.a -L/path/build/staging_dir/target-aarch64_cortex-a53_musl/usr/lib -lz -lhttp_parser -luv -lcares -lnghttp2 -lcrypto -lssl -ldl -Wl,--end-group
/path/build/build_dir/target-aarch64_cortex-a53_musl/node-v12.14.1/out/Release/obj.target/libnode/src/node_http2.o: In function `node::http2::Http2Stream::Provider::~Provider()':
node_http2.cc:(.text+0x2c): relocation truncated to fit: R_AARCH64_LD64_GOTPAGE_LO15 against symbol `node::http2::Http2Stream::Provider::~Provider()' defined in .text section in /path/build/build_dir/target-aarch64_cortex-a53_musl/node-v12.14.1/out/Release/obj.target/libnode/src/node_http2.o
node_http2.cc:(.text+0x2c): warning: Too many GOT entries for -fpic, please recompile with -fPIC
/path/build/staging_dir/toolchain-aarch64_cortex-a53_gcc-7.5.0_musl/lib/gcc/aarch64-openwrt-linux-musl/7.5.0/../../../../aarch64-openwrt-linux-musl/bin/ld: final link failed: Symbol needs debug section which does not exist
collect2: error: ld returned 1 exit status
cctest.target.mk:177: recipe for target '/path/build/build_dir/target-aarch64_cortex-a53_musl/node-v12.14.1/out/Release/cctest' failed
make[4]: *** [/path/build/build_dir/target-aarch64_cortex-a53_musl/node-v12.14.1/out/Release/cctest] Error 1
rm e55efe88db421443fcdd762ce570bca45649b2f7.intermediate 829b89c65180b63e4d6b29258e462c7ce517188d.intermediate 793a3c08b46f72a113b7822249c4fe8d72e7db1a.intermediate b6848839cda799a2113ff5a775f556faef47d0d3.intermediate 93f44726da9194e907a7c0d4e8e09f674f9157c0.intermediate
Makefile:101: recipe for target 'node' failed
```

For now, I've just add `PKG_ASLR_PIE:=0` to node Makefile, so I would be able to compile it for both targets.

nxhack, 2020-01-28:
Thank you @BKPepe 

set PKG_ASLR_PIE:=0

BKPepe, 2020-01-31:
Fixed by this commit https://github.com/nxhack/openwrt-node-packages/commit/9232b8024d22972b6e0abf66dd7c4712dfc3e3a7, but I think there could be a different issue underneath, but as workaround it's fine for me.

closed 2020-01-31 by BKPepe