nxhack/openwrt-node-packages #648: Node does not build for OpenWRT 19.07.1 (brcm2708/bcm2709)

dlpratte, 2020-02-07:
When trying to build for raspberry pi I get the following error trying to build node (host systen is Alpine 3.11 in Docker).

[This is the output of the build](https://github.com/nxhack/openwrt-node-packages/files/4168894/log_build.txt). It's essentially a series of failures to find some internal function in an imported library.

Can you help?

nxhack, 2020-02-07:
Hi @dlpratte 

I have no experience in Alpine linux, so I will answer with a guess.
To build a 32bit target of node.js, you need a multilib environment provided by gcc on host. Probably not provided by alpine musl. So it seems that the only solution is to create a 32-bit alpine linux environment and build with it.

dlpratte, 2020-02-07:
Understandable. Unfortunate that it is the only package not to compile on Alpine, but I updated my container to use Ubuntu instead and no longer have any problems.

Cheers.

closed 2020-02-07 by dlpratte