nxhack/openwrt-node-packages #653: cannot compile inksys wrt anymore

p3x-robot, 2020-03-01:
```text
  arm-openwrt-linux-muslgnueabi-g++ -Os -pipe -o /build/source/build_dir/target-arm_cortex-a9+vfpv3_musl_eabi/node-v12.16.1/out/Release/obj.target/libnode/src/node_util.o ../src/node_util.cc '-DV8_DEPRECATION_WARNINGS' '-DV8_IMMINENT_DEPRECATION_WARNINGS' '-D__STDC_FORMAT_MACROS' '-DNODE_ARCH="arm"' '-DNODE_PLATFORM="linux"' '-DNODE_WANT_INTERNALS=1' '-DV8_DEPRECATION_WARNINGS=1' '-DNODE_OPENSSL_SYSTEM_CERT_PATH=""' '-DHAVE_INSPECTOR=0' '-DNODE_REPORT' '-D__POSIX__' '-DNODE_USE_V8_PLATFORM=1' '-DHAVE_OPENSSL=1' -I/build/source/staging_dir/target-arm_cortex-a9+vfpv3_musl_eabi/usr/include -I../src -I/build/source/build_dir/target-arm_cortex-a9+vfpv3_musl_eabi/node-v12.16.1/out/Release/obj/gen -I../deps/histogram/src -I../deps/uvwasi/include -I../deps/v8/include -I../deps/llhttp/include -I../deps/brotli/c/include  -Wall -Wextra -Wno-unused-parameter -pthread -Wall -Wextra -Wno-unused-parameter -O3 -fno-omit-frame-pointer -fno-rtti -fno-exceptions -std=gnu++1y -MMD -MF /build/source/build_dir/target-arm_cortex-a9+vfpv3_musl_eabi/node-v12.16.1/out/Release/.deps//build/source/build_dir/target-arm_cortex-a9+vfpv3_musl_eabi/node-v12.16.1/out/Release/obj.target/libnode/src/node_util.o.d.raw  -Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -mfloat-abi=hard -iremap/build/source/build_dir/target-arm_cortex-a9+vfpv3_musl_eabi/node-v12.16.1:node-v12.16.1 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro  -I/build/source/staging_dir/target-arm_cortex-a9+vfpv3_musl_eabi/usr/include -I/build/source/staging_dir/target-arm_cortex-a9+vfpv3_musl_eabi/include -I/build/source/staging_dir/toolchain-arm_cortex-a9+vfpv3_gcc-7.5.0_musl_eabi/usr/include -I/build/source/staging_dir/toolchain-arm_cortex-a9+vfpv3_gcc-7.5.0_musl_eabi/include/fortify -I/build/source/staging_dir/toolchain-arm_cortex-a9+vfpv3_gcc-7.5.0_musl_eabi/include  -c
../src/node_util.cc: In function 'void node::util::Sleep(const v8::FunctionCallbackInfo<v8::Value>&)':
../src/node_util.cc:176:3: error: 'uv_sleep' was not declared in this scope
   uv_sleep(msec);
   ^~~~~~~~
../src/node_util.cc:176:3: note: suggested alternative: 'uv_stop'
   uv_sleep(msec);
   ^~~~~~~~
   uv_stop
libnode.target.mk:225: recipe for target '/build/source/build_dir/target-arm_cortex-a9+vfpv3_musl_eabi/node-v12.16.1/out/Release/obj.target/libnode/src/node_util.o' failed
make[5]: *** [/build/source/build_dir/target-arm_cortex-a9+vfpv3_musl_eabi/node-v12.16.1/out/Release/obj.target/libnode/src/node_util.o] Error 1
rm f039e06fde8f4acb054f3e8c9331351d422dc44c.intermediate 44c0ab1bdb078a2e1a2b8e06803b768026cdb872.intermediate 3685a6a9656a169a7b82794b374c38e6998c2717.intermediate 1bfd97abbc1c0f5f444d34feea08e9bc94acbc6e.intermediate a4d8587680e606c492ff369facde9cb9ee3d969f.intermediate
Makefile:101: recipe for target 'node' failed
make[4]: *** [node] Error 2
make[4]: Leaving directory '/build/source/build_dir/target-arm_cortex-a9+vfpv3_musl_eabi/node-v12.16.1'
Makefile:175: recipe for target '/build/source/build_dir/target-arm_cortex-a9+vfpv3_musl_eabi/node-v12.16.1/.built' failed
make[3]: *** [/build/source/build_dir/target-arm_cortex-a9+vfpv3_musl_eabi/node-v12.16.1/.built] Error 2
time: package/feeds/node/node/compile#3283.02#172.99#3448.53
docker@4c7e0df39978:/build/source/logs/package/feeds/node/node$
```

nxhack, 2020-03-02:
node.js 12.16 or higher requires the latest libuv.

https://github.com/openwrt/packages/commit/f1e15068cb59b86456f43900b51820e75e46ca4a

p3x-robot, 2020-03-02:
ok thanks

closed 2020-03-02 by p3x-robot

BKPepe, 2020-03-03:
The issue remains same in OpenWrt 19.07 using @nxhack feed.

p3x-robot, 2020-03-03:
i patched it myself and it works.