nxhack/openwrt-node-packages #655: Homebridge, crypto error

p3x-robot, 2020-03-07:
Hey NXHACK!

How are you? Do you know what it means?

```txt
crw------- 1 root root 10, 58 Mar 6 19:08 /dev/crypto

Fri Mar  6 20:09:40 2020 daemon.err homebridge[4766]: Could not open /dev/crypto: Permission denied
Fri Mar  6 20:09:41 2020 daemon.err homebridge[4766]: internal/modules/cjs/loader.js:796
Fri Mar  6 20:09:41 2020 daemon.err homebridge[4766]:     throw err;
Fri Mar  6 20:09:41 2020 daemon.err homebridge[4766]:     ^
Fri Mar  6 20:09:41 2020 daemon.err homebridge[4766]:
Fri Mar  6 20:09:41 2020 daemon.err homebridge[4766]: Error: Cannot find module 'hap-nodejs/lib/util/once'
Fri Mar  6 20:09:41 2020 daemon.err homebridge[4766]: Require stack:
Fri Mar  6 20:09:41 2020 daemon.err homebridge[4766]: - /usr/lib/node/homebridge/lib/server.js
Fri Mar  6 20:09:41 2020 daemon.err homebridge[4766]: - /usr/lib/node/homebridge/lib/cli.js
Fri Mar  6 20:09:41 2020 daemon.err homebridge[4766]: - /usr/lib/node/homebridge/bin/homebridge
Fri Mar  6 20:09:41 2020 daemon.err homebridge[4766]:     at Function.Module._resolveFilename (internal/modules/cjs/loader.js:793:17)
Fri Mar  6 20:09:41 2020 daemon.err homebridge[4766]:     at Function.Module._load (internal/modules/cjs/loader.js:686:27)
Fri Mar  6 20:09:41 2020 daemon.err homebridge[4766]:     at Module.require (internal/modules/cjs/loader.js:848:19)
Fri Mar  6 20:09:41 2020 daemon.err homebridge[4766]:     at require (internal/modules/cjs/helpers.js:74:18)
Fri Mar  6 20:09:41 2020 daemon.err homebridge[4766]:     at Object.<anonymous> (/usr/lib/node/homebridge/lib/server.js:10:12)
Fri Mar  6 20:09:41 2020 daemon.err homebridge[4766]:     at Module._compile (internal/modules/cjs/loader.js:955:30)
Fri Mar  6 20:09:41 2020 daemon.err homebridge[4766]:     at Object.Module._extensions..js (internal/modules/cjs/loader.js:991:10)
Fri Mar  6 20:09:41 2020 daemon.err homebridge[4766]:     at Module.load (internal/modules/cjs/loader.js:811:32)
Fri Mar  6 20:09:41 2020 daemon.err homebridge[4766]:     at Function.Module._load (internal/modules/cjs/loader.js:723:14)
Fri Mar  6 20:09:41 2020 daemon.err homebridge[4766]:     at Module.require (internal/modules/cjs/loader.js:848:19) {
Fri Mar  6 20:09:41 2020 daemon.err homebridge[4766]:   code: 'MODULE_NOT_FOUND',
Fri Mar  6 20:09:41 2020 daemon.err homebridge[4766]:   requireStack: [
Fri Mar  6 20:09:41 2020 daemon.err homebridge[4766]:     '/usr/lib/node/homebridge/lib/server.js',
Fri Mar  6 20:09:41 2020 daemon.err homebridge[4766]:     '/usr/lib/node/homebridge/lib/cli.js',
Fri Mar  6 20:09:41 2020 daemon.err homebridge[4766]:     '/usr/lib/node/homebridge/bin/homebridge'
Fri Mar  6 20:09:41 2020 daemon.err homebridge[4766]:   ]
Fri Mar  6 20:09:41 2020 daemon.err homebridge[4766]: }
```

p3x-robot, 2020-03-07:
https://github.com/patrikx3/openwrt-insomnia/issues/2

p3x-robot, 2020-03-07:
Hey NXHACK!

Do you think that `/dev/crypto` only allows `root` and Homebridge is using another user???

p3x-robot, 2020-03-07:
i think this is the blame:
https://github.com/nxhack/openwrt-node-packages/blame/master/node-homebridge/files/homebridge.init#L14

p3x-robot, 2020-03-07:
besides it looks like it is missing a package: `hap-nodejs`
for the blame is here:
https://github.com/nxhack/openwrt-node-packages/blame/master/node-homebridge/patches/001-turn_off_depends.patch#L7

nxhack, 2020-03-09:
There are different styles here. There is also a way to change cryptodev permissions. I follow the policy of not running large network services as root.

TLCary, 2020-03-11:
Doesn't OpenWRT typically runs everything as root?  To install homebridge on OpenWRT, I needed to first install avahi-dbus-daemon and libavahi-compat-libdnssd (it took me a while to figure that out), but avahi installs dbus automatically as dependency.  But dbus errors on install unless you find a way to first create a "netdev" user group, (there is no addgroup command).  Users and groups are a complication and installing homebridge on OpenWRT is not easy.  The homebridge installer creates a homebridge user, but apparently crypto doesn't like it.  I got hombridge from Insomnia working on a generic OpenWRT 19.07.2 install.  Thank you!
A hombridge service never appears in etc/avahi/services, and I can't see homebridge in my mDNS Discovery application. It's like no other mDNS service. But somehow it magically works.  I'm a little confused.  Is it possible to get the homebridge installer to automatically install avahi and the other dependency and create the required users?

nxhack, 2020-03-11:
Hi @TLCary 

> Doesn't OpenWRT typically runs everything as root?

 YES. but I am afraid of that. ;)

> To install homebridge on OpenWRT, I needed to first install avahi-dbus-daemon and libavahi-compat-libdnssd (it took me a while to figure that out),

The latest hap-nodejs no longer requires libdnssd to build.
Of course you need the avahi daemon or something to enable mDNS.

> but avahi installs dbus automatically as dependency. But dbus errors on install unless you find a way to first create a "netdev" user group, (there is no addgroup command). 

I remember adding netdev group.
However, I remember that dbus log was only a warning and did not affect operation.

> A hombridge service never appears in etc/avahi/services, and I can't see homebridge in my mDNS Discovery application. It's like no other mDNS service. 

```avahi-browse -t _hap._tcp```
Doesn't that show up if you run this command?

> But somehow it magically works. I'm a little confused. Is it possible to get the homebridge installer to automatically install avahi and the other dependency and create the required users?

'umdns' :  an openwrt-style light mdns service. (instead of avahi)
Homebridge using umdn maybe work.

Unfortunately, my HomeBridge has migrated to raspbian and has not made any new efforts with openwrt.

nxhack, 2020-03-11:
Since hap-nodejs build no longer depends on avahi, multiple mdns services can be selected. So I don't define package dependencies.

By the way, I tested using umdns, mDNS record was registered correctly.

```
root@OpenWrt:~# ubus call umdns update
root@OpenWrt:~# ubus call umdns browse
{
	"_hap._tcp": {
		"Homebridge-CAD8": {
			"port": 32837,
			"txt": "md=Homebridge",
			"txt": "pv=1.0",
			"txt": "id=CC:22:3D:E3:CE:30",
			"txt": "c#=2",
			"txt": "s#=1",
			"txt": "ff=0",
			"txt": "ci=2",
			"txt": "sf=1",
			"txt": "sh=AAzJSQ=="
		}
	}
}
```

nxhack, 2021-05-05:
Close the issue. If the problem still exists, feel free to reopen it.

closed 2021-05-05 by nxhack