nxhack/openwrt-node-packages #665: recursive dependency?

rgillan, 2020-04-20:
Hi, 
doing an OpenWRT rebuild on latest 19.07.02 from source.  I've included this repo in the feed as per normal and when I try devconfig or make, it throws up a few recursive dependency errors:

Error: recursive dependency detected!
For a resolution refer to Documentation/kbuild/kconfig-language.txt
subsection "Kconfig recursive dependency limitations"
tmp/.config-package.in:23254:	symbol PACKAGE_node-cylon-gpio is selected by PACKAGE_node-cylon-gpio
tmp/.config-package.in:23293:error: recursive dependency detected!
For a resolution refer to Documentation/kbuild/kconfig-language.txt
subsection "Kconfig recursive dependency limitations"
tmp/.config-package.in:23293:	symbol PACKAGE_node-cylon-i2c is selected by PACKAGE_node-cylon-i2c
tmp/.config-package.in:132630:error: recursive dependency detected!
For a resolution refer to Documentation/kbuild/kconfig-language.txt
subsection "Kconfig recursive dependency limitations"

In the node.js language menu in menuconfig, it's showing duplicate entries for those packages (and they are not included in our config, but still throwing errors:

`< > node-cylon-gpio........................................... CylonJS - GPIO                                                            │ │  
  │ │                                                                   < > node-cylon-gpio..... Cylon module for General Purpose Input/Output (GPIO)                                                            │ │  
  │ │                                                                   < > node-cylon-i2c............................................. CylonJS - I2C                                                            │ │  
  │ │                                                                   < > node-cylon-i2c.......................... Cylon.js drivers for i2c devices `

It's also throwing one on KeepaliveD which I'm just trying to find as well (can't find it in menuconfig as yet).

Thanks again for any help
Rob

rgillan, 2020-04-22:
I know it's a crazy time, but has anyone got an idea how to fix this (its a bit of a showstopper for us atm).

nxhack, 2020-04-22:
It's a conflict between the official package and the custom package.

Try to remove unnecessary packages.

For example.
```
rm ./package/feeds/packages/node
rm ./package/feeds/packages/node-*
./scripts/feeds install -a -p node
make defconfig
make
```

rgillan, 2020-04-23:
Thanks so much.  All good now

closed 2020-04-23 by rgillan