nxhack/openwrt-node-packages #667: Compile faild

DhanushSandy, 2020-04-23:
make[3]: Leaving directory '/home/dhanush/chaos_calmer/package/libs/openssl'
make[3]: Entering directory '/home/dhanush/chaos_calmer/feeds/node/node'
. /home/dhanush/chaos_calmer/include/shell.sh; xzcat /home/dhanush/chaos_calmer/dl/node-v8.17.0.tar.xz | tar -C /home/dhanush/chaos_calmer/build_dir/target-mipsel_24kec+dsp_uClibc-0.9.33.2/node-v8.17.0/.. -xf - 

Applying ./patches/v8.x/001-hardfloat.patch using plaintext: 
patching file deps/v8/src/base/cpu.cc

Applying ./patches/v8.x/002-addr_info.patch using plaintext: 
patching file deps/uv/src/unix/getaddrinfo.c

Applying ./patches/v8.x/003-path.patch using plaintext: 
patching file lib/module.js

Applying ./patches/v8.x/004-uclibc-no-execinfo.patch using plaintext: 
patching file deps/v8/src/base/debug/stack_trace_posix.cc
patching file deps/v8/src/compiler/node-matchers.h
patching file src/backtrace_posix.cc

Applying ./patches/v8.x/010-execvp-arg-list-too-long.patch using plaintext: 
patching file deps/npm/node_modules/node-gyp/gyp/pylib/gyp/generator/make.py
Hunk #1 succeeded at 151 (offset 2 lines).
Hunk #2 succeeded at 219 (offset 2 lines).
Hunk #3 succeeded at 1660 (offset 1 line).
patching file tools/gyp/pylib/gyp/generator/make.py

Applying ./patches/v8.x/012-changing-default-npm-settings.patch using plaintext: 
patching file deps/npm/lib/config/defaults.js
Hunk #1 succeeded at 175 (offset 4 lines).
Hunk #2 succeeded at 231 (offset 5 lines).

Applying ./patches/v8.x/013-deps_uv_uclibc.patch using plaintext: 
patching file deps/uv/src/unix/thread.c

Applying ./patches/v8.x/013-npm-unsafe-override-for-global-install.patch using plaintext: 
patching file deps/npm/lib/config/defaults.js
Hunk #1 FAILED at 238.
1 out of 1 hunk FAILED -- saving rejects to file deps/npm/lib/config/defaults.js.rej
Patch failed!  Please fix ./patches/v8.x/013-npm-unsafe-override-for-global-install.patch!
Makefile:159: recipe for target '/home/dhanush/chaos_calmer/build_dir/target-mipsel_24kec+dsp_uClibc-0.9.33.2/node-v8.17.0/.prepared_cc69890dfe3e7f7935032f15a75222e4' failed
make[3]: *** [/home/dhanush/chaos_calmer/build_dir/target-mipsel_24kec+dsp_uClibc-0.9.33.2/node-v8.17.0/.prepared_cc69890dfe3e7f7935032f15a75222e4] Error 1
make[3]: Leaving directory '/home/dhanush/chaos_calmer/feeds/node/node'
package/Makefile:191: recipe for target 'package/feeds/node/node/compile' failed
make[2]: *** [package/feeds/node/node/compile] Error 2
make[2]: Leaving directory '/home/dhanush/chaos_calmer'
package/Makefile:188: recipe for target '/home/dhanush/chaos_calmer/staging_dir/target-mipsel_24kec+dsp_uClibc-0.9.33.2/stamp/.package_compile' failed
make[1]: *** [/home/dhanush/chaos_calmer/staging_dir/target-mipsel_24kec+dsp_uClibc-0.9.33.2/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/dhanush/chaos_calmer'
/home/dhanush/chaos_calmer/include/toplevel.mk:181: recipe for target 'world' failed
make: *** [world] Error 2
dhanush@ubuntu:~/chaos_calmer$

nxhack, 2020-04-23:
That's the case with the 18.06 branch, isn't it?

I'll check it out.

#stayhome

nxhack, 2020-04-23:
Neither 18.06 nor 17.01 branch seems to be a problem.

Tell me about your environment.

DhanushSandy, 2020-04-24:
i am using 16.04

nxhack, 2020-04-24:
Please tell me which branch of the openwrt-node-packages repository you are using. And please tell me the version of openwrt.

DhanushSandy, 2020-04-24:
I am using one you posted in the wiki. I am following the same.
https://github.com/nxhack/openwrt-node-packages/wiki

https://github.com/openwrt/chaos_calmer.git

nxhack, 2020-04-24:
All right.
I'll check it out. Please wait.

nxhack, 2020-04-25:
Fixed.

Note that openwrt chaos calmer is completely obsolete.

#stayhome

nxhack, 2020-04-25:
Apparently npm is broken, so I'm dropping the version of node.js. Please wait.

nxhack, 2020-04-25:
I tried to make it buildable for now.

To maintain this, however, would be to ignore the vulnerability.

Branches for chaos calmer are no longer supported.

#stayhome

DhanushSandy, 2020-04-25:
Hhey, I found another way. it worked.

DhanushSandy, 2020-04-26:
Thanks for the support

closed 2020-04-26 by DhanushSandy