nxhack/openwrt-node-packages #676: Cannot build for ramips/mipsel _24kc (Seeed Studio LinkIt 7688)

sans-ltd, 2020-06-02:
Hello,
is this related to the dropped MIPS support:

```
  g++ -m32 -o /home/gonzo/tools/openwrt-sdk-19.07.2-ramips-mt76x8_gcc-7.5.0_musl.Linux-x86_64/build_dir/target-mipsel_24kc_musl/node-v12.16.3/out/Release/obj.host/v8_libbase/deps/v8/src/base/bits.o ../deps/v8/src/base/bits.cc '-DV8_GYP_BUILD' '-DV8_TYPED_ARRAY_MAX_SIZE_IN_HEAP=64' '-D__STDC_FORMAT_MACROS' '-DV8_TARGET_ARCH_MIPS' '-D__mips_soft_float=1' '-DV8_EMBEDDER_STRING="-node.35"' '-DENABLE_DISASSEMBLER' '-DV8_PROMISE_INTERNAL_FIELD_COUNT=1' '-DENABLE_MINOR_MC' '-DV8_USE_SNAPSHOT' '-DV8_CONCURRENT_MARKING' '-DV8_ENABLE_LAZY_SOURCE_POSITIONS' '-DV8_USE_SIPHASH' '-DDISABLE_UNTRUSTED_CODE_MITIGATIONS' '-DV8_WIN64_UNWINDING_INFO' '-DV8_ENABLE_REGEXP_INTERPRETER_THREADED_DISPATCH' '-DBUILDING_V8_BASE_SHARED' '-D_MIPS_TARGET_HW' '-D_MIPS_ARCH_MIPS32R2' '-DFPU_MODE_FP32' -I/home/gonzo/tools/openwrt-sdk-19.07.2-ramips-mt76x8_gcc-7.5.0_musl.Linux-x86_64/staging_dir/target-mipsel_24kc_musl/usr/include -I../deps/v8 -I../deps/v8/include  -pthread -Wno-unused-parameter -Wno-return-type -fno-strict-aliasing -m32 -O3 -fno-omit-frame-pointer -fdata-sections -ffunction-sections -O3 -fno-rtti -fno-exceptions -std=gnu++1y -MMD -MF /home/gonzo/tools/openwrt-sdk-19.07.2-ramips-mt76x8_gcc-7.5.0_musl.Linux-x86_64/build_dir/target-mipsel_24kc_musl/node-v12.16.3/out/Release/.deps//home/gonzo/tools/openwrt-sdk-19.07.2-ramips-mt76x8_gcc-7.5.0_musl.Linux-x86_64/build_dir/target-mipsel_24kc_musl/node-v12.16.3/out/Release/obj.host/v8_libbase/deps/v8/src/base/bits.o.d.raw   -c
In file included from /usr/lib/gcc/x86_64-linux-gnu/7/include/stdint.h:9:0,
                 from ../deps/v8/src/base/bits.h:8,
                 from ../deps/v8/src/base/bits.cc:5:
/usr/include/stdint.h:26:10: fatal error: bits/libc-header-start.h: No such file or directory
 #include <bits/libc-header-start.h>
          ^~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
```
Seems that the compiler pulls the stdint.h from the host, not from the cross environment.
My feeds.conf.default:
```
src-git base https://git.openwrt.org/openwrt/openwrt.git;v19.07.3
src-git packages https://git.openwrt.org/feed/packages.git^2e6bd4cb86682b224803325127d3f777d40b3231
src-git luci https://git.openwrt.org/project/luci.git^fb2f36306756d0d0782dcab8413a8bb7ec379e54
src-git routing https://git.openwrt.org/feed/routing.git^3f8571194c2765ed31aa73459e86c2ebf943d27d
src-git telephony https://git.openwrt.org/feed/telephony.git^036cd451c35b82b3d8cac519864986894d9f6958
src-git node https://github.com/nxhack/openwrt-node-packages.git
```
My build host system is Ubuntu 18.04.
Is there a fix for it?
Regards and thanks for the great project in advance
Florian

sans-ltd, 2020-06-04:
OK fixed the first issue (apt-get install gcc-multilib, g++-multilib)
Not it compiled:
```
 make[1] package/node/compile
 make[2] -C package/toolchain clean-build
 make[2] -C package/toolchain compile
 make[2] -C feeds/base/package/libs/nghttp2 clean-build
 make[2] -C feeds/base/package/libs/nghttp2 compile
 make[2] -C feeds/base/package/libs/openssl clean-build
 make[2] -C feeds/base/package/libs/openssl compile
 make[2] -C feeds/base/package/libs/zlib clean-build
 make[2] -C feeds/base/package/libs/zlib compile
 make[2] -C feeds/packages/libs/libhttp-parser clean-build
 make[2] -C feeds/packages/libs/libhttp-parser compile
 make[2] -C feeds/packages/libs/libuv clean-build
 make[2] -C feeds/packages/libs/libuv compile
 make[2] -C feeds/base/package/utils/bzip2 host-compile
 make[2] -C feeds/packages/libs/expat host-compile
 make[2] -C feeds/packages/lang/python/python host-compile
 make[2] -C feeds/packages/lang/node clean-build
 make[2] -C feeds/packages/lang/node compile

```
 But no .ipk packages for node were generated:
```
~/tools/openwrt-sdk-19.07.2-ramips-mt76x8_gcc-7.5.0_musl.Linux-x86_64$ find . -name *.ipk
./bin/targets/ramips/mt76x8/packages/libstdcpp6_7.5.0-2_mipsel_24kc.ipk
./bin/targets/ramips/mt76x8/packages/libatomic1_7.5.0-2_mipsel_24kc.ipk
./bin/targets/ramips/mt76x8/packages/libpthread_1.1.24-2_mipsel_24kc.ipk
./bin/targets/ramips/mt76x8/packages/libc_1.1.24-2_mipsel_24kc.ipk
./bin/targets/ramips/mt76x8/packages/librt_1.1.24-2_mipsel_24kc.ipk
./bin/targets/ramips/mt76x8/packages/libgcc1_7.5.0-2_mipsel_24kc.ipk
```
Another strange thing: Node.js is deactivated in make menuconfig (Node.js submenu is empty)

nxhack, 2020-06-04:
Try
```
rm ./package/feeds/packages/node
rm ./package/feeds/packages/node-*
./scripts/feeds install -a -p node
```

sans-ltd, 2020-06-06:
This did not help. The Node.js menu in menuconfig is still empty. Still no packages. Also set up the SDK from scratch. Same results. Only the few packages listed above were built. Is your build tested also with the SDK or only with buildroot?

sans-ltd, 2020-06-08:
Next observation when building with V=sc: 
A lot of files get instlalled below openwrt-sdk-19.07.2-ramips-mt76x8_gcc-7.5.0_musl.Linux-x86_64/build_dir/target-mipsel_24kc_musl/node-v12.16.3/ipkg-install but they are not packaged afterwards. Why?

sans-ltd, 2020-06-08:
OK it was my fault: With the default image (which is the base configuration for the SDK) the MIPS_FPU_EMULATION is not enabled. So you definitely need a custom image with enabled FPU emulation for the MIPS hardware.
Conclusion: Not possible to build with SDK for MIPS, only with buildroot.

closed 2020-06-08 by sans-ltd

muazmalas, 2020-12-28:
> This did not help. The Node.js menu in menuconfig is still empty. Still no packages. Also set up the SDK from scratch. Same results. Only the few packages listed above were built. Is your build tested also with the SDK or only with buildroot?

How did you manage to get the Node.js menu to populate can you elaborate, thanks

nxhack, 2020-12-29:
@muazmalas 

This is a wiki I wrote for the obsolete version, but it may help you build your firmware.

https://github.com/nxhack/openwrt-node-packages/wiki