nxhack/openwrt-node-packages #679: Node fails to compile due to error: 'nghttp2_option_set_max_settings' was not declared in this scope

BKPepe, 2020-06-07:
Trying to compile node 12.18.0 on environment: OpenWrt 19.07 (Turris Omnia - mvebu - cortex-a9 | Turris MOX - mvebu - cortexa53)

Log:

```
-I/path/build/staging_dir/toolchain-arm_cortex-a9+vfpv3-d16_gcc-7.5.0_musl_eabi/include/fortify -I/path/build/staging_dir/toolchain-arm_cortex-a9+vfpv3-d16_gcc-7.5.0_musl_eabi/include  -c
../src/node_http2.cc: In constructor 'node::http2::Http2Options::Http2Options(node::Environment*, node::http2::nghttp2_session_type)':
../src/node_http2.cc:208:5: error: 'nghttp2_option_set_max_settings' was not declared in this scope
     nghttp2_option_set_max_settings(
     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
../src/node_http2.cc:208:5: note: suggested alternative: 'nghttp2_session_get_local_settings'
     nghttp2_option_set_max_settings(
     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
     nghttp2_session_get_local_settings
libnode.target.mk:228: recipe for target '/path/build/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/node-v12.18.0/out/Release/obj.target/libnode/src/node_http2.o' failed
make[4]: *** [/path/build/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/node-v12.18.0/out/Release/obj.target/libnode/src/node_http2.o] Error 1
```

If you need more logs, I'd be glad to provide them.

sans-ltd, 2020-06-08:
I'm also on this problem. The issue is that node does not pick the nghttp2 header from its deps folder but from staging_dir. The nghttp2 (1.39.2) there from openwrt is too old and lacks the symbol it complains about.
But I have no clue for now how to fix this.

nxhack, 2020-06-09:
You need to backport the changes applied to master.

https://github.com/openwrt/openwrt/commit/4e65838871e27b476e910f4fdfb746dd61998e43

sans-ltd, 2020-06-09:
Yes this works, thanks nxhack!

BKPepe, 2020-06-09:
> You need to backport the changes applied to master.

Shouldn't we rather ask OpenWrt main repo to backport it to OpenWrt 19.07? Also, I think it can be good if there is a branch for openwrt-19.07 the same as in the packages feed.

BKPepe, 2020-07-05:
Closing as nghttp2 was updated in OpenWrt 19.07!

Good job, @ja-pa !

closed 2020-07-05 by BKPepe