nxhack/openwrt-node-packages #680: Node Red not listed in System/Startup

sans-ltd, 2020-06-12:
In `node-red/files/node-red.init` the line
`START=99`
is missing. Would be great if somebody could add it. I would do it on myself but I currently do not have the permissions to do so. 
Another good thing would be to run node-red as a separate user with a special home directory (e.g. under `/var/lib/node-red`) because for some node-red applications it is sometimes necessary to make modifications in `$HOME/.node-red`
It is not nice that this is currently `/root/.node-red` especially when the full application should be packaged later in development.
In addition, running such a complex application as root is a big security threat.
I could help to achieve this, if you like.

BKPepe, 2020-06-16:
ping @ja-pa as he added init script

nxhack, 2020-06-26:
Sorry for the delay in responding, fix.

# The coronavirus has prevented me from responding to you as usual...

closed 2020-06-26 by nxhack