nxhack/openwrt-node-packages #693: Build problem OpenWrt 18.06 r8043

Raivon, 2020-08-13:
Traceback (most recent call last):
  File "../../deps/v8/tools/run.py", line 12, in <module>
    sys.exit(subprocess.call(sys.argv[1:]))
  File "/usr/lib/python2.7/subprocess.py", line 172, in call
    return Popen(*popenargs, **kwargs).wait()
  File "/usr/lib/python2.7/subprocess.py", line 394, in __init__
    errread, errwrite)
  File "/usr/lib/python2.7/subprocess.py", line 1047, in _execute_child
    raise child_exception
OSError: [Errno 8] Exec format error
make[5]: *** [tools/v8_gypfiles/generate_bytecode_builtins_list.target.mk:13: /home/vova/openwrt/build_dir/target-mips_24kc_musl/node-v12.14.0/out/Release/obj/gen/generate-bytecode-output-root/builtins-generated/bytecodes-builtins-list.h] Error 1
make[4]: *** [Makefile:101: node] Error 2
make[4]: Leaving directory '/home/vova/openwrt/build_dir/target-mips_24kc_musl/node-v12.14.0'
make[3]: *** [Makefile:197: /home/vova/openwrt/build_dir/target-mips_24kc_musl/node-v12.14.0/.built] Error 2
make[3]: Leaving directory '/home/vova/openwrt/feeds/node/node'
time: package/feeds/node/node/compile#0.28#1.20#1.48
make[2]: *** [package/Makefile:113: package/feeds/node/node/compile] Error 2
make[2]: Leaving directory '/home/vova/openwrt'
make[1]: *** [package/Makefile:107: /home/vova/openwrt/staging_dir/target-mips_24kc_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/vova/openwrt'
make: *** [/home/vova/openwrt/include/toplevel.mk:218: world] Error 2

Build system Debian 10.3 x64

nxhack, 2020-08-14:
@Raivon 

I was able to build it here.
Can you show me a slightly earlier log of this log?

Raivon, 2020-08-14:
[buildlog.txt](https://github.com/nxhack/openwrt-node-packages/files/5072619/buildlog.txt)
This is a complete log

nxhack, 2020-08-14:
@Raivon 

Run this command and let me know the result
```
file /home/vova/openwrt/build_dir/target-mips_24kc_musl/node-v12.14.0/out/Release/bytecode_builtins_list_generator
```

nxhack, 2020-08-14:
You probably don't have multilib installed in your host environment.
Run this command to install multilib.

```
sudo apt-get install gcc-multilib g++-multilib
```

Raivon, 2020-08-16:
> @Raivon
> 
> Run this command and let me know the result
> 
> ```
> file /home/vova/openwrt/build_dir/target-mips_24kc_musl/node-v12.14.0/out/Release/bytecode_builtins_list_generator
> ```
`/home/vova/openwrt/build_dir/target-mips_24kc_musl/node-v12.14.0/out/Release/bytecode_builtins_list_generator: ELF 32-bit LSB pie executable, Intel 80386, version 1 (GNU/Linux), dynamically linked, interpreter /lib/ld-linux.so.2, for GNU/Linux 3.2.0, BuildID[sha1]=08f1f4ff2b5fc3aa50c1af2291a4e23bd2d25403, not stripped`

> You probably don't have multilib installed in your host environment.
> Run this command to install multilib.
> 
> ```
> sudo apt-get install gcc-multilib g++-multilib
> ```
I have installed libraries beforehand
`g++-multilib is already the newest version (4:8.3.0-1).
gcc-multilib is already the newest version (4:8.3.0-1).`

Maybe the problem is in the Windows WSL ? But other modules build without errors.

nxhack, 2020-08-17:
https://github.com/microsoft/WSL/issues/711

Raivon, 2020-08-20:
I confirm. The problem is in the Windows WSL subsystem.
After upgrade Windows 10 to build 2004 or later you must upgrade installed WSL images to WSL2 or reinstall it.
Thank you!

closed 2020-08-20 by Raivon