nxhack/openwrt-node-packages #695: node-homebridge-cmd4: Node modules dependency problem

nxhack, 2020-08-26:
From this issue: https://github.com/openwrt/packages/issues/13196

zokl, 2020-08-26:
> @zokl
> 
> The "deasync" module, which is a dependency, was a binnary module. I'll think of a solution.

The same issue afected instalation of other packages - homebridge-camera-ffmpeg

nxhack, 2020-08-26:
```
homebridge-cmd4
  +system-sleep
    +deasync-promise
      +deasync (native module)
```
Because of this dependency structure, I've separated the packages so that deacync can build them successfully.

The build worked fine, but the homebridge plugin test seems to have failed.

```
Wed Aug 26 07:21:43 2020 daemon.err homebridge[4179]: [8/26/2020, 7:21:43 AM] No plugins found. See the README for information on installing plugins.
Wed Aug 26 07:21:43 2020 daemon.info homebridge[4179]: [8/26/2020, 7:21:43 AM] Loading 1 platforms...
Wed Aug 26 07:21:43 2020 daemon.err homebridge[4179]: [8/26/2020, 7:21:43 AM] Error loading platform requested in your config.json at position 1
Wed Aug 26 07:21:43 2020 daemon.err homebridge[4179]: [8/26/2020, 7:21:43 AM] Error: The requested platform 'Cmd4' was not registered by any plugin.
Wed Aug 26 07:21:43 2020 daemon.err homebridge[4179]:     at PluginManager.getPluginForPlatform (/usr/lib/node/homebridge/src/pluginManager.ts:224:15)
Wed Aug 26 07:21:43 2020 daemon.err homebridge[4179]:     at /usr/lib/node/homebridge/src/server.ts:387:37
Wed Aug 26 07:21:43 2020 daemon.err homebridge[4179]:     at Array.forEach (<anonymous>)
Wed Aug 26 07:21:43 2020 daemon.err homebridge[4179]:     at Server.loadPlatforms (/usr/lib/node/homebridge/src/server.ts:374:27)
Wed Aug 26 07:21:43 2020 daemon.err homebridge[4179]:     at Server.start (/usr/lib/node/homebridge/src/server.ts:153:29)
Wed Aug 26 07:21:43 2020 daemon.err homebridge[4179]:     at cli (/usr/lib/node/homebridge/src/cli.ts:80:10)
Wed Aug 26 07:21:43 2020 daemon.err homebridge[4179]:     at Object.<anonymous> (/usr/lib/node/homebridge/bin/homebridge:17:22)
Wed Aug 26 07:21:43 2020 daemon.err homebridge[4179]:     at Module._compile (internal/modules/cjs/loader.js:1137:30)
Wed Aug 26 07:21:43 2020 daemon.err homebridge[4179]:     at Object.Module._extensions..js (internal/modules/cjs/loader.js:1157:10)
Wed Aug 26 07:21:43 2020 daemon.err homebridge[4179]:     at Module.load (internal/modules/cjs/loader.js:985:32)
```

Verification please.

nxhack, 2020-08-26:
On the target machine.
```
npm install -g homebridge-cmdswitch2
```
Change config.json to cmdswitch2 and it starts up fine. Homebridge seems to be working fine.

```
Wed Aug 26 07:54:14 2020 daemon.info homebridge[8228]: [8/26/2020, 7:54:14 AM] Loaded plugin: homebridge-cmdswitch2@0.2.10
Wed Aug 26 07:54:14 2020 daemon.info homebridge[8228]: [8/26/2020, 7:54:14 AM] Registering platform 'homebridge-cmdswitch2.cmdSwitch2'
Wed Aug 26 07:54:14 2020 daemon.info homebridge[8228]: [8/26/2020, 7:54:14 AM] ---
Wed Aug 26 07:54:14 2020 daemon.info homebridge[8228]: [8/26/2020, 7:54:14 AM] Loading 1 platforms...
Wed Aug 26 07:54:14 2020 daemon.info homebridge[8228]: [8/26/2020, 7:54:14 AM] [cmdSwitch2] Initializing cmdSwitch2 platform...
Wed Aug 26 07:54:14 2020 daemon.info homebridge[8228]: Setup Payload:
```

nxhack, 2020-08-26:
Now that we've identified the reason why the plugin was not detected, we'll fix it.

zokl, 2020-08-26:
> ```
> homebridge-cmd4
>   +system-sleep
>     +deasync-promise
>       +deasync (native module)
> ```
>
>
> Because of this dependency structure, I've separated the packages so that deacync can build them [...]
I'll try it on a live system over the weekend.

Is there a reason why you put modules in /usr/lib/node instead of /usr/lib/node_modules? The default npm install uses /usr/lib/node_modules

nxhack, 2020-08-26:
Changed the path to be managed by OpenWrt's opkg package manager. We changed the path to add a search path to the node program itself.

The changes to homebridge are necessary because of homebridge search for the plugin module path itself.

nxhack, 2020-08-27:
It seems to be working.
```
Thu Aug 27 02:00:37 2020 daemon.info homebridge[3604]: [8/27/2020, 2:00:37 AM] Loaded plugin: homebridge-cmd4@2.3.2
Thu Aug 27 02:00:37 2020 daemon.info homebridge[3604]: [8/27/2020, 2:00:37 AM] Registering accessory 'homebridge-cmd4.Cmd4'
Thu Aug 27 02:00:37 2020 daemon.info homebridge[3604]: [8/27/2020, 2:00:37 AM] Registering platform 'homebridge-cmd4.Cmd4'
Thu Aug 27 02:00:39 2020 daemon.info homebridge[3604]: [8/27/2020, 2:00:39 AM] ---
Thu Aug 27 02:00:39 2020 daemon.info homebridge[3604]: [8/27/2020, 2:00:39 AM] Loading 1 platforms...
Thu Aug 27 02:00:39 2020 daemon.info homebridge[3604]: [8/27/2020, 2:00:39 AM] [Cmd4] Initializing Cmd4 platform...
Thu Aug 27 02:00:39 2020 daemon.info homebridge[3604]: [8/27/2020, 2:00:39 AM] [Cmd4] Fetching config.json devices.
Thu Aug 27 02:00:39 2020 daemon.info homebridge[3604]: [8/27/2020, 2:00:39 AM] [Cmd4] Processing accessory: PS_4
Thu Aug 27 02:00:39 2020 daemon.info homebridge[3604]: [8/27/2020, 2:00:39 AM] [Cmd4] Processing accessory: My_v1Fan
Thu Aug 27 02:00:39 2020 daemon.info homebridge[3604]: [8/27/2020, 2:00:39 AM] [Cmd4] Processing accessory: My_Fan
Thu Aug 27 02:00:39 2020 daemon.info homebridge[3604]: [8/27/2020, 2:00:39 AM] [Cmd4] Processing accessory: My_v2Fan
Thu Aug 27 02:00:40 2020 daemon.info homebridge[3604]: [8/27/2020, 2:00:40 AM] [Cmd4] Processing accessory: My_GarageDoorOpener
Thu Aug 27 02:00:40 2020 daemon.info homebridge[3604]: [8/27/2020, 2:00:40 AM] [Cmd4] Processing accessory: My_OnOff_Light
Thu Aug 27 02:00:40 2020 daemon.info homebridge[3604]: [8/27/2020, 2:00:40 AM] [Cmd4] Processing accessory: My_Dimmable_Light
Thu Aug 27 02:00:40 2020 daemon.info homebridge[3604]: [8/27/2020, 2:00:40 AM] [Cmd4] Processing accessory: My_LockManagement
Thu Aug 27 02:00:40 2020 daemon.err homebridge[3604]: [8/27/2020, 2:00:40 AM] [Cmd4] ****** Removing TLV8 required characteristic :LockControlPoint
Thu Aug 27 02:00:40 2020 daemon.err homebridge[3604]: [8/27/2020, 2:00:40 AM] [Cmd4] ****** Removing TLV8 required characteristic :Logs
Thu Aug 27 02:00:40 2020 daemon.info homebridge[3604]: [8/27/2020, 2:00:40 AM] [Cmd4] Processing accessory: My_LockMechanism
Thu Aug 27 02:00:40 2020 daemon.info homebridge[3604]: [8/27/2020, 2:00:40 AM] [Cmd4] Processing accessory: My_Outlet
Thu Aug 27 02:00:40 2020 daemon.info homebridge[3604]: [8/27/2020, 2:00:40 AM] [Cmd4] Processing accessory: My_Switch
Thu Aug 27 02:00:40 2020 daemon.info homebridge[3604]: [8/27/2020, 2:00:40 AM] [Cmd4] Processing accessory: My_Thermostat
Thu Aug 27 02:00:40 2020 daemon.info homebridge[3604]: [8/27/2020, 2:00:40 AM] [Cmd4] Processing accessory: My_AirQualitySensor
Thu Aug 27 02:00:40 2020 daemon.info homebridge[3604]: [8/27/2020, 2:00:40 AM] [Cmd4] Processing accessory: My_SecuritySystem
Thu Aug 27 02:00:40 2020 daemon.info homebridge[3604]: [8/27/2020, 2:00:40 AM] [Cmd4] Processing accessory: My_CO_Sensor
Thu Aug 27 02:00:40 2020 daemon.info homebridge[3604]: [8/27/2020, 2:00:40 AM] [Cmd4] Processing accessory: My_ContactSensor
Thu Aug 27 02:00:40 2020 daemon.info homebridge[3604]: [8/27/2020, 2:00:40 AM] [Cmd4] Processing accessory: My_Door
Thu Aug 27 02:00:40 2020 daemon.info homebridge[3604]: [8/27/2020, 2:00:40 AM] [Cmd4] Processing accessory: My_HumiditySensor
Thu Aug 27 02:00:40 2020 daemon.err homebridge[3604]: [8/27/2020, 2:00:40 AM] [Cmd4] config.storage:fs for:My_HumiditySensor set but polling is not enabled.
Thu Aug 27 02:00:40 2020 daemon.err homebridge[3604]: [8/27/2020, 2:00:40 AM] [Cmd4]       History will not be updated continiously.
Thu Aug 27 02:00:40 2020 daemon.info homebridge[3604]: [8/27/2020, 2:00:40 AM] [Cmd4] Processing accessory: My_LeakSensor
Thu Aug 27 02:00:40 2020 daemon.info homebridge[3604]: [8/27/2020, 2:00:40 AM] [Cmd4] Processing accessory: My_LightSensor
Thu Aug 27 02:00:40 2020 daemon.info homebridge[3604]: [8/27/2020, 2:00:40 AM] [Cmd4] Processing accessory: My_MotionSensor
Thu Aug 27 02:00:40 2020 daemon.info homebridge[3604]: [8/27/2020, 2:00:40 AM] [Cmd4] Processing accessory: My_OccupancySensor
Thu Aug 27 02:00:40 2020 daemon.info homebridge[3604]: [8/27/2020, 2:00:40 AM] [Cmd4] Processing accessory: My_SmokeSensor
Thu Aug 27 02:00:40 2020 daemon.info homebridge[3604]: [8/27/2020, 2:00:40 AM] [Cmd4] Processing accessory: My_StatefulProgrammableSwitch
Thu Aug 27 02:00:40 2020 daemon.info homebridge[3604]: [8/27/2020, 2:00:40 AM] [Cmd4] Processing accessory: My_StatelessProgrammableSwitch
Thu Aug 27 02:00:40 2020 daemon.info homebridge[3604]: [8/27/2020, 2:00:40 AM] [Cmd4] Processing accessory: My_TemperatureSensor
Thu Aug 27 02:00:40 2020 daemon.err homebridge[3604]: HAP Warning: Characteristic 00000010-0000-1000-8000-0026BB765291 not in required or optional characteristics for service 0000008A-0000-1000-8000-0026BB765291. Adding anyway.
Thu Aug 27 02:00:40 2020 daemon.info homebridge[3604]: [8/27/2020, 2:00:40 AM] [Cmd4] Processing accessory: My_Window
Thu Aug 27 02:00:40 2020 daemon.info homebridge[3604]: [8/27/2020, 2:00:40 AM] [Cmd4] Processing accessory: My_WindowCovering
Thu Aug 27 02:00:40 2020 daemon.info homebridge[3604]: [8/27/2020, 2:00:40 AM] [Cmd4] Processing accessory: My_BatteryService
Thu Aug 27 02:00:41 2020 daemon.info homebridge[3604]: [8/27/2020, 2:00:41 AM] [Cmd4] Processing accessory: My_CO2_Sensor
Thu Aug 27 02:00:41 2020 daemon.info homebridge[3604]: [8/27/2020, 2:00:41 AM] [Cmd4] Processing accessory: My_Microphone
[... cut ...]
Thu Aug 27 02:00:42 2020 daemon.info homebridge[3604]: [8/27/2020, 2:00:42 AM] Initializing platform accessory 'My_TimeInformation'...
Thu Aug 27 02:00:42 2020 daemon.info homebridge[3604]: [8/27/2020, 2:00:42 AM] Initializing platform accessory 'My_Valve'...
Thu Aug 27 02:00:42 2020 daemon.info homebridge[3604]: [8/27/2020, 2:00:42 AM] Initializing platform accessory 'My_HeaterCooler'...
Thu Aug 27 02:00:42 2020 daemon.info homebridge[3604]: [8/27/2020, 2:00:42 AM] Initializing platform accessory 'My_HumidifierDehumidifier'...
Thu Aug 27 02:00:42 2020 daemon.info homebridge[3604]: Setup Payload:
```

zokl, 2020-08-27:
Yes, i try it and the plugin was load correctly.

zokl, 2020-08-27:
> Changed the path to be managed by OpenWrt's opkg package manager. We changed the path to add a search path to the node program itself.
> 
> The changes to homebridge are necessary because of homebridge search for the plugin module path itself.

In my packages modification I am using path /usr/lib/node_modules and it works. Opkg and compulation on host system too.

zokl, 2020-08-27:
One problem that I find is a running on homebridge user. May be it is for differend issue and maybe fir homebridge-cmd4 packager maintainer. 
If I call a shell script which use sysfs or call ubus command it does no work. If I change in homebridge init script running owner to root everything works.

nxhack, 2020-08-27:
I think it's a policy issue here. I don't want to run network services as root. In my environment, I was using sudo to kick the script from homebridge.

l leave it up to your policy.

zokl, 2020-08-27:
OK, sudo is a good solution. Thank you.

nxhack, 2021-05-05:
Close the issue. If the problem still exists, feel free to reopen it.

closed 2021-05-05 by nxhack