nxhack/openwrt-node-packages #731: Receiving missing dependency and reference to self errors....

xptsp, 2020-10-27:
After adding the repo to the `feeds.conf.default` file and running `make menuconfig`, I get these warnings/errors before seeing the menu:
```
WARNING: Makefile 'package/feeds/node/node-red-contrib-gpio/Makefile' has a dependency on 'node-red', which does not exist
WARNING: Makefile 'package/feeds/node/node-red-contrib-homekit/Makefile' has a dependency on 'node-red', which does not exist
WARNING: Makefile 'package/feeds/node/node-red-contrib-modbus/Makefile' has a dependency on 'node-red', which does not exist
WARNING: Makefile 'package/feeds/node/node-red-node-arduino/Makefile' has a dependency on 'node-red', which does not exist
tmp/.config-package.in:23497:error: recursive dependency detected!
tmp/.config-package.in:23497:	symbol PACKAGE_node-cylon-gpio is selected by PACKAGE_node-cylon-gpio
For a resolution refer to Documentation/kbuild/kconfig-language.rst
subsection "Kconfig recursive dependency limitations"

tmp/.config-package.in:23534:error: recursive dependency detected!
tmp/.config-package.in:23534:	symbol PACKAGE_node-cylon-i2c is selected by PACKAGE_node-cylon-i2c
```

Here is my `feeds.conf.default` file:
```
src-git packages https://git.openwrt.org/feed/packages.git
src-git luci https://git.openwrt.org/project/luci.git
src-git routing https://git.openwrt.org/feed/routing.git
src-git telephony https://git.openwrt.org/feed/telephony.git
src-git freifunk https://github.com/freifunk/openwrt-packages.git
#src-git video https://github.com/openwrt/video.git
#src-git targets https://github.com/openwrt/targets.git
#src-git management https://github.com/openwrt-management/packages.git
#src-git oldpackages http://git.openwrt.org/packages.git
#src-link custom /usr/src/openwrt/custom-feed
src-git node https://github.com/nxhack/openwrt-node-packages.git
```

I would really like to add Node-RED to my OpenWRT image, but am having problems doing so!  Please help!

nxhack, 2020-10-28:
https://github.com/nxhack/openwrt-node-packages#usage

```
./scripts/feeds update node
rm ./package/feeds/packages/node
rm ./package/feeds/packages/node-*
./scripts/feeds install -a -p node
```

xptsp, 2020-10-28:
Done those commands.  Duplicate dependency errors are gone, but still getting:
```
Collecting package info: done
WARNING: Makefile 'package/feeds/node/node-red-contrib-gpio/Makefile' has a dependency on 'node-red', which does not exist
WARNING: Makefile 'package/feeds/node/node-red-contrib-homekit/Makefile' has a dependency on 'node-red', which does not exist
WARNING: Makefile 'package/feeds/node/node-red-contrib-modbus/Makefile' has a dependency on 'node-red', which does not exist
WARNING: Makefile 'package/feeds/node/node-red-node-arduino/Makefile' has a dependency on 'node-red', which does not exist
```

nxhack, 2020-10-28:
```
make defconfig 
```

xptsp, 2020-10-29:
```
doug@Doug-PC:~/openwrt$ make defconfig
WARNING: Makefile 'package/feeds/node/node-red-contrib-gpio/Makefile' has a dependency on 'node-red', which does not exist
WARNING: Makefile 'package/feeds/node/node-red-contrib-homekit/Makefile' has a dependency on 'node-red', which does not exist
WARNING: Makefile 'package/feeds/node/node-red-contrib-modbus/Makefile' has a dependency on 'node-red', which does not exist
WARNING: Makefile 'package/feeds/node/node-red-node-arduino/Makefile' has a dependency on 'node-red', which does not exist
#
# No change to .config
#
doug@Doug-PC:~/openwrt$ make menuconfig
WARNING: Makefile 'package/feeds/node/node-red-contrib-gpio/Makefile' has a dependency on 'node-red', which does not exist
WARNING: Makefile 'package/feeds/node/node-red-contrib-homekit/Makefile' has a dependency on 'node-red', which does not exist
WARNING: Makefile 'package/feeds/node/node-red-contrib-modbus/Makefile' has a dependency on 'node-red', which does not exist
WARNING: Makefile 'package/feeds/node/node-red-node-arduino/Makefile' has a dependency on 'node-red', which does not exist

*** End of the configuration.
*** Execute 'make' to start the build or try 'make help'.
```

xptsp, 2020-10-29:
Updated feeds again.  Still getting:
```
doug@Doug-PC:~/openwrt$ ./scripts/feeds install -a -p node
WARNING: Makefile 'package/feeds/node/node-red-contrib-gpio/Makefile' has a dependency on 'node-red', which does not exist
WARNING: Makefile 'package/feeds/node/node-red-contrib-homekit/Makefile' has a dependency on 'node-red', which does not exist
WARNING: Makefile 'package/feeds/node/node-red-contrib-modbus/Makefile' has a dependency on 'node-red', which does not exist
WARNING: Makefile 'package/feeds/node/node-red-node-arduino/Makefile' has a dependency on 'node-red', which does not exist
Installing all packages from feed node.
```

xptsp, 2020-10-29:
```
doug@Doug-PC:~/openwrt$ ./scripts/feeds update node
Updating feed 'node' from 'https://github.com/nxhack/openwrt-node-packages.git' ...
Already up to date.
Create index file './feeds/node.index' 
doug@Doug-PC:~/openwrt$ rm ./package/feeds/packages/node
rm: cannot remove './package/feeds/packages/node': No such file or directory
doug@Doug-PC:~/openwrt$ rm ./package/feeds/packages/node-*
rm: cannot remove './package/feeds/packages/node-*': No such file or directory
doug@Doug-PC:~/openwrt$ ./scripts/feeds install -a -p node
WARNING: Makefile 'package/feeds/node/node-red-contrib-gpio/Makefile' has a dependency on 'node-red', which does not exist
WARNING: Makefile 'package/feeds/node/node-red-contrib-homekit/Makefile' has a dependency on 'node-red', which does not exist
WARNING: Makefile 'package/feeds/node/node-red-contrib-modbus/Makefile' has a dependency on 'node-red', which does not exist
WARNING: Makefile 'package/feeds/node/node-red-node-arduino/Makefile' has a dependency on 'node-red', which does not exist
Installing all packages from feed node.
doug@Doug-PC:~/openwrt$ make defconfig 
WARNING: Makefile 'package/feeds/node/node-red-contrib-gpio/Makefile' has a dependency on 'node-red', which does not exist
WARNING: Makefile 'package/feeds/node/node-red-contrib-homekit/Makefile' has a dependency on 'node-red', which does not exist
WARNING: Makefile 'package/feeds/node/node-red-contrib-modbus/Makefile' has a dependency on 'node-red', which does not exist
WARNING: Makefile 'package/feeds/node/node-red-node-arduino/Makefile' has a dependency on 'node-red', which does not exist
#
# No change to .config
#
doug@Doug-PC:~/openwrt$ 
```

nxhack, 2020-10-29:
That's very strange. Please try this one.
```
./scripts/feeds uninstall node-red
rm -rf ./tmp
./scripts/feeds install node-red
make defconfig
```

xptsp, 2020-10-29:
Yay!  I'm not receiving the missing dependency errors anymore!  Trying to build my OpenWRT install atm...

xptsp, 2020-10-29:
Okay, that didn't work.  So I removed my entire OpenWRT directory and pulled it again.  Followed the instructions on how to add your repo to the mix and now no missing dependency or reference to self errors are present.  Unfortunately, I have to select all my packages again, so further testing will take some time while I wait for my system to compile everything again.... (sigh)

xptsp, 2020-10-29:
After many hours of compiling and impatiently waiting, I inserted the compiled image into my BPi R2 and SUCCESS!!!  I'm so happy!  Thanks for all your help!

closed 2020-10-29 by xptsp