nxhack/openwrt-node-packages #736: OpenWrt HEAD drop KERNEL_MIPS_FPU_EMULATOR

nxhack, 2020-11-04:
In the latest version of OpenWrt, support for MIPS FPU emulators has disappeared. 
This removes mips(eb,el) and mips64(eb,el) from the active list.

https://github.com/openwrt/openwrt/commit/ac5671f46cb4a300e2a8ae00db70c248828367f5

nxhack, 2020-11-04:
linux kernel 5.x drop this option.
OpenWrt-18.06 linux-4.14 has this option.

Anyway, the development branch does not support mips.

predators46, 2021-08-22:
{standard input}: Assembler messages:
{standard input}:126: Error: opcode not supported on this processor: mips32r2 (mips32r2) `ldc1 $f0,24($fp)'
{standard input}:127: Error: opcode not supported on this processor: mips32r2 (mips32r2) `mtc1 $t0,$f1'
{standard input}:128: Error: opcode not supported on this processor: mips32r2 (mips32r2) `sdc1 $f0,24($fp)'

closed 2023-04-19 by nxhack