nxhack/openwrt-node-packages #758: build error openwrt 19.07.4 nodejs 14

p3x-robot, 2020-11-21:
ciao, i configured from node 12 to 14 and i get this error. do you have an idea?

```txt
arm-openwrt-linux-muslgnueabi-gcc-ar crsT /build/source/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/node-v14.15.1/out/Release/obj.target/deps/uv/libuv.a @/build/s$
  arm-openwrt-linux-muslgnueabi-gcc -Os -pipe -o /build/source/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/node-v14.15.1/out/Release/obj.target/uvwasi/deps/uvwasi$
  arm-openwrt-linux-muslgnueabi-gcc -Os -pipe -o /build/source/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/node-v14.15.1/out/Release/obj.target/uvwasi/deps/uvwasi$
  arm-openwrt-linux-muslgnueabi-gcc -Os -pipe -o /build/source/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/node-v14.15.1/out/Release/obj.target/uvwasi/deps/uvwasi$
  arm-openwrt-linux-muslgnueabi-gcc -Os -pipe -o /build/source/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/node-v14.15.1/out/Release/obj.target/uvwasi/deps/uvwasi$
  arm-openwrt-linux-muslgnueabi-gcc -Os -pipe -o /build/source/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/node-v14.15.1/out/Release/obj.target/uvwasi/deps/uvwasi$
  arm-openwrt-linux-muslgnueabi-gcc -Os -pipe -o /build/source/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/node-v14.15.1/out/Release/obj.target/uvwasi/deps/uvwasi$
  arm-openwrt-linux-muslgnueabi-gcc -Os -pipe -o /build/source/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/node-v14.15.1/out/Release/obj.target/uvwasi/deps/uvwasi$
  arm-openwrt-linux-muslgnueabi-gcc -Os -pipe -o /build/source/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/node-v14.15.1/out/Release/obj.target/uvwasi/deps/uvwasi$
rm -f /build/source/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/node-v14.15.1/out/Release/obj.target/deps/uvwasi/libuvwasi.a ar-file-list; mkdir -p `dirname /buil$
arm-openwrt-linux-muslgnueabi-gcc-ar crsT /build/source/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/node-v14.15.1/out/Release/obj.target/deps/uvwasi/libuvwasi.a @$
  LD_LIBRARY_PATH=/build/source/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/node-v14.15.1/out/Release/lib.host:/build/source/build_dir/target-arm_cortex-a9+vfpv3-$
  arm-openwrt-linux-muslgnueabi-g++ -Os -pipe -o /build/source/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/node-v14.15.1/out/Release/obj.target/libnode/src/api/as$
  arm-openwrt-linux-muslgnueabi-g++ -Os -pipe -o /build/source/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/node-v14.15.1/out/Release/obj.target/libnode/src/api/ca$
  arm-openwrt-linux-muslgnueabi-g++ -Os -pipe -o /build/source/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/node-v14.15.1/out/Release/obj.target/libnode/src/api/en$
  arm-openwrt-linux-muslgnueabi-g++ -Os -pipe -o /build/source/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/node-v14.15.1/out/Release/obj.target/libnode/src/api/en$
../src/api/environment.cc: In function 'void node::DefaultProcessExitHandler(node::Environment*, int)':
../src/api/environment.cc:712:3: error: 'uv_library_shutdown' was not declared in this scope
   uv_library_shutdown();
   ^~~~~~~~~~~~~~~~~~~
../src/api/environment.cc:712:3: note: suggested alternative: 'uv_shutdown'
   uv_library_shutdown();
   ^~~~~~~~~~~~~~~~~~~
   uv_shutdown
libnode.target.mk:223: recipe for target '/build/source/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/node-v14.15.1/out/Release/obj.target/libnode/src/api/environme$
make[5]: *** [/build/source/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/node-v14.15.1/out/Release/obj.target/libnode/src/api/environment.o] Error 1
Makefile:104: recipe for target 'node' failed
make[4]: *** [node] Error 2
make[4]: Leaving directory '/build/source/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/node-v14.15.1'
Makefile:193: recipe for target '/build/source/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/node-v14.15.1/.built' failed
make[3]: *** [/build/source/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/node-v14.15.1/.built] Error 2
time: package/feeds/node/node/compile#20.45#2.06#21.42
```

p3x-robot, 2020-11-21:
what does this mean?
**The libatomic package is required. (Install the 32-bit library at the same time, if necessary).**

is it inside openwrt or in the host i have to install it (libatomic)?

p3x-robot, 2020-11-21:
it says:
```txt
libatomic1/stable,now 8.3.0-6 amd64 [installed,automatic]
```

p3x-robot, 2020-11-21:
i am building on a linksys wrt3200acm

p3x-robot, 2020-11-21:
the first build was on docker stretch container, now running the buster container, maybe that works...

p3x-robot, 2020-11-21:
i built the same config i just replaced from v12 to v14...

p3x-robot, 2020-11-21:
ok, same error on different container. full error log:
https://cdn.corifeus.com/issues/ISSUE-openwrt-19.07.4-node-v15.txt

nxhack, 2020-11-22:
Is the version of libuv 1.40.0? The latest version of node.js requires 1.40.0.

p3x-robot, 2020-11-22:
so with the openwrt 19.07.4 it is not working? Or do I have to install something else as well?

p3x-robot, 2020-11-22:
i have not installed libuv....

p3x-robot, 2020-11-22:
so i guess i have to create a new Makefile for libuv and override?

p3x-robot, 2020-11-22:
ok i copied the libuv 1.4 Makefile from master and package in 19.07.4

thanks for the tip!

closed 2020-11-22 by p3x-robot