nxhack/openwrt-node-packages #775: error: 'uv_library_shutdown' was not declared in this scope

itProfi, 2020-11-27:
Good day! When compile form source openwrt 19.07  - under Debian 10 i always try this error:

nxhack, 2020-11-27:
Check feeds.conf. Probably the packages feed of 19.07.4 has an older version of libuv.

Change the relevant line to this if possible.
```
rc-git packages https://git.openwrt.org/feed/packages.git;openwrt-19.07
```

If I get more issues of this kind, I will make changes to my node package.

itProfi, 2020-11-27:
> Check feeds.conf. Probably the packages feed of 19.07.4 has an older version of libuv.
> 
> Change the relevant line to this if possible.
> 
> ```
> rc-git packages https://git.openwrt.org/feed/packages.git;openwrt-19.07
> ```
> 
> If I get more issues of this kind, I will make changes to my node package.

I use this line in feeds.conf

nxhack, 2020-11-27:
This was not the case?
```
src-git packages https://git.openwrt.org/feed/packages.git^eae1bb397f064befe003b01a36ae318321e81d35
```

itProfi, 2020-11-27:
I try build opewrt for this branche - https://github.com/devbis/openwrt/tree/lumi-19.07
This my feeds.conf
```
src-git packages https://git.openwrt.org/feed/packages.git^eae1bb397f064befe003b01a36ae318321e81d35
src-git luci https://git.openwrt.org/project/luci.git^0d0ab01a64f25d15325d3c98a7df252085382f1d
src-git routing https://git.openwrt.org/feed/routing.git^b77498bd56d5e45ab4577a1f4ad6ffc55b4a86b7
src-git telephony https://git.openwrt.org/feed/telephony.git^91b34e7eefa1d37a18390d0c6881978cd76804a1
src-git node https://github.com/nxhack/openwrt-node-packages.git;openwrt-19.07
````
Maybe this reason?

nxhack, 2020-11-27:
"packages" line.

Since there seems to be a lot of issues, I'll just fix my 19.07 branch for now.

Please be patient, I will fix my 19.07 branch.

nxhack, 2020-11-27:
same : https://github.com/nxhack/openwrt-node-packages/issues/758

nxhack, 2020-11-27:
Fixed.
 Set shared-libuv to disable.

@itProfi  test please.

itProfi, 2020-11-27:
> Fixed.
> Set shared-libuv to disable.
> 
> @itProfi test please.

All works fine! Thanks!
[image]

closed 2020-11-27 by itProfi