nxhack/openwrt-node-packages #857: Zigbee2mqtt error

ANARHIST1984, 2021-03-16:
Hello
I have router WE3526 MediaTek MT7621 and install there Openwrt 19.07.07
I manage to compile
node_v12.20.0-2_mipsel_24kc
node-npm_v12.20.0-2_mipsel_24kc.ipk
node-zigbee2mqtt_1.17.1-1_mipsel_24kc.ipk

Stik for zigbee is CC2531

when i try to start z2m i have error
when i try to start "npm start" i receive this
root@OpenWrt:/opt/zigbee2mqtt# npm start
npm info it worked if it ends with ok
npm info using npm@6.14.8
npm info using node@v14.15.1
npm info lifecycle zigbee2mqtt@1.16.2prestart: zigbee2mqtt@1.16.2
npm info lifecycle zigbee2mqtt@1.16.2start: zigbee2mqtt@1.16.2

zigbee2mqtt@1.16.2 start /opt/zigbee2mqtt
node index.js

Zigbee2MQTT:info 2021-03-14 22:43:56: Logging to console and directory: '/opt/zigbee2mqtt/data/log/2021-03-14.22-43-49' filename: log.txt
Zigbee2MQTT:info 2021-03-14 22:43:57: Starting Zigbee2MQTT version 1.16.2 (commit #unknown)
Zigbee2MQTT:info 2021-03-14 22:43:57: Starting zigbee-herdsman (0.13.37)
Zigbee2MQTT:error 2021-03-14 22:43:57: Error while starting zigbee-herdsman
Zigbee2MQTT:error 2021-03-14 22:43:57: Failed to start zigbee
Zigbee2MQTT:error 2021-03-14 22:43:57: Exiting...
Zigbee2MQTT:error 2021-03-14 22:43:57: Error: Error relocating /opt/zigbee2mqtt/node_modules/zigbee-herdsman/node_modules/@serialport/bindings/build/Release/bindings.node: _ZN2v82V88MakeWeakEPPNS_8internal6ObjectEPvPFvRKNS_16WeakCallbackInfoIvEEENS_16WeakCallbackTypeE: symbol not found
at Object.Module._extensions..node (internal/modules/cjs/loader.js:1122:18)
at Module.load (internal/modules/cjs/loader.js:928:32)
at Function.Module._load (internal/modules/cjs/loader.js:769:14)
at Module.require (internal/modules/cjs/loader.js:952:19)
at require (internal/modules/cjs/helpers.js:88:18)
at bindings (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/node_modules/bindings/bindings.js:112:48)
at Object. (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/node_modules/@serialport/bindings/lib/linux.js:2:36)
at Module._compile (internal/modules/cjs/loader.js:1063:30)
at Object.Module._extensions..js (internal/modules/cjs/loader.js:1092:10)
at Module.load (internal/modules/cjs/loader.js:928:32)
npm info lifecycle zigbee2mqtt@1.16.2~start: Failed to exec start script
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! zigbee2mqtt@1.16.2 start: node index.js
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the zigbee2mqtt@1.16.2 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.
npm timing npm Completed in 20808ms

npm ERR! A complete log of this run can be found in:
npm ERR! /root/.npm/_logs/2021-03-14T22_43_58_259Z-debug.log

Stik CC2531
root@OpenWrt:/opt/zigbee2mqtt# ls -la /dev | grep ttyACM0
crw-rw---- 1 root dialout 166, 0 Mar 15 17:16 ttyACM0

configuration.yaml
Home Assistant integration (MQTT discovery)
homeassistant: false

I also try to use 
clear install z2m and receive this

awn args   'build',
gyp info spawn args   '-Goutput_dir=.'
gyp info spawn args ]
gyp info spawn make
gyp info spawn args [ 'BUILDTYPE=Release', '-C', 'build' ]
make: Entering directory '/opt/zigbee2mqtt/node_modules/zigbee-herdsman/node_modules/@serialport/bindings/build'
  CXX(target) Release/obj.target/bindings/src/serialport.o
  CXX(target) Release/obj.target/bindings/src/serialport_unix.o
../src/serialport_unix.cpp: In function 'int setup(int, OpenBaton*)':
../src/serialport_unix.cpp:166:5: warning: '%s' directive output may be truncated writing up to 1023 bytes into a region of size 1005 [-Wformat-truncation=]
 int setup(int fd, OpenBaton *data) {
     ^~~~~
../src/serialport_unix.cpp:176:13: note: 'snprintf' output 20 or more bytes (assuming 1043) into a destination of size 1024
     snprintf(data->errorString, sizeof(data->errorString), "Error %s Cannot open %s", strerror(errno), data->path);
     ~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
../src/serialport_unix.cpp: In function 'void EIO_Open(uv_work_t*)':
../src/serialport_unix.cpp:79:6: warning: '%s' directive output may be truncated writing up to 1023 bytes into a region of size 1003 [-Wformat-truncation=]
 void EIO_Open(uv_work_t* req) {
      ^~~~~~~~
../src/serialport_unix.cpp:86:13: note: 'snprintf' output 22 or more bytes (assuming 1045) into a destination of size 1024
     snprintf(data->errorString, sizeof(data->errorString), "Error: %s, cannot open %s", strerror(errno), data->path);
     ~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
  CXX(target) Release/obj.target/bindings/src/poller.o
  CXX(target) Release/obj.target/bindings/src/serialport_linux.o
  SOLINK_MODULE(target) Release/obj.target/bindings.node
/usr/bin/ld: cannot find -lpthread
collect2: error: ld returned 1 exit status
make: *** [bindings.target.mk:134: Release/obj.target/bindings.node] Error 1
make: Leaving directory '/opt/zigbee2mqtt/node_modules/zigbee-herdsman/node_modules/@serialport/bindings/build'
gyp ERR! build error 
gyp ERR! stack Error: `make` failed with exit code: 2
gyp ERR! stack     at ChildProcess.onExit (/usr/lib/node_modules/npm/node_modules/node-gyp/lib/build.js:194:23)
gyp ERR! stack     at ChildProcess.emit (events.js:315:20)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:277:12)
gyp ERR! System Linux 4.14.221
gyp ERR! command "/usr/bin/node" "/usr/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "rebuild"
gyp ERR! cwd /opt/zigbee2mqtt/node_modules/zigbee-herdsman/node_modules/@serialport/bindings
gyp ERR! node -v v14.15.1
gyp ERR! node-gyp -v v5.1.0
gyp ERR! not ok 
npm info lifecycle @serialport/bindings@9.0.4~install: Failed to exec install script
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! @serialport/bindings@9.0.4 install: `prebuild-install --tag-prefix @serialport/bindings@ || node-gyp rebuild`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the @serialport/bindings@9.0.4 install script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.
npm timing npm Completed in 143233ms

npm ERR! A complete log of this run can be found in:
npm ERR!     /root/.npm/_logs/2021-03-16T16_50_54_176Z-debug.log

debug log
1782 info lifecycle @serialport/bindings@9.0.4~install: @serialport/bindings@9.0.4
1783 verbose lifecycle @serialport/bindings@9.0.4~install: unsafe-perm in lifecycle false
1784 verbose lifecycle @serialport/bindings@9.0.4~install: PATH: /usr/lib/node_modules/npm/node_modules/npm-lifecycle/node-gyp-bin:/o>
1785 verbose lifecycle @serialport/bindings@9.0.4~install: CWD: /opt/zigbee2mqtt/node_modules/zigbee-herdsman/node_modules/@serialpor>
1786 silly lifecycle @serialport/bindings@9.0.4~install: Args: [
1786 silly lifecycle   '-c',
1786 silly lifecycle   'prebuild-install --tag-prefix @serialport/bindings@ || node-gyp rebuild'
1786 silly lifecycle ]
1787 silly lifecycle @serialport/bindings@9.0.4~install: Returned: code: 1  signal: null
1788 info lifecycle @serialport/bindings@9.0.4~install: Failed to exec install script
1789 verbose stack Error: @serialport/bindings@9.0.4 install: `prebuild-install --tag-prefix @serialport/bindings@ || node-gyp rebuil>
1789 verbose stack Exit status 1
1789 verbose stack     at EventEmitter.<anonymous> (/usr/lib/node_modules/npm/node_modules/npm-lifecycle/index.js:332:16)
1789 verbose stack     at EventEmitter.emit (events.js:315:20)
1789 verbose stack     at ChildProcess.<anonymous> (/usr/lib/node_modules/npm/node_modules/npm-lifecycle/lib/spawn.js:55:14)
1789 verbose stack     at ChildProcess.emit (events.js:315:20)
1789 verbose stack     at maybeClose (internal/child_process.js:1048:16)
1789 verbose stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:288:5)
1790 verbose pkgid @serialport/bindings@9.0.4
1791 verbose cwd /opt/zigbee2mqtt
1792 verbose Linux 4.14.221
1793 verbose argv "/usr/bin/node" "/usr/bin/npm" "rebuild"
1794 verbose node v14.15.1
1795 verbose npm  v6.14.8
1796 error code ELIFECYCLE
1797 error errno 1
1798 error @serialport/bindings@9.0.4 install: `prebuild-install --tag-prefix @serialport/bindings@ || node-gyp rebuild`
1798 error Exit status 1
1799 error Failed at the @serialport/bindings@9.0.4 install script.
1799 error This is probably not a problem with npm. There is likely additional logging output above.
1800 verbose exit [ 1, true ]

what i do wrong?

nxhack, 2021-03-16:
Hi @ANARHIST1984 

You say you have installed version 1.17.1 of zigbee2mqtt.
>I manage to compile
node_v12.20.0-2_mipsel_24kc
node-npm_v12.20.0-2_mipsel_24kc.ipk
node-zigbee2mqtt_1.17.1-1_mipsel_24kc.ipk

However, the logs show the output as 1.16.2.
>npm info lifecycle zigbee2mqtt@1.16.2prestart: zigbee2mqtt@1.16.2
npm info lifecycle zigbee2mqtt@1.16.2start: zigbee2mqtt@1.16.2

Do you have a different version of the package already installed?

Our zigbee2mqtt package is extracted to /opt/zigbee2mqtt.

Check /usr/lib/node_modules etc.

ANARHIST1984, 2021-03-17:
> Hi @ANARHIST1984
>
>
>
> You say you have installed version 1.17.1 of zigbee2mqtt.
>
>
> node_v12.20.0-2_mipsel_24kc
>
> node-npm_v12.20.0-2_mipsel_24kc.ipk
>
> node-zigbee2mqtt_1.17.1-1_mipsel_24kc.ipk [...]

I try different version of Z2m
16 and 17
And try clear install all.
Use different node 10,12,14.
Always same error!
Also. Stick inside the router or not in, same error!

nxhack, 2021-03-17:
@ANARHIST1984 

I see. Please give me some time to create a test environment.

ANARHIST1984, 2021-03-17:
> @ANARHIST1984 
> 
> 
> 
> I see. Please give me some time to create a test environment.

If need any assistant of information, please let me know.

nxhack, 2021-03-17:
>Zigbee2MQTT:error 2021-03-14 22:43:57: Error: Error relocating /opt/zigbee2mqtt/node_modules/zigbee-herdsman/node_modules/@serialport/bindings/build/Release/bindings.node: _ZN2v82V88MakeWeakEPPNS_8internal6ObjectEPvPFvRKNS_16WeakCallbackInfoIvEEENS_16WeakCallbackTypeE: symbol not found

It looks like a faile in the serialport module.
(There seems to be no error in my environment. mipsel and x86_64)

Can you please try it?
```
cd /opt/zigbee2mqtt/node_modules/zigbee-herdsman/node_modules/serialport
node test.js
```
```

nxhack, 2021-03-17:
It probably looks like a libstdc++ inconsistency.

Do the node-related modules you installed match the libstdc++ you linked in when you build it?

ANARHIST1984, 2021-03-17:
> It probably looks like a libstdc++ inconsistency.
> 
> Do the node-related modules you installed match the libstdc++ you linked in when you build it?

libstdc++ was instaled!
cd /opt/zigbee2mqtt/node_modules/zigbee-herdsman/node_modules/serialport
node test.js not stat
after i decide install 19.07.05

I compile 19.07.05 
Install all packages and now its works)
tomorrow will try install again all in to 19.07.07 Make clear install

nxhack, 2021-05-05:
Close the issue. If the problem still exists, feel free to reopen it.

closed 2021-05-05 by nxhack