nxhack/openwrt-node-packages #862: Build failed node-red > node-bufferutil

G1K, 2021-03-29:
Maybe you can help me

https://github.com/G1K/openwrt-fastbuild-actions/runs/2219336225?check_suite_focus=true#step:19:1236

nxhack, 2021-03-30:
Hi @G1K 

> 2021-03-29T16:43:41.0495940Z Applying ./patches/001-use_prebuild-install.patch using plaintext:
2021-03-29T16:43:41.0511128Z patching file package.json
2021-03-29T16:43:41.0511664Z Hunk #1 FAILED at 4.
2021-03-29T16:43:41.0512110Z Hunk #2 FAILED at 22.
2021-03-29T16:43:41.0514497Z 2 out of 2 hunks FAILED -- saving rejects to file package.json.rej

Check node repo (openwrt-19.07)
https://github.com/nxhack/openwrt-node-packages/tree/openwrt-19.07/node-bufferutil

This is not present in the repository.
```
. /patches/001-use_prebuild-install.patch 
```

To fix this, delete . /feeds/node*, and then try to build it.

G1K, 2021-03-30:
Thanks. I also came to this conclusion. Very very strange, it was not deleted and conflicted. Now the issue is resolved. But there is another one. Removing node from the packages feed, do I have to do it, or will the latest one from your feed still be used ?

closed 2021-03-30 by G1K

nxhack, 2021-03-31:
". /scripts/feeds" script. This script does a git pull to synchronize.

I don't know why the old files were still there, but what often happens is that when you make a modification (adding a file) locally, it stays there.