nxhack/openwrt-node-packages #868: serialport not work on MT7688 (19.07)

arno608rw, 2021-04-02:
/usr/lib/node/@serialport/bindings/node_modules/bindings/bindings.js:121
        throw e;
        ^

Error: Error relocating /usr/lib/node/@serialport/bindings/build/Release/bindings.node: _ZN2v816FunctionTemplate3NewEPNS_7IsolateEPFvRKNS_20FunctionCallbackInfoINS_5ValueEEEENS_5LocalIS4_EENSA_INS_9SignatureEEEiNS_19ConstructorBehaviorENS_14SideEffectTypeE: symbol not found
    at Object.Module._extensions..node (internal/modules/cjs/loader.js:1122:18)
    at Module.load (internal/modules/cjs/loader.js:928:32)
    at Function.Module._load (internal/modules/cjs/loader.js:769:14)
    at Module.require (internal/modules/cjs/loader.js:952:19)
    at require (internal/modules/cjs/helpers.js:88:18)
    at bindings (/usr/lib/node/@serialport/bindings/node_modules/bindings/bindings.js:112:48)
    at Object.<anonymous> (/usr/lib/node/@serialport/bindings/lib/linux.js:2:36)
    at Module._compile (internal/modules/cjs/loader.js:1063:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:1092:10)
    at Module.load (internal/modules/cjs/loader.js:928:32)

nxhack, 2021-04-02:
Hi @arno608rw 

>Error: Error relocating /usr/lib/node/@serialport/bindings/build/Release/bindings.node: _ZN2v816FunctionTemplate3NewEPNS_7IsolateEPFvRKNS_20FunctionCallbackInfoINS_5ValueEEEENS_5LocalIS4_EENSA_INS_9SignatureEEEiNS_19ConstructorBehaviorENS_14SideEffectTypeE: symbol not found

It probably looks like a libstdc++ inconsistency.

Do the node-related modules you installed match the libstdc++ you linked in when you build it?

arno608rw, 2021-04-02:
I compile complete but it not work on nodejs application.

const SerialPort = require('serialport')

nxhack, 2021-04-02:
@arno608rw 

I recommend that you do a clean build and rebuild the firmware to fix the inconsistency.

arno608rw, 2021-04-02:
@nxhack Thank you
It work. I make clean and remake firmware

closed 2021-04-02 by arno608rw