nxhack/openwrt-node-packages #873: Compile error mraa and node V12

arno608rw, 2021-04-18:
V8_WARN_UNUSED_RESULT MaybeLocal<Function> GetFunction(
                                              ^~~~~~~~~~~
/home/user/openwrt-node-19.07/staging_dir/target-mipsel_24kc_musl/usr/include/node/v8.h:6126:46: note:   candidate expects 1 argument, 0 provided
/home/user/openwrt-node-19.07/build_dir/target-mipsel_24kc_musl/mraa-2.0.0/src/javascript/CMakeFiles/mraajs.dir/mraajsJAVASCRIPT_wrap.cxx:12406:77: error: no matching function for call to 'v8::FunctionTemplate::GetFunction()'
 v8::Handle<v8::Object> _exports_Aio_obj = _exports_Aio_class_0->GetFunction();
                                                                             ^
In file included from /home/user/openwrt-node-19.07/staging_dir/target-mipsel_24kc_musl/usr/include/node/node.h:67:0,
                 from /home/user/openwrt-node-19.07/build_dir/target-mipsel_24kc_musl/mraa-2.0.0/src/javascript/CMakeFiles/mraajs.dir/mraajsJAVASCRIPT_wrap.cxx:171:
/home/user/openwrt-node-19.07/staging_dir/target-mipsel_24kc_musl/usr/include/node/v8.h:6126:46: note: candidate: v8::MaybeLocal<v8::Function> v8::FunctionTemplate::GetFunction(v8::Local<v8::Context>)
   V8_WARN_UNUSED_RESULT MaybeLocal<Function> GetFunction(
                                              ^~~~~~~~~~~
/home/user/openwrt-node-19.07/staging_dir/target-mipsel_24kc_musl/usr/include/node/v8.h:6126:46: note:   candidate expects 1 argument, 0 provided
/home/user/openwrt-node-19.07/build_dir/target-mipsel_24kc_musl/mraa-2.0.0/src/javascript/CMakeFiles/mraajs.dir/mraajsJAVASCRIPT_wrap.cxx:12412:79: error: no matching function for call to 'v8::FunctionTemplate::GetFunction()'
 v8::Handle<v8::Object> _exports_Uart_obj = _exports_Uart_class_0->GetFunction();
                                                                               ^
In file included from /home/user/openwrt-node-19.07/staging_dir/target-mipsel_24kc_musl/usr/include/node/node.h:67:0,
                 from /home/user/openwrt-node-19.07/build_dir/target-mipsel_24kc_musl/mraa-2.0.0/src/javascript/CMakeFiles/mraajs.dir/mraajsJAVASCRIPT_wrap.cxx:171:
/home/user/openwrt-node-19.07/staging_dir/target-mipsel_24kc_musl/usr/include/node/v8.h:6126:46: note: candidate: v8::MaybeLocal<v8::Function> v8::FunctionTemplate::GetFunction(v8::Local<v8::Context>)
   V8_WARN_UNUSED_RESULT MaybeLocal<Function> GetFunction(
                                              ^~~~~~~~~~~
/home/user/openwrt-node-19.07/staging_dir/target-mipsel_24kc_musl/usr/include/node/v8.h:6126:46: note:   candidate expects 1 argument, 0 provided
/home/user/openwrt-node-19.07/build_dir/target-mipsel_24kc_musl/mraa-2.0.0/src/javascript/CMakeFiles/mraajs.dir/mraajsJAVASCRIPT_wrap.cxx:12418:77: error: no matching function for call to 'v8::FunctionTemplate::GetFunction()'
 v8::Handle<v8::Object> _exports_Led_obj = _exports_Led_class_0->GetFunction();
                                                                             ^
In file included from /home/user/openwrt-node-19.07/staging_dir/target-mipsel_24kc_musl/usr/include/node/node.h:67:0,
                 from /home/user/openwrt-node-19.07/build_dir/target-mipsel_24kc_musl/mraa-2.0.0/src/javascript/CMakeFiles/mraajs.dir/mraajsJAVASCRIPT_wrap.cxx:171:
/home/user/openwrt-node-19.07/staging_dir/target-mipsel_24kc_musl/usr/include/node/v8.h:6126:46: note: candidate: v8::MaybeLocal<v8::Function> v8::FunctionTemplate::GetFunction(v8::Local<v8::Context>)
   V8_WARN_UNUSED_RESULT MaybeLocal<Function> GetFunction(
                                              ^~~~~~~~~~~
/home/user/openwrt-node-19.07/staging_dir/target-mipsel_24kc_musl/usr/include/node/v8.h:6126:46: note:   candidate expects 1 argument, 0 provided
make[6]: *** [src/javascript/CMakeFiles/mraajs.dir/build.make:63: src/javascript/CMakeFiles/mraajs.dir/CMakeFiles/mraajs.dir/mraajsJAVASCRIPT_wrap.cxx.o] Error 1
make[6]: Leaving directory '/home/user/openwrt-node-19.07/build_dir/target-mipsel_24kc_musl/mraa-2.0.0'
make[5]: *** [CMakeFiles/Makefile2:480: src/javascript/CMakeFiles/mraajs.dir/all] Error 2
make[5]: Leaving directory '/home/user/openwrt-node-19.07/build_dir/target-mipsel_24kc_musl/mraa-2.0.0'
make[4]: *** [Makefile:141: all] Error 2
make[4]: Leaving directory '/home/user/openwrt-node-19.07/build_dir/target-mipsel_24kc_musl/mraa-2.0.0'
make[3]: *** [Makefile:128: /home/user/openwrt-node-19.07/build_dir/target-mipsel_24kc_musl/mraa-2.0.0/.built] Error 2
make[3]: Leaving directory '/home/user/openwrt-node-19.07/feeds/packages/libs/libmraa'
time: package/feeds/packages/libmraa/compile#15.08#2.40#18.28
make[2]: *** [package/Makefile:113: package/feeds/packages/libmraa/compile] Error 2
make[2]: Leaving directory '/home/user/openwrt-node-19.07'
make[1]: *** [package/Makefile:107: /home/user/openwrt-node-19.07/staging_dir/target-mipsel_24kc_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/user/openwrt-node-19.07'
make: *** [/home/user/openwrt-node-19.07/include/toplevel.mk:227: world] Error 2

nxhack, 2021-04-18:
Hi @arno608rw 

The mraa and upm packages in OpenWrt-19.07 do not support the latest version of node.js. There are no plans to do so.

If you want to use 19.07, please backport the 21.02 packages yourself.

nxhack, 2021-04-19:
& swig

swig & libmraa & libupm

arno608rw, 2021-04-19:
Thank @nxhack 
I change feeds.conf and recompile. It work!!!!. I testing on device
#src-git packages https://git.openwrt.org/feed/packages.git;openwrt-19.07
src-git packages https://git.openwrt.org/feed/packages.git;openwrt-21.02
src-git luci https://git.openwrt.org/project/luci.git;openwrt-19.07
src-git routing https://git.openwrt.org/feed/routing.git;openwrt-19.07
src-git telephony https://git.openwrt.org/feed/telephony.git;openwrt-19.07
src-git freifunk https://github.com/freifunk/openwrt-packages.git;openwrt-19.07
src-git node https://github.com/nxhack/openwrt-node-packages.git;openwrt-19.07

closed 2021-04-19 by arno608rw