nxhack/openwrt-node-packages #882: v21.02.0-rc1

spot320, 2021-05-04:
add instructions for building 21.02.0-rc1, but I can't find what to add to feeds.conf.default. from 19.07 does not work.

nxhack, 2021-05-04:
I've already created the openwrt-21.02 branch.

```
src-git node https://github.com/nxhack/openwrt-node-packages.git;openwrt-21.02
```

spot320, 2021-05-04:
> I've already created the openwrt-21.02 branch.
> 
> ```
> src-git node https://github.com/nxhack/openwrt-node-packages.git;openwrt-21.02
> ```

>  
found a branch, thanks. I do everything according to the instructions, I get an error 
./scripts/feeds install -a -p node
Collecting package info: done
WARNING: Makefile 'package/feeds/packages/adguardhome/Makefile' has a build dependency on 'node/host', which does not exist
WARNING: Makefile 'package/feeds/packages/libmraa/Makefile' has a build dependency on 'node/host', which does not exist
WARNING: Makefile 'package/feeds/packages/libmraa/Makefile' has a build dependency on 'node', which does not exist
WARNING: Makefile 'package/feeds/node/node-actions-on-google/Makefile' has a dependency on 'node', which does not exist
WARNING: Makefile 'package/feeds/node/node-actions-on-google/Makefile' has a build dependency on 'node/host', which does not exist
WARNING: Makefile 'package/feeds/node/node-alexa-app/Makefile' has a dependency on 'node', which does not exist
WARNING: Makefile 'package/feeds/node/node-alexa-app/Makefile' has a build dependency on 'node/host', which does not exist
WARNING: Makefile 'package/feeds/node/node-ask-sdk/Makefile' has a dependency on 'node', which does not exist
WARNING: Makefile 'package/feeds/node/node-ask-sdk/Makefile' has a build dependency on 'node/host', which does not exist
WARNING: Makefile 'package/feeds/node/node-authenticate-pam/Makefile' has a dependency on 'node', which does not exist
WARNING: Makefile 'package/feeds/node/node-authenticate-pam/Makefile' has a build dependency on 'node/host', which does not exist
WARNING: Makefile 'package/feeds/node/node-autobahn/Makefile' has a dependency on 'node', which does not exist
WARNING: Makefile 'package/feeds/node/node-autobahn/Makefile' has a build dependency on 'node/host', which does not exist
WARNING: Makefile 'package/feeds/node/node-aws-crt/Makefile' has a dependency on 'node', which does not exist
WARNING: Makefile 'package/feeds/node/node-aws-crt/Makefile' has a build dependency on 'node/host', which does not exist
WARNING: Makefile 'package/feeds/node/node-aws-iot-device-sdk/Makefile' has a dependency on 'node', which does not exist
WARNING: Makefile 'package/feeds/node/node-aws-iot-device-sdk/Makefile' has a build dependency on 'node/host', which does not exist
WARNING: Makefile 'package/feeds/node/node-aws-iot-device-sdk-v2/Makefile' has a dependency on 'node', which does not exist
WARNING: Makefile 'package/feeds/node/node-aws-iot-device-sdk-v2/Makefile' has a build dependency on 'node/host', which does not exist
WARNING: Makefile 'package/feeds/node/node-aws-sdk/Makefile' has a dependency on 'node', which does not exist
WARNING: Makefile 'package/feeds/node/node-aws-sdk/Makefile' has a build dependency on 'node/host', which does not exist
WARNING: Makefile 'package/feeds/node/node-azure-event-hubs/Makefile' has a dependency on 'node', which does not exist
WARNING: Makefile 'package/feeds/node/node-azure-event-hubs/Makefile' has a build dependency on 'node/host', which does not exist
WARNING: Makefile 'package/feeds/node/node-azure-iot-device/Makefile' has a dependency on 'node', which does not exist
WARNING: Makefile 'package/feeds/node/node-azure-iot-device/Makefile' has a build dependency on 'node/host', which does not exist
WARNING: Makefile 'package/feeds/node/node-azure-iot-device-amqp/Makefile' has a dependency on 'node', which does not exist
WARNING: Makefile 'package/feeds/node/node-azure-iot-device-amqp/Makefile' has a build dependency on 'node/host', which does not exist
WARNING: Makefile 'package/feeds/node/node-azure-iot-device-http/Makefile' has a dependency on 'node', which does not exist
WARNING: Makefile 'package/feeds/node/node-azure-iot-device-http/Makefile' has a build dependency on 'node/host', which does not exist
WARNING: Makefile 'package/feeds/node/node-azure-iot-device-mqtt/Makefile' has a dependency on 'node', which does not exist
WARNING: Makefile 'package/feeds/node/node-azure-iot-device-mqtt/Makefile' has a build dependency on 'node/host', which does not exist
WARNING: Makefile 'package/feeds/node/node-azure-iothub/Makefile' has a dependency on 'node', which does not exist
WARNING: Makefile 'package/feeds/node/node-azure-iothub/Makefile' has a build dependency on 'node/host', which does not exist
WARNING: Makefile 'package/feeds/node/node-bcrypt/Makefile' has a dependency on 'node', which does not exist
WARNING: Makefile 'package/feeds/node/node-bcrypt/Makefile' has a build dependency on 'node/host', which does not exist
WARNING: Makefile 'package/feeds/node/node-bignum/Makefile' has a dependency on 'node', which does not exist
WARNING: Makefile 'package/feeds/node/node-bignum/Makefile' has a build dependency on 'node/host', which does not exist
WARNING: Makefile 'package/feeds/node/node-binaryjs/Makefile' has a dependency on 'node', which does not exist
WARNING: Makefile 'package/feeds/node/node-binaryjs/Makefile' has a build dependency on 'node/host', which does not exist
WARNING: Makefile 'package/feeds/node/node-bleacon/Makefile' has a dependency on 'node', which does not exist
WARNING: Makefile 'package/feeds/node/node-bleacon/Makefile' has a build dependency on 'node/host', which does not exist
WARNING: Makefile 'package/feeds/node/node-bleno/Makefile' has a dependency on 'node', which does not exist
WARNING: Makefile 'package/feeds/node/node-bleno/Makefile' has a build dependency on 'node/host', which does not exist
[... cut ...]
WARNING: Makefile 'package/feeds/node/node-zigbee2mqtt/Makefile' has a build dependency on 'node/host', which does not exist
Installing all packages from feed node.
Installing package 'node' from node
Installing package 'node-arduino-firmata' from node
Installing package 'node-serialport' from node
Installing package 'node-serialport-bindings' from node
Installing package 'node-cylon' from node
Installing package 'node-hid' from node
Installing package 'node-homebridge' from node
Installing package 'node-javascript-obfuscator' from node
Installing package 'node-yarn' from node

nxhack, 2021-05-04:
feeds.conf for OpenWrt v21.02.0-rc1 branch
```
src-git packages https://git.openwrt.org/feed/packages.git^4ceeb8fc90ed2c2e650ddddc855e7ed1df071c22
src-git luci https://git.openwrt.org/project/luci.git^7d913b997601d533cca187cfc1b3057c3c98effc
src-git routing https://git.openwrt.org/feed/routing.git^5b4d4c7fb6a97cac68c7d8b156fd0ab27bab4dcc
src-git telephony https://git.openwrt.org/feed/telephony.git^178822957123b821407e1216e9e7314161512ac6
src-git node https://github.com/nxhack/openwrt-node-packages.git;openwrt-21.02
```

and run
```
./scripts/feeds update -a
./scripts/feeds install -a -f
rm ./package/feeds/packages/node
rm ./package/feeds/packages/node-*
./scripts/feeds install -a -p node
make defconfig
```

spot320, 2021-05-05:
I do everything as described, but there are errors in the process and in the end there are no points

     Global build settings> Kernel build options
     [*] Compile the kernel with MIPS FPU Emulator

     Languages> Node.js
     -M- node
     Languages> Node.js> node> Configuration ---> Version Selection
     10.x Maintenance LTS / 12.x Maintenance LTS / 14.x Active LTS
     Languages> Node.js
     <M> node-npm 

.

/scripts/feeds update -a
Updating feed 'packages' from 'https://git.openwrt.org/feed/packages.git^4ceeb8fc90ed2c2e650ddddc855e7ed1df071c22' ...
Клонирование в «./feeds/packages»…
remote: Enumerating objects: 142729, done.
remote: Counting objects: 100% (142729/142729), done.
remote: Compressing objects: 100% (58609/58609), done.
remote: Total 142729 (delta 81774), reused 136932 (delta 76732)
Получение объектов: 100% (142729/142729), 32.43 MiB | 232.00 KiB/s, готово.
Определение изменений: 100% (81774/81774), готово.
Переключено на новую ветку «4ceeb8fc90ed2c2e650ddddc855e7ed1df071c22»
/home/spot320/openwrt
Create index file './feeds/packages.index' 
Checking 'working-make'... ok.
Checking 'case-sensitive-fs'... ok.
Checking 'proper-umask'... ok.
Checking 'gcc'... ok.
Checking 'working-gcc'... ok.
Checking 'g++'... ok.
Checking 'working-g++'... ok.
Checking 'ncurses'... ok.
Checking 'perl-data-dumper'... ok.
Checking 'perl-thread-queue'... ok.
Checking 'tar'... ok.
Checking 'find'... ok.
Checking 'bash'... ok.
Checking 'xargs'... ok.
Checking 'patch'... ok.
Checking 'diff'... ok.
Checking 'cp'... ok.
Checking 'seq'... ok.
Checking 'awk'... ok.
Checking 'grep'... ok.
Checking 'egrep'... ok.
Checking 'getopt'... ok.
Checking 'stat'... ok.
Checking 'unzip'... ok.
Checking 'bzip2'... ok.
Checking 'wget'... ok.
Checking 'perl'... ok.
Checking 'python2-cleanup'... ok.
Checking 'python'... ok.
Checking 'python3'... ok.
Checking 'git'... ok.
Checking 'file'... ok.
Checking 'rsync'... ok.
Checking 'which'... ok.
Checking 'ldconfig-stub'... ok.
Collecting package info: feeds/packages/lang/python/python3-django-etesync-journ
Collecting package info: done
Collecting target info: done
Updating feed 'luci' from 'https://git.openwrt.org/project/luci.git^7d913b997601d533cca187cfc1b3057c3c98effc' ...
Клонирование в «./feeds/luci»…
remote: Enumerating objects: 260919, done.
remote: Counting objects: 100% (260919/260919), done.
remote: Compressing objects: 100% (53837/53837), done.
remote: Total 260919 (delta 143393), reused 256960 (delta 140608)
[... 1855 lines cut ...]
WARNING: Makefile 'package/feeds/node/node-wot-binding-mqtt/Makefile' has a build dependency on 'node/host', which does not exist
WARNING: Makefile 'package/feeds/node/node-wot-binding-opcua/Makefile' has a dependency on 'node', which does not exist
WARNING: Makefile 'package/feeds/node/node-wot-binding-opcua/Makefile' has a build dependency on 'node/host', which does not exist
WARNING: Makefile 'package/feeds/node/node-wot-binding-websockets/Makefile' has a dependency on 'node', which does not exist
WARNING: Makefile 'package/feeds/node/node-wot-binding-websockets/Makefile' has a build dependency on 'node/host', which does not exist
WARNING: Makefile 'package/feeds/node/node-wot-cli/Makefile' has a dependency on 'node', which does not exist
WARNING: Makefile 'package/feeds/node/node-wot-cli/Makefile' has a build dependency on 'node/host', which does not exist
WARNING: Makefile 'package/feeds/node/node-wot-core/Makefile' has a dependency on 'node', which does not exist
WARNING: Makefile 'package/feeds/node/node-wot-core/Makefile' has a build dependency on 'node/host', which does not exist
WARNING: Makefile 'package/feeds/node/node-wot-td-tools/Makefile' has a dependency on 'node', which does not exist
WARNING: Makefile 'package/feeds/node/node-wot-td-tools/Makefile' has a build dependency on 'node/host', which does not exist
WARNING: Makefile 'package/feeds/node/node-ws/Makefile' has a dependency on 'node', which does not exist
WARNING: Makefile 'package/feeds/node/node-ws/Makefile' has a build dependency on 'node/host', which does not exist
WARNING: Makefile 'package/feeds/node/node-zigbee2mqtt/Makefile' has a dependency on 'node', which does not exist
WARNING: Makefile 'package/feeds/node/node-zigbee2mqtt/Makefile' has a build dependency on 'node/host', which does not exist
Installing all packages from feed node.
Installing package 'node' from node
Installing package 'node-arduino-firmata' from node
Installing package 'node-serialport' from node
Installing package 'node-serialport-bindings' from node
Installing package 'node-cylon' from node
Installing package 'node-hid' from node
Installing package 'node-homebridge' from node
Installing package 'node-javascript-obfuscator' from node
Installing package 'node-yarn' from node

nxhack, 2021-05-05:
Ah! You're not using 21.02.

As it is concluded, the Linux kernel of OpenWrt-21.02 does not support the MIPS FPU Emulator.

If you plan to use mips as a target, node.js will not work.

If you are not targeting mips, please update your source code in the following way

```
git pull                                                                                                      
git reset --hard 2ce89a35788fcd768a4a872d3d4f5a3a131b1d43                                                     
cp feeds.conf.default feeds.conf
echo 'src-git node https://github.com/nxhack/openwrt-node-packages.git;openwrt-21.02' >> feeds.conf
./scripts/feeds update -a
./scripts/feeds install -a -f
rm ./package/feeds/packages/node
rm ./package/feeds/packages/node-*
./scripts/feeds install -a -p node
make defconfig
```

spot320, 2021-05-05:
> Ah! You're not using 21.02.
>
> As it is concluded, the Linux kernel of OpenWrt-21.02 does not support the MIPS FPU Emulator.
>
> If you plan to use mips as a target, node.js will not work.
>
> If you are [...]

thanks for the help, but it does not work, there are simply no points in the builder. Everything worked for me newifi2 ramips m7621 router in version 19.07. Did I understand correctly that 21.02 does not support the MIPS FPU Emulator core?

nxhack, 2021-05-05:
Your target machine: newifi2  (https://openwrt.org/toh/hwdata/d-team/d-team_newifi_d2)
Your target CPU: ramips (m7621)
Your target CPU ARCH: mipsel_24kc  (***No hardware FPU***)

The version of OpenWrt you want to install: v21.02.0-rc1
Linux kernel version for v21.02.0-rc1: 5.4 (https://github.com/openwrt/openwrt/blob/2ce89a35788fcd768a4a872d3d4f5a3a131b1d43/target/linux/ramips/Makefile#L13)

 kernel: Use new symbol to deactivate MIPS FPU support: (https://github.com/openwrt/openwrt/commit/69d179ec81dacf92e1f693783c4ddcbf5a68b231)

Node.js support platform list: https://github.com/nodejs/node/blob/master/BUILDING.md#platform-list

Discussion in Node.js: https://github.com/nodejs/node/issues/26179

Node.js is not certified to run MIPS.
In V8 development status, mips64el is maintained, but mipsel is not considered.

spot320, 2021-05-06:
> Your target machine: newifi2 (https://openwrt.org/toh/hwdata/d-team/d-team_newifi_d2)
> Your target CPU: ramips (m7621)
> Your target CPU ARCH: mipsel_24kc (_**No hardware FPU**_)
> 
> The version of OpenWrt you want to install: v21.02.0-rc1
> Linux kernel version for v21.02.0-rc1: 5.4 (https://github.com/openwrt/openwrt/blob/2ce89a35788fcd768a4a872d3d4f5a3a131b1d43/target/linux/ramips/Makefile#L13)
> 
> kernel: Use new symbol to deactivate MIPS FPU support: ([openwrt/openwrt@69d179e](https://github.com/openwrt/openwrt/commit/69d179ec81dacf92e1f693783c4ddcbf5a68b231))
> 
> Node.js support platform list: https://github.com/nodejs/node/blob/master/BUILDING.md#platform-list
> 
> Discussion in Node.js: [nodejs/node#26179](https://github.com/nodejs/node/issues/26179)
> 
> Node.js is not certified to run MIPS.
> In V8 development status, mips64el is maintained, but mipsel is not considered.

I do not understand English well, and Google translator does not allow me to understand the situation as you describe it.

When built on 07/19/7, everything worked like this:
git clone git: //github.com/openwrt/openwrt.git -b v19.07.7 && cd ./openwrt

Add the following line to it:
src-git node https://github.com/nxhack/openwrt-node-packages.git;openwrt-19.07

WE UPDATE SOURCES AND INSTALL PACKAGES FROM THEM:

./scripts/feeds update -a
./scripts/feeds install -a

WE DO THE SAME BUT SEPARATELY FOR Node.js:

./scripts/feeds update node
rm ./package/feeds/packages/node
rm ./package/feeds/packages/node-*
./scripts/feeds install -a -p node

DOWNLOAD THE OFFICIAL ASSEMBLY CONFIGURATION FILE:

wget -O ./.config https://downloads.openwrt.org/releases/19.07.7/targets/ramips/mt7621/config.buildinfo

RUNNING THE CONFIGURATION UTILITY:

make menuconfig

and puts appeared and you can select them:

Global build settings> Kernel build options
[*] Compile the kernel with MIPS FPU Emulator
Languages> Node.js
-M- node
Languages> Node.js> node> Configuration ---> Version Selection
10.x Maintenance LTS / 12.x Maintenance LTS / 14.x Active LTS
Languages> Node.js
<M> node-npm

_______ Now I'm doing this and it doesn't work:
git clone git: //github.com/openwrt/openwrt.git -b v21.02.0-rc1 && cd ./openwrt

OPENING THE FILE WITH THE LIST OF SOURCES:

nano ./feeds.conf.default
And add the following line to it:
src-git node https://github.com/nxhack/openwrt-node-packages.git;openwrt-21.02

WE UPDATE SOURCES AND INSTALL PACKAGES FROM THEM:

./scripts/feeds update -a
./scripts/feeds install -a

WE DO THE SAME BUT SEPARATELY FOR Node.js:

./scripts/feeds update node
rm ./package/feeds/packages/node
rm ./package/feeds/packages/node-*
./scripts/feeds install -a -p node

DOWNLOAD THE OFFICIAL ASSEMBLY CONFIGURATION FILE:

wget -O ./.config https://downloads.openwrt.org/releases/21.02.0-rc1/targets/ramips/mt7621/config.buildinfo

 RUNNING THE CONFIGURATION UTILITY:

make menuconfig

After there are no parameters:

Global build settings> Kernel build options
[*] Compile the kernel with MIPS FPU Emulator
Languages> Node.js
-M- node
Languages> Node.js> node> Configuration ---> Version Selection
10.x Maintenance LTS / 12.x Maintenance LTS / 14.x Active LTS
Languages> Node.js
<M> node-npm

Can you more simply explain what my mistake is?

nxhack, 2021-05-06:
Openwrt has fine documentation.

https://openwrt.org/docs/guide-developer/build-system/use-buildsystem

If you want to use v21.02.0-rc1 with newifi2, you can build the firmware as follows Of course, you can't use node.js. You can't use node.js, or even select the build target.

```
mkdir SOME_WHERE
cd SOME_WHERE
git clone https://git.openwrt.org/openwrt/openwrt.git openwrt
cd openwrt
git pull
scripts/feeds update -a
scripts/feeds install -a
git checkout v21.02.0-rc1 
cp feeds.conf.default feeds.conf
echo 'src-git node https://github.com/nxhack/openwrt-node-packages.git;openwrt-21.02' >> feeds.conf
./scripts/feeds update -a
./scripts/feeds install -a -f
rm ./package/feeds/packages/node
rm ./package/feeds/packages/node-*
./scripts/feeds install -a -p node
make menuconfig
```
Select
```
Target System (MediaTek Ralink MIPS)  --->
Subtarget (MT7621 based boards)  --->
Target Profile (Newifi D2
```
***Of course, you can't use node.js. You can't use node.js, or even select the build target.***

nxhack, 2021-05-06:
If you select ramips in 21.02, you will not see this display you commented on.

>Global build settings> Kernel build options
[*] Compile the kernel with MIPS FPU Emulator
Languages> Node.js
-M- node
Languages> Node.js> node> Configuration ---> Version Selection
10.x Maintenance LTS / 12.x Maintenance LTS / 14.x Active LTS
Languages> Node.js
node-npm

nxhack, 2021-05-06:
>Can you more simply explain what my mistake is?

The reason is what I commented on and this code.

https://github.com/nxhack/openwrt-node-packages/issues/882#issuecomment-832394185

https://github.com/nxhack/openwrt-node-packages/blob/openwrt-21.02/node/Makefile#L54

nxhack, 2021-05-06:
To put it simply, if you want to run node.js on OpenWrt ramips, you have to use 19.07.

spot320, 2021-05-06:
> To put it simply, if you want to run node.js on OpenWrt ramips, you have to use 19.07.

thanks for the detailed answer, now it is clear that openwrt 21.02.0-rc1 is not suitable for using mips fpu emulator

closed 2021-05-06 by nxhack

nxhack, 2021-05-07:
@spot320 
К сожалению, OpenWrt-21.02 и более поздние версии не поддерживают программное обеспечение MIPS FPU. Основной движок node.js, V8, требует FPU. Поэтому node.js не поддерживается в OpenWrt-21.02 и более поздних версиях. Кроме того, проект разработки node.js официально не поддерживает MIPS. В настоящее время минимальным окружением для запуска node.js является ядро arm с 32-битным аппаратным FPU. Из того, что я видел, этого достаточно для запуска node.js; использование node.js больше невозможно на MIPS.

[DeepL](https://www.deepl.com/translator) - это отличная программа для перевода. Я тоже им пользуюсь.

spot320, 2021-05-09:
Спасибо вам большое. Я до этого читал что из-за отсутствия подержки аппаратного fpu на mips был вами разработан эмулятор, именно по этому  я и пользовался им на 19.07 сейчас хотел обновить прошивку до 21.02 но понял что нет смысла.  Сейчас я пытаюсь найти вариант установки node.js на 21.02, изучаю форумы, нашел https://github.com/patrikx3/openwrt-insomnia  но с моим уровнем знаний это тяжело, пытаюсь разобраться что к чему. И спасибо за переводчик, действительно годный.

andreasrsa, 2021-12-17:
da