nxhack/openwrt-node-packages #912: Cannot compile on kali-linux 64 on RaspberryPi

CyanTaro, 2021-05-20:
**The error is as follows：**
```
g++: error: unrecognized command-line option '-m32'
make[5]: *** [tools/icu/icutools.host.mk:568: /home/kali/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/node-v14.17.0/out/Release/obj.host/icutools/deps/icu-small/source/tools/toolutil/pkgitems.o] Error 1
make[4]: *** [Makefile:104: node] Error 2
make[4]: Leaving directory '/home/kali/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/node-v14.17.0'
make[3]: *** [Makefile:197: /home/kali/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/node-v14.17.0/.built] Error 2
make[3]: Leaving directory '/home/kali/lede/feeds/node/node'
time: package/feeds/node/node/compile#335.10#53.47#378.99
make[2]: *** [package/Makefile:113: package/feeds/node/node/compile] Error 2
make[2]: Leaving directory '/home/kali/lede'
make[1]: *** [package/Makefile:107: /home/kali/lede/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/kali/lede'
make: *** [/home/kali/lede/include/toplevel.mk:224: world] Error 2
```
**And after editting Makefile of node (delete -m32), the error is as follows：**
```
../src/api/environment.cc: In function 'void node::DefaultProcessExitHandler(node::Environment*, int)':
../src/api/environment.cc:712:3: error: 'uv_library_shutdown' was not declared in this scope
   uv_library_shutdown();
   ^~~~~~~~~~~~~~~~~~~
../src/api/environment.cc:712:3: note: suggested alternative: 'uv_shutdown'
   uv_library_shutdown();
   ^~~~~~~~~~~~~~~~~~~
   uv_shutdown
make[5]: *** [libnode.target.mk:322: /home/kali/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/node-v14.17.0/out/Release/obj.target/libnode/src/api/environment.o] Error 1
rm 6a9f142617387aa8fd26f66844a83de323ff660b.intermediate
make[4]: *** [Makefile:104: node] Error 2
make[4]: Leaving directory '/home/kali/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/node-v14.17.0'
make[3]: *** [Makefile:197: /home/kali/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/node-v14.17.0/.built] Error 2
make[3]: Leaving directory '/home/kali/lede/feeds/node/node'
time: package/feeds/node/node/compile#1229.55#145.98#1365.62
make[2]: *** [package/Makefile:113: package/feeds/node/node/compile] Error 2
make[2]: Leaving directory '/home/kali/lede'
make[1]: *** [package/Makefile:107: /home/kali/lede/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/kali/lede'
make: *** [/home/kali/lede/include/toplevel.mk:224: world] Error 2
```

nxhack, 2021-05-20:
Hi @CyanTaro 

(1)
> g++: error: unrecognized command-line option '-m32'

Please read my README.md, I guess multilib is not installed.

kali-linux:
```
apt-get install gcc-multilib g++-multilib
```

(2)
> ../src/api/environment.cc:712:3: error: 'uv_library_shutdown' was not declared in this scope
   uv_library_shutdown();

You have an old version of libuv.
What version of OpenWrt do you want to use?

If it's 19.07, use the matching branch.
https://github.com/nxhack/openwrt-node-packages/tree/openwrt-19.07#usage

CyanTaro, 2021-05-21:
> Hi @CyanTaro
> 
> (1)
> 
> > g++: error: unrecognized command-line option '-m32'
> 
> Please read my README.md, I guess multilib is not installed.
> 
> kali-linux:
> 
> ```
> sudo apt-get install gcc-multilib g++-multilib
> ```
> 
> (2)
> 
> > ../src/api/environment.cc:712:3: error: 'uv_library_shutdown' was not declared in this scope
> > uv_library_shutdown();
> 
> You have an old version of libuv.
> What version of OpenWrt do you want to use?
> 
> If it's 19.07, use the matching branch.
> https://github.com/nxhack/openwrt-node-packages/tree/openwrt-19.07#usage

Thanks fot reply! 

(1) Sorry, I didn't describe my problem well. The raspberryPi 4B is the Host, my goal is compiling in arm64 mechine, this environment don't support multilib.

(2)The version of Openwrt I'm using is R21.4.18 by Lean, a third party branch, thank you for giving directions, I will try other branchs.

nxhack, 2021-05-21:
First of all, I can't support anything other than the official branch of OpenWrt.

> (1) Sorry, I didn't describe my problem well. The raspberryPi 4B is the Host, my goal is compiling in arm64 mechine, this environment don't support multilib.

The OpenWrt toolchain should be built on x86_64.
Please do not use raspberryPi (aarch64) as a build environment for OpenWrt.

It is recommended that you read the official documentation before building OpenWrt.

https://openwrt.org/docs/guide-developer/build-system/start

CyanTaro, 2021-05-21:
> First of all, I can't support anything other than the official branch of OpenWrt.
>
>
> The OpenWrt toolchain should be built on x86_64.
> Please do not use raspberryPi (aarch64) as a build environment [...]

Got it, thank you for help. I'll try to fix it myself.

closed 2021-05-21 by CyanTaro