nxhack/openwrt-node-packages #920: Please add node-argon2

kele6ra, 2021-05-24:
Hello, can you add [node-argon2](https://github.com/ranisalt/node-argon2) ?

nxhack, 2021-05-25:
@kele6ra 

It ready. Please check if it is working.

nxhack, 2021-05-26:
You will need to build the package for your target machine by yourself.

Please take a look at the OprnWrt documentations.

https://openwrt.org/docs/guide-user/additional-software/imagebuilder
https://forum.openwrt.org/t/custom-build-package-inside-image-builder-image/40350

kele6ra, 2021-05-26:
I found that pre-build binary is made with the help of the node-gyp during the openWRT building. And I can find it in /usr/lib/node/argon2/lib. But nodeJS tries to install it localy when I make 'npm i' instead of using global version form /usr/lib.

kele6ra, 2021-05-26:
I copy argon2.node binary from lib dir to local node_modules and all is working now. I think it is nodeJS issue, not argon2 library. I had the same problem with sqlite3.

closed 2021-05-26 by kele6ra

nxhack, 2021-05-26:
The node-related packages in this repository are designed to be installed in a "global" installation.

If you are using it in a "local" installation, then you are correct to do as you did.

Incidentally, OpenWrt is designed as a router OS optimized for underpowered embedded MPUs, unlike regular server OSs.

Because of this design philosophy, various services can be run under the so-called "Run As Root" policy.
Or, they can be run with the privileges of a user dedicated to that service.

It is not a problem in the design philosophy of OpenWrt that node modules also use packages for global installation.

kele6ra, 2021-05-27:
I compiled openWRT rootfs with inckuding sqlite3 and argon2, after I tried to use these modules as "global" installed and node can't recognize them as globally installed packages.
They become to work only after merging global and local version.

nxhack, 2021-05-27:
Hmmm... In my environment, I can use modules on "global" without any problem.

If you can figure out what's causing your problem, I'd appreciate it.

kele6ra, 2021-06-17:
> Hmmm... In my environment, I can use modules on "global" without any problem.
> 
> If you can figure out what's causing your problem, I'd appreciate it.

Should I install packet globally after OpenWRT start or it should works out of the box? I try to install packets from package.json and it tries to install packet locally and build sqlite3 lib instead of using it from /usr/lib/node/sqlite3.

nxhack, 2021-06-17:
I want to make sure that you have set your "NODE_PATH". If you do, what is the value?

kele6ra, 2021-06-22:
> I want to make sure that you have set your "NODE_PATH". If you do, what is the value?

Thank you, you are right, I have started application without "NODE_PATH". I thought that it should be set out-of-the-box in the global PATH.