nxhack/openwrt-node-packages #974: Issue in make menuconfig.

miteshdube, 2021-07-27:
I'm trying to build for LinkIt Smart 7688 Duo. I am following all necessary steps provided in this [documentation](https://github.com/nxhack/openwrt-node-packages/wiki). But I'm getting error in make menuconfig, error is as below,

```
$ make menuconfig
Checking 'working-make'... ok.
Checking 'case-sensitive-fs'... ok.
Checking 'gcc'... ok.
Checking 'working-gcc'... ok.
Checking 'g++'... ok.
Checking 'working-g++'... ok.
Checking 'ncurses'... ok.
Checking 'zlib'... ok.
Checking 'libssl'... ok.
Checking 'tar'... ok.
Checking 'find'... ok.
Checking 'bash'... ok.
Checking 'patch'... ok.
Checking 'diff'... ok.
Checking 'cp'... ok.
Checking 'seq'... ok.
Checking 'awk'... ok.
Checking 'grep'... ok.
Checking 'getopt'... ok.
Checking 'stat'... ok.
Checking 'md5sum'... ok.
Checking 'unzip'... ok.
Checking 'bzip2'... ok.
Checking 'wget'... ok.
Checking 'perl'... ok.
Checking 'python'... ok.
Checking 'svn'... ok.
Checking 'git'... failed.
Checking 'file'... ok.
Checking 'openssl'... ok.
Checking 'ldconfig-stub'... ok.

Build dependency: Please install Git (git-core) >= 1.6.5

/home/mitesh-fountlab/fountlab/linkit_smart/build_1/chaos_calmer/include/prereq.mk:12: recipe for target 'prereq' failed
Prerequisite check failed. Use FORCE=1 to override.
/home/mitesh-fountlab/fountlab/linkit_smart/build_1/chaos_calmer/include/toplevel.mk:140: recipe for target 'staging_dir/host/.prereq-build' failed
make: *** [staging_dir/host/.prereq-build] Error 1

```
But I have already installed **git** on host system 

```
$ git --version
git version 2.17.1
```
May I know where am I doing wrong?

nxhack, 2021-07-28:
The version you are trying to build, OpenWrt 15.05, is completely obsolete and does not have full conunity support. It is completely dangerous and you should not be using this version.

If you use it at your own risk, please take care of it yourself.

By the way, I think this will solve the git problem, but I will not support it afterwards.

```diff
--- a/include/prereq-build.mk
+++ b/include/prereq-build.mk
@@ -145,7 +145,7 @@ $(eval $(call SetupHostCommand,svn,Please install the Subversion client, \
        svn --version | grep Subversion))
 
 $(eval $(call SetupHostCommand,git,Please install Git (git-core) >= 1.6.5, \
-       git clone 2>&1 | grep -- --recursive))
+       git --exec-path | xargs -I % -- grep -q -- --recursive %/git-submodule))
 
 $(eval $(call SetupHostCommand,file,Please install the 'file' package, \
        file --version 2>&1 | grep file))
```

miteshdube, 2021-07-28:
Thanks for the response. May I get latest source code for build. My host system is Ubuntu 18.04.

miteshdube, 2021-07-28:
@nxhack Is there any latest source code available for LinkIt Smart 7688 Duo? Because as you mentioned in previous comment that lines doesn't exist in source that I have. My host system is Ubuntu 18.04.5 LTS.

miteshdube, 2021-07-28:
make menuconfig issue has been solved.
Now I am getting below error. Please provide reference to working source code.

```
$ make -j1 V=s
make[1]: Entering directory '/home/mitesh/linkit_smart/build_1/chaos_calmer'
make[2]: Entering directory '/home/mitesh/linkit_smart/build_1/chaos_calmer'
+ mkdir -p /home/mitesh/linkit_smart/build_1/chaos_calmer/staging_dir/target-mipsel_24kec+dsp_uClibc-0.9.33.2
+ cd /home/mitesh/linkit_smart/build_1/chaos_calmer/staging_dir/target-mipsel_24kec+dsp_uClibc-0.9.33.2
+ mkdir -p bin lib include stamp
mkdir -p /home/mitesh/linkit_smart/build_1/chaos_calmer/build_dir/target-mipsel_24kec+dsp_uClibc-0.9.33.2/stamp
touch /home/mitesh/linkit_smart/build_1/chaos_calmer/staging_dir/target-mipsel_24kec+dsp_uClibc-0.9.33.2/.prepared
make[3]: Entering directory '/home/mitesh/linkit_smart/build_1/chaos_calmer/toolchain/gdb'
make[3]: Nothing to be done for 'prepare'.
make[3]: Leaving directory '/home/mitesh/linkit_smart/build_1/chaos_calmer/toolchain/gdb'
make[3]: Entering directory '/home/mitesh/linkit_smart/build_1/chaos_calmer/toolchain/gdb'
make[3]: Nothing to be done for 'compile'.
make[3]: Leaving directory '/home/mitesh/linkit_smart/build_1/chaos_calmer/toolchain/gdb'
make[3]: Entering directory '/home/mitesh/linkit_smart/build_1/chaos_calmer/toolchain/gdb'
make[3]: Nothing to be done for 'install'.
make[3]: Leaving directory '/home/mitesh/linkit_smart/build_1/chaos_calmer/toolchain/gdb'
make[3]: Entering directory '/home/mitesh/linkit_smart/build_1/chaos_calmer/toolchain/binutils'
make[3]: Nothing to be done for 'prepare'.
make[3]: Leaving directory '/home/mitesh/linkit_smart/build_1/chaos_calmer/toolchain/binutils'
make[3]: Entering directory '/home/mitesh/linkit_smart/build_1/chaos_calmer/toolchain/binutils'
make[3]: Nothing to be done for 'compile'.
make[3]: Leaving directory '/home/mitesh/linkit_smart/build_1/chaos_calmer/toolchain/binutils'
make[3]: Entering directory '/home/mitesh/linkit_smart/build_1/chaos_calmer/toolchain/binutils'
make[3]: Nothing to be done for 'install'.
make[3]: Leaving directory '/home/mitesh/linkit_smart/build_1/chaos_calmer/toolchain/binutils'
make[3]: Entering directory '/home/mitesh/linkit_smart/build_1/chaos_calmer/toolchain/gcc/minimal'
make[3]: Nothing to be done for 'prepare'.
make[3]: Leaving directory '/home/mitesh/linkit_smart/build_1/chaos_calmer/toolchain/gcc/minimal'
make[3]: Entering directory '/home/mitesh/linkit_smart/build_1/chaos_calmer/toolchain/gcc/minimal'
(cd /home/mitesh/linkit_smart/build_1/chaos_calmer/build_dir/toolchain-mipsel_24kec+dsp_gcc-4.8-linaro_uClibc-0.9.33.2/gcc-linaro-4.8-2014.04-minimal && rm -f config.cache; SHELL="bash"  /home/mitesh/linkit_smart/build_1/chaos_calmer/build_dir/toolchain-mipsel_24kec+dsp_gcc-4.8-linaro_uClibc-0.9.33.2/gcc-linaro-4.8-2014.04/configure --with-bugurl=https://dev.openwrt.org/ --with-pkgversion="OpenWrt/Linaro GCC 4.8-2014.04 r49632" --prefix=/home/mitesh/linkit_smart/build_1/chaos_calmer/staging_dir/toolchain-mipsel_24kec+dsp_gcc-4.8-linaro_uClibc-0.9.33.2 --build=x86_64-linux-gnu --host=x86_64-linux-gnu --target=mipsel-openwrt-linux-uclibc --with-gnu-ld --enable-target-optspace --disable-libgomp --disable-libmudflap --disable-multilib --disable-nls  --with-host-libstdcxx=-lstdc++ --with-float=soft   --with-gmp=/home/mitesh/linkit_smart/build_1/chaos_calmer/staging_dir/host --with-mpfr=/home/mitesh/linkit_smart/build_1/chaos_calmer/staging_dir/host --with-mpc=/home/mitesh/linkit_smart/build_1/chaos_calmer/staging_dir/host --disable-decimal-float --with-mips-plt --disable-libssp --disable-__cxa_atexit --with-newlib --without-headers --enable-languages=c --disable-libsanitizer --disable-libssp --disable-shared --disable-threads );
checking build system type... x86_64-pc-linux-gnu
checking host system type... x86_64-pc-linux-gnu
checking target system type... mipsel-openwrt-linux-uclibc
checking for a BSD-compatible install... /usr/bin/install -c
checking whether ln works... yes
checking whether ln -s works... yes
checking for a sed that does not truncate output... /home/mitesh/linkit_smart/build_1/chaos_calmer/staging_dir/host/bin/sed
checking for gawk... gawk
checking for libatomic support... yes
checking for libitm support... no
checking for x86_64-linux-gnu-gcc... x86_64-linux-gnu-gcc
checking for C compiler default output file name... a.out
checking whether the C compiler works... yes
checking whether we are cross compiling... no
checking for suffix of executables... 
checking for suffix of object files... o
checking whether we are using the GNU C compiler... yes
checking whether x86_64-linux-gnu-gcc accepts -g... yes
checking for x86_64-linux-gnu-gcc option to accept ISO C89... none needed
checking for x86_64-linux-gnu-g++... x86_64-linux-gnu-g++
checking whether we are using the GNU C++ compiler... yes
checking whether x86_64-linux-gnu-g++ accepts -g... yes
checking whether g++ accepts -static-libstdc++ -static-libgcc... yes
checking for x86_64-linux-gnu-gnatbind... no
checking for gnatbind... no
checking for x86_64-linux-gnu-gnatmake... no
checking for gnatmake... no
checking whether compiler driver understands Ada... no
checking how to compare bootstrapped objects... cmp --ignore-initial=16 $$f1 $$f2
checking for objdir... .libs
checking for the correct version of gmp.h... no
configure: error: Building GCC requires GMP 4.2+, MPFR 2.4.0+ and MPC 0.8.0+.
Try the --with-gmp, --with-mpfr and/or --with-mpc options to specify
[... cut ...]
toolchain/Makefile:80: recipe for target '/home/mitesh/linkit_smart/build_1/chaos_calmer/staging_dir/toolchain-mipsel_24kec+dsp_gcc-4.8-linaro_uClibc-0.9.33.2/stamp/.toolchain_install' failed
make[1]: *** [/home/mitesh/linkit_smart/build_1/chaos_calmer/staging_dir/toolchain-mipsel_24kec+dsp_gcc-4.8-linaro_uClibc-0.9.33.2/stamp/.toolchain_install] Error 2
make[1]: Leaving directory '/home/mitesh/linkit_smart/build_1/chaos_calmer'
/home/mitesh/linkit_smart/build_1/chaos_calmer/include/toplevel.mk:181: recipe for target 'world' failed
make: *** [world] Error 2
```

nxhack, 2021-07-28:
I am now not using MT7688. I don't have the latest information either.
The reason is that the MIPS24KE does not have a hardware FPU, so it will not run the latest node.js.
The problem you are facing is that OpenWrt 15.05 is too old to support the latest toolchain (git, gcc, python3, etc).
I can't support you.

miteshdube, 2021-07-30:
Okay thanks for the your kind support and help.

closed 2021-07-30 by miteshdube