nxhack/openwrt-node-packages #989: zigbee2mqtt typescript

stek29, 2021-08-09:
since 1.21.0 zigbee2mqtt uses typescript, so tsc compilation should be handled during build

see Koenkk/zigbee2mqtt#8074

G1K, 2021-08-09:
Since we use archives and registry npm js, we already get the dist directory with the collected files via tsc and lib.

G1K, 2021-08-09:
At first, I did npm build, but here it is not needed. Here for the build from the dev branch and the github source I use tsc build

G1K, 2021-08-09:
There is a problem with running 1.21 via procd, but it consists in adding hash validation and reassembly via tsc. The check is sewn into the index.js without taking into account the dirrectory relative to which the launch is taking place. As a result, the file with the hash is not detected when running using procd, and tsc is launched, which is missing in the package. The fix has already been made in the dev branch, there is only 1 line of changes in index.js . Unfortunately, I have not figured out how to fix 1.21 so that it runs without this fix. Procd does not support changing the working directory

https://github.com/Koenkk/zigbee2mqtt/issues/8222
https://github.com/Koenkk/zigbee2mqtt/commit/62d763614a428f4dbb87ef575c0ede527cbbe647

stek29, 2021-08-09:
yeah, looks like there is no way to set cwd with procd :\

https://git.openwrt.org/?p=project/procd.git;a=blob;f=service/instance.c;h=f1242d6a5b7fa50b1f11db9bc4cf54f233ff2fac;hb=HEAD

so i guess it's zigbee2mqtt issue and not openwrt-node-packages issue, and patching index.js manually should work for now until next z2m release.
I'm closing this issue.

closed 2021-08-09 by stek29

G1K, 2021-08-09:
can only add a temporary patch that will make these changes to the build now

stek29, 2021-08-09:
@G1K not sure if it's worth it