nxhack/openwrt-node-packages #996: error compile node-v14.17.4 openwrt 19.07.4

predators46, 2021-08-22:
`{standard input}: Assembler messages:
{standard input}:126: Error: opcode not supported on this processor: mips32r2 (mips32r2) ldc1 $f0,24($fp)' {standard input}:127: Error: opcode not supported on this processor: mips32r2 (mips32r2) mtc1 $t0,$f1'
{standard input}:128: Error: opcode not supported on this processor: mips32r2 (mips32r2) `sdc1 $f0,24($fp)'   `

nxhack, 2021-08-24:
Hi @predators46 

Show me diffconfig.
(Please let me know the result of running this command.)
```
./scripts/diffconfig.sh
```

predators46, 2021-08-24:
> Hi @predators46
> 
> Show me diffconfig.
> (Please let me know the result of running this command.)
> 
> ```
> ./scripts/diffconfig.sh
> ```

I use openwrt-sdk

predators46, 2021-08-24:
adguardhome-0.106.3 eror compile openwrt-19.07.4

Log eror detail

https://pastebin.com/Utx3qG2G

` Browserslist: caniuse-lite is outdated. Please run:
npx browserslist@latest --update-db
Browserslist: caniuse-lite is outdated. Please run:
npx browserslist@latest --update-db
   1134 modules

WARNING in 
src/components/Logs/Logs.css
 174:5   ⚠  Expected empty line before declaration   declaration-empty-line-before
 439:18  ⚠  Expected numeric font-weight notation    font-weight-notation

yarn --cwd client2 build
yarn run v1.22.10                                                                                       $ webpack --config ./scripts/webpack/webpack.config.prod.js                                             (node:2547) [DEP_WEBPACK_COMPILATION_OPTIMIZE_CHUNK_ASSETS] DeprecationWarning: optimizeChunkAssets is deprecated (use Compilation.hook.processAssets instead and use one of Compilation.PROCESS_ASSETS_STAGE_* as stage option)
(Use `node --trace-deprecation ...` to show where the warning was created)
Browserslist: caniuse-lite is outdated. Please run:
npx browserslist@latest --update-db

Why you should do it regularly:
https://github.com/browserslist/browserslist#browsers-data-updating

<--- Last few GCs --->

[2547:0x564daffaa840]    45382 ms: Mark-sweep (reduce) 488.4 (494.4) -> 487.3 (495.7) MB, 990.7 / 0.0 ms  (average mu = 0.107, current mu = 0.032) allocation failure scavenge might not succeed
[2547:0x564daffaa840]    46458 ms: Mark-sweep (reduce) 488.3 (494.7) -> 487.4 (495.9) MB, 1059.0 / 0.0 ms  (average mu = 0.060, current mu = 0.016) allocation failure scavenge might not succeed

<--- JS stacktrace --->

FATAL ERROR: Ineffective mark-compacts near heap limit Allocation failed - JavaScript heap out of memory 1: 0x564dac3d1b60 node::Abort() [webpack]
 2: 0x564dac3d24be node::OnFatalError(char const*, char const*) [webpack]
 3: 0x564dac57eac2 v8::Utils::ReportOOMFailure(v8::internal::Isolate*, char const*, bool) [webpack]
 4: 0x564dac57ee3f v8::internal::V8::FatalProcessOutOfMemory(v8::internal::Isolate*, char const*, bool) [webpack]
 5: 0x564dac77e185  [webpack]
 6: 0x564dac77f013 v8::internal::Heap::RecomputeLimits(v8::internal::GarbageCollector) [webpack]
 7: 0x564dac78ed66 v8::internal::Heap::PerformGarbageCollection(v8::internal::GarbageCollector, v8::GCCallbackFlags) [webpack]
 8: 0x564dac78f6a7 v8::internal::Heap::CollectGarbage(v8::internal::AllocationSpace, v8::internal::GarbageCollectionReason, v8::GCCallbackFlags) [webpack]
 9: 0x564dac7930fc v8::internal::Heap::AllocateRawWithRetryOrFailSlowPath(int, v8::internal::AllocationType, v8::internal::AllocationOrigin, v8::internal::AllocationAlignment) [webpack]
10: 0x564dac755abc v8::internal::Factory::AllocateRaw(int, v8::internal::AllocationType, v8::internal::AllocationAlignment) [webpack]
11: 0x564dac751aa9 v8::internal::FactoryBase<v8::internal::Factory>::AllocateRawArray(int, v8::internal::AllocationType) [webpack]
12: 0x564dac751b74 v8::internal::FactoryBase<v8::internal::Factory>::NewFixedArrayWithFiller(v8::internal::Handle<v8::internal::Map>, int, v8::internal::Handle<v8::internal::Oddball>, v8::internal::AllocationType) [webpack]
13: 0x564dac9041ed  [webpack]
14: 0x564dac9043aa  [webpack]
15: 0x564dacadc005 v8::internal::Runtime_GrowArrayElements(int, unsigned long*, v8::internal::Isolate*) [webpack]
16: 0x564dacef4299  [webpack]
Aborted (core dumped)
error Command failed with exit code 134.                                                                info Visit https://yarnpkg.com/en/docs/cli/run for documentation about this command.                    Makefile:75: recipe for target 'js-build' failed
make[3]: *** [js-build] Error 134
make[3]: Leaving directory '/home/open/21/build_dir/target-mipsel_24kc_musl/adguardhome-0.106.3'
building box ../../build/static
packing file assets/apple-touch-icon-180x180.png
packed file assets/apple-touch-icon-180x180.png
packing file assets/favicon.png
packed file assets/favicon.png
packing file assets/safari-pinned-tab.svg
packed file assets/safari-pinned-tab.svg
packing file index.html
packed file index.html
packing file install.ca0ae8ce76dbc4eeb056.css
packed file install.ca0ae8ce76dbc4eeb056.css
packing file install.ca0ae8ce76dbc4eeb056.js
packed file install.ca0ae8ce76dbc4eeb056.js
packing file install.html
packed file install.html
packing file login.ca0ae8ce76dbc4eeb056.css
packed file login.ca0ae8ce76dbc4eeb056.css
packing file login.ca0ae8ce76dbc4eeb056.js
packed file login.ca0ae8ce76dbc4eeb056.js
packing file login.html
packed file login.html
packing file main.ca0ae8ce76dbc4eeb056.css
packed file main.ca0ae8ce76dbc4eeb056.css
packing file main.ca0ae8ce76dbc4eeb056.js
packed file main.ca0ae8ce76dbc4eeb056.js
built box ../../build/static with ["assets/apple-touch-icon-180x180.png" "assets/favicon.png" "assets/safari-pinned-tab.svg" "index.html" "install.ca0ae8ce76dbc4eeb056.css" "install.ca0ae8ce76dbc4eeb056.js" "install.html" "login.ca0ae8ce76dbc4eeb056.css" "login.ca0ae8ce76dbc4eeb056.js" "login.html" "main.ca0ae8ce76dbc4eeb056.css" "main.ca0ae8ce76dbc4eeb056.js"]
building box ../../build2/static
Error: lstat build2: no such file or directory
Usage:
  packr [flags]
  packr [command]

Available Commands:
  build       Wraps the go build command with packr
  clean       removes any *-packr.go files
  help        Help about any command
  install     Wraps the go install command with packr
  version     prints packr version

Flags:
  -z, --compress       compress box contents
  -h, --help           help for packr
  -i, --input string   path to scan for packr Boxes (default "/home/open/21/build_dir/target-mipsel_24kc_musl/adguardhome-0.106.3")
  -v, --verbose        print verbose logging information

Use "packr [command] --help" for more information about a command.
 `

predators46, 2021-08-24:
Fix thanks

closed 2021-08-24 by predators46