nxhack/openwrt-node-packages #997: Error: opcode not supported on this processor: mips32r2 (mips32r2) `ldc1 $f0,24($fp)'

predators46, 2021-08-29:
I reopen this problem. i have tried openwrt 18.06.4 and 19.07.4 but still found this problem. I use openwrt-sdk. nodes always fail when I create it

nxhack, 2021-09-01:
Hi @predators46 

I tried to use openwrt-sdk, but I can't change the kernel config and I can't configure the "MIPS FPU Emulator". Therefore, I can't even select node.js.

Please tell me the exact steps you took.

predators46, 2021-09-01:
> Hi @predators46
> 
> I tried to use openwrt-sdk, but I can't change the kernel config and I can't configure the "MIPS FPU Emulator". Therefore, I can't even select node.js.
> 
> Please tell me the exact steps you took.

thank you for your reply. I didn't change the kernel configuration

nxhack, 2021-09-01:
> I didn't change the kernel configuration

Then you shouldn't be able to build node.js.

This is the procedure I tried. node.js is not an available option.
```bash
wget https://downloads.openwrt.org/releases/19.07.7/targets/ramips/mt7620/openwrt-sdk-19.07.7-ramips-mt7620_gcc-7.5.0_musl.Linux-x86_64.tar.xz
tar xfJ openwrt-sdk-19.07.7-ramips-mt7620_gcc-7.5.0_musl.Linux-x86_64.tar.xz 
cd openwrt-sdk-19.07.7-ramips-mt7620_gcc-7.5.0_musl.Linux-x86_64
echo 'src-git node https://github.com/nxhack/openwrt-node-packages.git;openwrt-19.07' >> feeds.conf.default
./scripts/feeds update -a
rm ./package/feeds/packages/node
rm ./package/feeds/packages/node-*
./scripts/feeds install -a -p node
make menuconfig 
```
ref: https://openwrt.org/docs/guide-developer/using_the_sdk

closed 2021-09-02 by predators46