nxhack/openwrt-node-packages #1080: node zigbee2mqtt

ANARHIST1984, 2021-11-14:
Do not start without install Winston-transport

nxhack, 2021-11-14:
I am not familiar with the usage of zigbee2mqtt.

You might want to look for similar issues on upstream.

https://github.com/Koenkk/zigbee2mqtt/search?q=winston-transport&type=issues

ANARHIST1984, 2021-11-17:
to fix this need open 
cd /opt/zigbee2mqtt nom install winston transport

nxhack, 2021-11-17:
@ANARHIST1984 
Check your "NODE_PATH".

G1K, 2021-12-05:
The same error was encountered by the guys who tried to build zigbee2mqtt. To reduce the package size, I am building a package with the production flag for zigbee2mqtt and its two main dependencies. Otherwise, the packet becomes about 200 MB. unpacked. These edits are in a separate patch. With them, the assembly takes place correctly and fits in the xiaomi dgnwg05lm gateway, but they may not be optimal and not beautiful. That's why I don't do PR.

https://github.com/openlumi/openwrt-fastbuild-actions/blob/master/user/21.02.0/patches/001_zigbee2mqtt_optimize.patch

Maybe you'll take a look and decide what can be moved to the main repository.

nxhack, 2021-12-06:
I'm going to test using this patch as a reference. Please give me some time.

nxhack, 2021-12-06:
https://github.com/nxhack/openwrt-node-packages/pull/1093

Explicitly specified NODE_PATH.
This resolved the issue.
However, the root cause will be investigated later.

Added modeclean.

I would like feedback on the handling of configuration.yaml.

G1K, 2022-01-16:
@nxhack Good afternoon.
There was no way to check last time . The main thing in my size reduction patch is not modclean, which potentially removes garbage, but can be dangerous. 

The main thing is a re -installation with the production flag .

First npm i , and after `npm ci --production` , it gives a huge effect

I currently have no way to test this in the openwrt build , but on a regular machine with npm it is reproduced like this
after 

`npm i --production zigbee2mqtt`

the size of the directory is almost 600 MB
[image]

If you now go to the zigbee2mqtt directory and make npm ci --production
the size will be reduced to **65 MB** 

If you also remove the duplicate dependency `node_modules/zigbee-herdsman-converters/node_modules/zigbee-herdsman`

Which is almost not needed, but because of the difference in the version numbers of dependencies, it is dragged repeatedly.
https://github.com/Koenkk/zigbee-herdsman-converters/issues/2189

The size is reduced by another 10 MB 

And after that I did the cleaning with modclean. The final package in the archive had a size of about 5 MB, unpacked on the xiaomi gateway up to 20 MB

I do not know why the production flag does not work with direct installation via npm and therefore used such a workaround hack.

I would appreciate it if you re-use it. Then it will be possible not to update this patch in the auto-assembly of the zigbee2mqtt package for the xiaomi gateway.

G1K, 2022-01-16:
About configuration.yml.
I moved it to etc and declared it as config
```
define Package/node-zigbee2mqtt/conffiles
/etc/zigbee2mqtt/configuration.yml
endef
```
This allows you to keep the config and data separate from the application. When updating the application, they are not overwritten, and can also be backed up with openwrt tools.

In order for zigbee2mqtt to work with another data folder, I added a script to init 

```
procd_set_param env ZIGBEE2MQTT_DATA=/etc/zigbee2mqtt/
```
This solution is not perfect, and sometimes leads to user errors. So I wanted to take another step. Replace the data directory with configuration.yml in the directory with the program with a symbolic link leading to etc. 

This would allow users to ignore and edit the config from anywhere.