nxhack/openwrt-node-packages #1141: Homebridge

ANARHIST1984, 2022-01-10:
I compile Homebridge and install to the router, but in apple home app I didnt found device to add in to Homekit.
I saw one solution. Need to install Avahi and it has to be configured. But I haven't succeed. Spend few days on this(
Next, I try use docker image and have same problem, bonjour/mdns dosnt work. But Homebrige start and work
After I install Homebridge 1.2.4 from OpenWrt repo and its work fine! I can see "device" in home kit app.
Nest, when I open this https://github.com/openwrt/packages/tree/38f01ad2c9738c6c5a5dbda85019bb9cf5bc10f8/lang/node-homebridge and I saw, this makefile made by you!
What different between "official" package and openwrt-node-packages/node-homebridge?
BTW I use node and node-npm from this repo for start node-homebridge

nxhack, 2022-01-11:
@ANARHIST1984 
> I saw one solution. Need to install Avahi and it has to be configured.

There are several mDNS packages available for OpenWrt. It is up to the user to decide which one to choose.
 (umdns / avahi-dbus-daemon / mdnsresponder)

I think this thread will help you to figure out how to solve the mDNS related problem.
https://github.com/nxhack/openwrt-node-packages/issues/526

The version is different from the official OpenWrt community package. Please give me feedback if you can identify the problem.

ANARHIST1984, 2022-01-11:
Update
HB from this repo, HB have internet and check updates working fine, but not working mdns
install HB from OWRT repo, always have check updates! mdns working

ANARHIST1984, 2022-01-11:
Update
"mdns": {
   "interface": "192.168.1.1"
}
or 
"mdns": {
   "interface": "eth0"
}
successfully paired and working!

nxhack, 2022-01-11:
@ANARHIST1984 

Thank you! for the report.

I checked it out and it seems that the behavior has changed since v1.3.0.
https://github.com/homebridge/homebridge/releases/tag/v1.3.0

This wiki is an example of using the Homebridge UI.
https://github.com/homebridge/homebridge/wiki/mDNS-Options

SooHandsome, 2023-01-14:
Hi, I’m super noob here,
But I manage to run Homebridge and Homebridge UI in openwrt.
Adding mdns in config.json works well, and I successfully paired my device. However, in system log it complains that the mdns option in config.json being deprecated. But when I used advertiser and bind option, it complains dbus don’t have /usr/share/dbus-1/org.freedesktop…. File and then crashed.
Is dbus package in openwrt somehow has limited functionality?

nxhack, 2023-01-15:
@SooHandsome 

OpenWrt's avahi package does not install that file. Attempting to install that file manually seems to circumvent the error.

https://github.com/lathiat/avahi/blob/master/avahi-daemon/org.freedesktop.Avahi.Server.xml

SooHandsome, 2023-01-16:
@nxhack 
ok, I have tried it, it doesn't complain anymore when using **avahi** in `advertiser` option in `bridge` section after putting the introspection file.
Also, I remove the `mdns` section and the homebridge is still discoverable.
Thx