nxhack/openwrt-node-packages #1205: TypeError: Class extends value undefined is not a constructor or null at Object

WYC-2020, 2022-03-08:
On openwrt, compiling yarn with node-v16.14.0 gives error
log:
make[4]: Leaving directory '/home/hzybb/lede/build_dir/hostpkg/node-v16.14.0' mkdir -p /home/hzybb/lede/staging_dir/hostpkg/stamp touch /home/hzybb/lede/build_dir/hostpkg/node-v16.14.0/.built touch /home/hzybb/lede/staging_dir/hostpkg/stamp/.node_installed make[3]: Leaving directory '/home/hzybb/lede/feeds/packages/lang/node' time: package/feeds/packages/node/host-compile#5197.23#310.51#5514.01 make[3]: Entering directory '/home/hzybb/lede/feeds/packages/lang/node-yarn' mkdir -p /home/hzybb/lede/dl SHELL= flock /home/hzybb/lede/tmp/.yarn-1.22.17.tgz.flock -c ' /home/hzybb/lede/scripts/download.pl "/home/hzybb/lede/dl" "yarn-1.22.17.tgz" "267982c61119a055ba2b23d9cf90b02d3d16c202c03cb0c3a53b9633eae37249" "" "https://registry.npmjs.org/yarn/-/" ' . /home/hzybb/lede/include/shell.sh; gzip -dc /home/hzybb/lede/dl/yarn-1.22.17.tgz | tar -C /home/hzybb/lede/build_dir/hostpkg/node-yarn-1.22.17 -xf - --strip-components 1 [ ! -d ./src/ ] || cp -fpR ./src/* /home/hzybb/lede/build_dir/hostpkg/node-yarn-1.22.17 touch /home/hzybb/lede/build_dir/hostpkg/node-yarn-1.22.17/.prepareddde67abb946bbb04ee50db61d61a453d_6664517399ebbbc92a37c5bb081b5c53 (cd /home/hzybb/lede/build_dir/hostpkg/node-yarn-1.22.17/; if [ -x configure ]; then cp -fpR /home/hzybb/lede/scripts/config.{guess,sub} /home/hzybb/lede/build_dir/hostpkg/node-yarn-1.22.17// && CC="gcc" CFLAGS="-O2 -I/home/hzybb/lede/staging_dir/host/include -I/home/hzybb/lede/staging_dir/hostpkg/include -I/home/hzybb/lede/staging_dir/target-x86_64_musl/host/include" CXX="g++" CPPFLAGS="-I/home/hzybb/lede/staging_dir/host/include -I/home/hzybb/lede/staging_dir/hostpkg/include -I/home/hzybb/lede/staging_dir/target-x86_64_musl/host/include" LDFLAGS="-L/home/hzybb/lede/staging_dir/host/lib -L/home/hzybb/lede/staging_dir/hostpkg/lib -L/home/hzybb/lede/staging_dir/target-x86_64_musl/host/lib" CONFIG_SHELL="/usr/bin/env bash" bash ./configure --target=x86_64-pc-linux-gnu --host=x86_64-pc-linux-gnu --build=x86_64-pc-linux-gnu --program-prefix="" --program-suffix="" --prefix=/home/hzybb/lede/staging_dir/hostpkg --exec-prefix=/home/hzybb/lede/staging_dir/hostpkg --sysconfdir=/home/hzybb/lede/staging_dir/hostpkg/etc --localstatedir=/home/hzybb/lede/staging_dir/hostpkg/var --sbindir=/home/hzybb/lede/staging_dir/hostpkg/bin ; fi ) touch /home/hzybb/lede/build_dir/hostpkg/node-yarn-1.22.17/.configured cd /home/hzybb/lede/build_dir/hostpkg/node-yarn-1.22.17; CFLAGS="-O2 -I/home/hzybb/lede/staging_dir/host/include -I/home/hzybb/lede/staging_dir/hostpkg/include -I/home/hzybb/lede/staging_dir/target-x86_64_musl/host/include" CPPFLAGS="-I/home/hzybb/lede/staging_dir/host/include -I/home/hzybb/lede/staging_dir/hostpkg/include -I/home/hzybb/lede/staging_dir/target-x86_64_musl/host/include" CXXFLAGS="" LDFLAGS="-L/home/hzybb/lede/staging_dir/host/lib -L/home/hzybb/lede/staging_dir/hostpkg/lib -L/home/hzybb/lede/staging_dir/target-x86_64_musl/host/lib" npm_config_nodedir=/home/hzybb/lede/staging_dir/target-x86_64_musl/usr/ npm_config_cache=/home/hzybb/lede/tmp/npm-cache-iQiLTEL0tm npm_config_tmp=/home/hzybb/lede/tmp/npm-tmp-iQiLTEL0tm npm install --production --global-style --no-save --omit=dev --no-package-lock TypeError: Class extends value undefined is not a constructor or null at Object. (/home/hzybb/lede/staging_dir/hostpkg/lib/node_modules/npm/node_modules/socks-proxy-agent/dist/agent.js:114:44) at Module._compile (node:internal/modules/cjs/loader:1103:14) at Object.Module._extensions..js (node:internal/modules/cjs/loader:1155:10) at Module.load (node:internal/modules/cjs/loader:981:32) at Function.Module._load (node:internal/modules/cjs/loader:822:12) at Module.require (node:internal/modules/cjs/loader:1005:19) at require (node:internal/modules/cjs/helpers:102:18) at Object. (/home/hzybb/lede/staging_dir/hostpkg/lib/node_modules/npm/node_modules/socks-proxy-agent/dist/index.js:5:33) at Module._compile (node:internal/modules/cjs/loader:1103:14) at Object.Module._extensions..js (node:internal/modules/cjs/loader:1155:10) at Module.load (node:internal/modules/cjs/loader:981:32) at Function.Module._load (node:internal/modules/cjs/loader:822:12) at Module.require (node:internal/modules/cjs/loader:1005:19) at require (node:internal/modules/cjs/helpers:102:18) at Object. (/home/hzybb/lede/staging_dir/hostpkg/lib/node_modules/npm/node_modules/make-fetch-happen/lib/agent.js:161:25) at Module._compile (node:internal/modules/cjs/loader:1103:14) make[3]: *** [Makefile:115: /home/hzybb/lede/build_dir/hostpkg/node-yarn-1.22.17/.built] Error 1 make[3]: Leaving directory '/home/hzybb/lede/feeds/packages/lang/node-yarn' time: package/feeds/packages/node-yarn/host-compile#0.35#0.38#0.96 ERROR: package/feeds/packages/node-yarn [host] failed to build. make[2]: *** [package/Makefile:114: package/feeds/packages/node-yarn/host/compile] Error 1 make[2]: Leaving directory '/home/hzybb/lede' make[1]: *** [package/Makefile:110: /home/hzybb/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2 make[1]: Leaving directory '/home/hzybb/lede' make: *** [/home/hzybb/lede/include/toplevel.mk:230: world] Error 2

nxhack, 2022-03-15:
@WYC-2020 

There seems to be an error in the host' s node.
Please try building the host' s node once.

```
make V=s package/feeds/node/node/host/{clean,compile}
```

nxhack, 2022-04-03:
@WYC-2020 

I have applied your patch. Thank you very much.
https://github.com/nxhack/openwrt-node-packages/commit/f50e0b05fe9ef934c7ac1fc6cde6dfd28b4366c9

closed 2022-04-03 by nxhack