nxhack/openwrt-node-packages #1217: Zigbee2Mqtt doesn't work

Tao173, 2022-03-24:
I have Z2M and Node-MQtt installed. But Z2M seems to be unable to connect to the MQTT server. The log I see clearly in TMP /log/ Z2M shows that the MQTT server cannot be connected. How to configure it?
This is my .config for OpenWrt.
`https://github.com/Tao173/Actions-OpenWrt-Node/releases/download/2022.03.24-0616/config.buildinfo`
This is screen for OpenWrt installed z2m and mqtt.
[image]
This is z2m logs.
`info  2022-03-24 20:34:22: Logging to console and directory: '/tmp/log/zigbee2mqtt/2022-03-24.20-34-20' filename: log.txt
info  2022-03-24 20:34:22: Starting Zigbee2MQTT version 1.24.0 (commit #7a2ddf2)
info  2022-03-24 20:34:22: Starting zigbee-herdsman (0.14.20)
info  2022-03-24 20:35:03: zigbee-herdsman started (resumed)
info  2022-03-24 20:35:03: Coordinator firmware version: '{"meta":{"maintrel":2,"majorrel":2,"minorrel":7,"product":2,"revision":20200327,"transportrev":2},"type":"zStack30x"}'
info  2022-03-24 20:35:03: Currently 0 devices are joined:
warn  2022-03-24 20:35:03: `permit_join` set to  `true` in configuration.yaml.
warn  2022-03-24 20:35:03: Allowing new devices to join.
warn  2022-03-24 20:35:03: Set `permit_join` to `false` once you joined all devices.
info  2022-03-24 20:35:03: Zigbee: allowing new devices to join.
info  2022-03-24 20:35:03: Connecting to MQTT server at mqtt://localhost
error 2022-03-24 20:35:03: MQTT failed to connect: connect ECONNREFUSED 127.0.0.1:1883
error 2022-03-24 20:35:03: Exiting...
info  2022-03-24 20:35:03: Stopping zigbee-herdsman...
info  2022-03-24 20:35:41: Stopped zigbee-herdsman
`

nxhack, 2022-03-24:
@Tao173 

Ask the zigbee2mqtt community about zigbee2mqtt specifications.

> 2022-03-24 20:35:03: Connecting to MQTT server at mqtt://localhost error
> 022-03-24 20:35:03: MQTT failed to connect: connect ECONNREFUSED 127.0.0.1:1883 error

This error message is related to the mqtt server. I cannot support this more.

Tao173, 2022-03-24:
I installed Node-MQTT, it comes from your repository, I don't know how to configure it, just make sure it is installed into the system, does it prove that the OP contains an MQTT server inside? Or I still need to install another MQTT server. Just like mosquitos. Install other projects from your repository while I install Zigbee2Mqtt? maybe they have dependencies.

nxhack, 2022-03-24:
https://flemmingss.com/how-to-set-up-zigbee2mqtt-on-a-raspberry-pi-and-integrate-it-with-home-assistant/
https://www.youtube.com/watch?v=-YeObwEZqL4
https://www.zigbee2mqtt.io/guide/installation/

As the name suggests, "zigbee2mqtt" connects the zigbee device interface to the mqtt broker, so you will need to prepare those two yourself.

Tao173, 2022-03-24:
> https://flemmingss.com/how-to-set-up-zigbee2mqtt-on-a-raspberry-pi-and-integrate-it-with-home-assistant/ https://www.youtube.com/watch?v=-YeObwEZqL4 https://www.zigbee2mqtt.io/guide/installation/ [...]

OK,Thank you.I will try it.

closed 2022-03-24 by Tao173