nxhack/openwrt-node-packages #1223: Update node openwrt

predators46, 2022-04-03:
Hy @nxhack 

looks like node should be updated in openwrt master branch

`root@OpenWrt:~/discord# npm install discord.js
npm notice created a lockfile as package-lock.json. You should commit this file.
npm WARN notsup Unsupported engine for discord.js@13.6.0: wanted: {"node":">=16.6.0","npm":">=7.0.0"} (current: {"node":"14.18.3","npm":"6.14.15"})
npm WARN notsup Not compatible with your version of node/npm: discord.js@13.6.0
npm WARN notsup Unsupported engine for @sapphire/async-queue@1.3.1: wanted: {"node":">=v14.0.0","npm":">=7.0.0"} (current: {"node":"14.18.3","npm":"6.14.15"})
npm WARN notsup Not compatible with your version of node/npm: @sapphire/async-queue@1.3.1
npm WARN notsup Unsupported engine for @discordjs/builders@0.11.0: wanted: {"node":">=16.0.0","npm":">=7.0.0"} (current: {"node":"14.18.3","npm":"6.14.15"})
npm WARN notsup Not compatible with your version of node/npm: @discordjs/builders@0.11.0
npm WARN notsup Unsupported engine for @discordjs/collection@0.4.0: wanted: {"node":">=16.0.0","npm":">=7.0.0"} (current: {"node":"14.18.3","npm":"6.14.15"})
npm WARN notsup Not compatible with your version of node/npm: @discordjs/collection@0.4.0
npm WARN discord@1.0.0 No description
npm WARN discord@1.0.0 No repository field.

+ discord.js@13.6.0
added 24 packages from 81 contributors and audited 24 packages in 18.674s

2 packages are looking for funding
  run `npm fund` for details

found 0 vulnerabilities`

nxhack, 2022-04-03:
@predators46 

Node v16.x will be released with npm v8.x as standard, which will require a lot of modifications to node-related packages in openwrt's community packages. So I am holding off on upgrading to v16.

Of course, the custom packages in this repository are already supported.

predators46, 2022-04-03:
> @predators46
> 
> Node v16.x will be released with npm v8.x as standard, which will require a lot of modifications to node-related packages in openwrt's community packages. So I am holding off on upgrading to v16.
> 
> Of course, the custom packages in this repository are already supported.

Thanks friend

nxhack, 2022-04-03:
@predators46 

I do know that it is time to upgrade the version.
Ok I will update to the master branch of the community package, the inconsistency of the npm package may be handled on a case-by-case basis.

predators46, 2022-04-03:
> @predators46
> 
> I do know that it is time to upgrade the version. Ok I will update to the master branch of the community package, the inconsistency of the npm package may be handled on a case-by-case basis.

I will wait and test it

predators46, 2022-04-03:
@nxhack 

i found an error while creating in openwrt 21.02.2

`musl/node-v16.14.2/out/Release/obj.host/torque_base/deps/v8/src/torque/implementation-visitor.o.d.raw   -c
../deps/v8/src/torque/implementation-visitor.cc: In instantiation of 'v8::internal::torque::{anonymous}::CppClassGenerator::GenerateFieldAccessors(const v8::internal::torque::Field&, std::vector<const v8::internal::torque::Field*>&)::<lambda(auto:3&)> [with auto:3 = std::basic_ostream<char>]':                  /usr/include/c++/6/type_traits:2481:26:   required by substitution of 'template<class _Fn, class ... _Args> static std::__result_of_success<decltype (declval<_Fn>()((declval<_Args>)()...)), std::__invoke_other> std::__result_of_other_impl::_S_test(int) [with _Fn = v8::internal::torque::{anonymous}::CppClassGenerator::GenerateFieldAccessors(const v8::internal::torque::Field&, std::vector<const v8::internal::torque::Field*>&)::<lambda(auto:3&)>&; _Args = {std::basic_ostream<char, std::char_traits<char> >&}]'         /usr/include/c++/6/type_traits:2492:55:   required from 'struct std::__result_of_impl<false, false, v8::internal::torque::{anonymous}::CppClassGenerator::GenerateFieldAccessors(const v8::internal::torque::Field&, std::vector<const v8::internal::torque::Field*>&)::<lambda(auto:3&)>&, std::basic_ostream<char, std::char_traits<char> >&>'                                                                                /usr/include/c++/6/type_traits:2496:12:   required from 'class std::result_of<v8::internal::torque::{anonymous}::CppClassGenerator::GenerateFieldAccessors(const v8::internal::torque::Field&, std::vector<const v8::internal::torque::Field*>&)::<lambda(auto:3&)>&(std::basic_ostream<char>&)>'                       /usr/include/c++/6/functional:1913:9:   required by substitution of 'template<class _Functor, class, class> std::function<_Res(_ArgTypes ...)>::function(_Functor) [with _Functor = v8::internal::torque::{anonymous}::CppClassGenerator::GenerateFieldAccessors(const v8::internal::torque::Field&, std::vector<const v8::internal::torque::Field*>&)::<lambda(auto:3&)>; <template-parameter-1-2> = void; <template-parameter-1-3> = <missing>]'                                                                                      ../deps/v8/src/torque/implementation-visitor.cc:4354:6:   required from here                            ../deps/v8/src/torque/implementation-visitor.cc:4352:7: error: cannot call member function 'void v8::internal::torque::{anonymous}::CppClassGenerator::EmitLoadFieldStatement(std::ostream&, const v8::internal::torque::Field&, std::vector<const v8::internal::torque::Field*>&)' without object                             EmitLoadFieldStatement(stream, class_field, struct_fields);                                             ^~~~~~~~~~~~~~~~~~~~~~                                                                           ../deps/v8/src/torque/implementation-visitor.cc: In instantiation of 'v8::internal::torque::{anonymous}::CppClassGenerator::GenerateFieldAccessors(const v8::internal::torque::Field&, std::vector<const v8::internal::torque::Field*>&)::<lambda(auto:4&)> [with auto:4 = std::basic_ostream<char>]':                  /usr/include/c++/6/type_traits:2481:26:   required by substitution of 'template<class _Fn, class ... _Args> static std::__result_of_success<decltype (declval<_Fn>()((declval<_Args>)()...)), std::__invoke_other> std::__result_of_other_impl::_S_test(int) [with _Fn = v8::internal::torque::{anonymous}::CppClassGenerator::GenerateFieldAccessors(const v8::internal::torque::Field&, std::vector<const v8::internal::torque::Field*>&)::<lambda(auto:4&)>&; _Args = {std::basic_ostream<char, std::char_traits<char> >&}]'         /usr/include/c++/6/type_traits:2492:55:   required from 'struct std::__result_of_impl<false, false, v8::internal::torque::{anonymous}::CppClassGenerator::GenerateFieldAccessors(const v8::internal::torque::Field&, std::vector<const v8::internal::torque::Field*>&)::<lambda(auto:4&)>&, std::basic_ostream<char, std::char_traits<char> >&>'                                                                                /usr/include/c++/6/type_traits:2496:12:   required from 'class std::result_of<v8::internal::torque::{anonymous}::CppClassGenerator::GenerateFieldAccessors(const v8::internal::torque::Field&, std::vector<const v8::internal::torque::Field*>&)::<lambda(auto:4&)>&(std::basic_ostream<char>&)>'                       /usr/include/c++/6/functional:1913:9:   required by substitution of 'template<class _Functor, class, class> std::function<_Res(_ArgTypes ...)>::function(_Functor) [with _Functor = v8::internal::torque::{anonymous}::CppClassGenerator::GenerateFieldAccessors(const v8::internal::torque::Field&, std::vector<const v8::internal::torque::Field*>&)::<lambda(auto:4&)>; <template-parameter-1-2> = void; <template-parameter-1-3> = <missing>]'                                                                                      ../deps/v8/src/torque/implementation-visitor.cc:4381:6:   required from here                            ../deps/v8/src/torque/implementation-visitor.cc:4380:7: error: cannot call member function 'void v8::internal::torque::{anonymous}::CppClassGenerator::EmitStoreFieldStatement(std::ostream&, const v8::internal::torque::Field&, std::vector<const v8::internal::torque::Field*>&)' without object                            EmitStoreFieldStatement(stream, class_field, struct_fields);                                            ^~~~~~~~~~~~~~~~~~~~~~~                                                                          tools/v8_gypfiles/torque_base.host.mk:159: recipe for target '/home/admin/21/build_dir/target-aarch64_cortex-a53_musl/node-v16.14.2/out/Release/obj.host/torque_base/deps/v8/src/torque/implementation-visitor.o' failed
make[4]: *** [/home/admin/21/build_dir/target-aarch64_cortex-a5`

predators46, 2022-04-03:
`error: cannot call member function 'void v8::internal::torque::{anonymous}::CppClassGenerator::EmitLoadFieldStatement(std::ostream&, const v8::internal::torque::Field&, std::vector<const v8::internal::torque::Field*>&)' without object                             EmitLoadFieldStatement(stream, class_field, struct_fields);                                             ^~~~~~~~~~~~~~~~~~~~~~`

predators46, 2022-04-03:
can you change the node version to 17.x.x

https://www.linuxquestions.org/questions/showthread.php?p=6312663#post6312663

predators46, 2022-04-03:
https://www.linuxquestions.org/questions/showthread.php?p=6312663#post6312663

closed 2022-04-03 by predators46