nxhack/openwrt-node-packages #1227: ejdb2

predators46, 2022-04-04:
Error ejdb2 build

`cp -fpR /home/ubuntu/21/build_dir/target-aarch64_cortex-a53_musl/node-ejdb2_node-2.62.5/linux-x64/ejdb2_node.node /home/ubuntu/21/build_dir/target-aarch64_cortex-a53_musl/node-ejdb2_node-2.62.5/.pkgdir/node-ejdb2_node/usr/lib/node/ejdb2_node/build/Release
cp: cannot stat '/home/ubuntu/21/build_dir/target-aarch64_cortex-a53_musl/node-ejdb2_node-2.62.5/linux-x64/ejdb2_node.node': No such file or directory
Makefile:91: recipe for target '/home/ubuntu/21/build_dir/target-aarch64_cortex-a53_musl/node-ejdb2_node-2.62.5/.pkgdir/node-ejdb2_node.installed' failed
make[2]: *** [/home/ubuntu/21/build_dir/target-aarch64_cortex-a53_musl/node-ejdb2_node-2.62.5/.pkgdir/node-ejdb2_node.installed] Error 1`

nxhack, 2022-04-04:
Hi @predators46 

Please tell us about your cross-compiling environment.

Machine architecture 
Operation System

```
arch
uname -a
cat /etc/lsb-release
```

predators46, 2022-04-05:
@nxhack 

`ubuntu@ip-172-26-7-209:~$ arch
x86_64
ubuntu@ip-172-26-7-209:~$ uname -a                                                                      Linux ip-172-26-7-209 4.15.0-1021-aws #21-Ubuntu SMP Tue Aug 28 10:23:07 UTC 2018 x86_64 x86_64 x86_64 GNU/Linux
ubuntu@ip-172-26-7-209:~$ cat /etc/lsb-release                                                          DISTRIB_ID=Ubuntu
DISTRIB_RELEASE=18.04
DISTRIB_CODENAME=bionic
DISTRIB_DESCRIPTION="Ubuntu 18.04.6 LTS"
ubuntu@ip-172-26-7-209:~$`

nxhack, 2022-04-05:
It's just like my environment!
I am able to build successfully.

Run this command to show all log.
```
make V=s package/feeds/node/node-ejdb2_node/{clean,compile}
```

predators46, 2022-04-05:
I'll give it a try in a moment. and now I'm trying to make node v16 for mips but it's not possible to mips using node again. i have added fpu emulator and fp

`{standard input}: Assembler messages:
{standard input}:130: Error: opcode not supported on this processor: mips32r2 (mips32r2) `ldc1 $f0,24($fp)'
{standard input}:131: Error: opcode not supported on this processor: mips32r2 (mips32r2) `mtc1 $t0,$f1'
{standard input}:132: Error: opcode not supported on this processor: mips32r2 (mips32r2) `sdc1 $f0,24($fp)'
tools/v8_gypfiles/v8_libbase.target.mk:189: recipe for target '/home/ubuntu/gg/build_dir/target-mipsel_24kc_musl/node-v16.14.2/out/Release/obj.target/v8_libbase/deps/v8/src/base/cpu.o' failed
make[4]: *** [/home/ubuntu/gg/build_dir/target-mipsel_24kc_musl/node-v16.14.2/out/Release/obj.target/v8_libbase/deps/v8/src/base/cpu.o] Error 1`

nxhack, 2022-04-05:
It does not work on mips.

predators46, 2022-04-05:
yes i know that

predators46, 2022-04-05:
@nxhack 

looks like mips can still use node in version 14

BusyBox v1.33.2 (2022-03-23 12:52:20 UTC) built-in shell (ash)

  _______                     ________        __
 |       |.-----.-----.-----.|  |  |  |.----.|  |_
 |   -   ||  _  |  -__|     ||  |  |  ||   _||   _|
 |_______||   __|_____|__|__||________||__|  |____|
          |__| W I R E L E S S   F R E E D O M
 -----------------------------------------------------
 OpenWrt 21.02.2, r16495-bf0c965af0
 -----------------------------------------------------
=== WARNING! =====================================
There is no root password defined on this device!
Use the "passwd" command to set up a new password
in order to prevent unauthorized SSH logins.
--------------------------------------------------
root@OpenWrt:~# uname -a                                                  
Linux OpenWrt 5.4.179 #0 Wed Feb 16 20:29:10 2022 mips GNU/Linux          
root@OpenWrt:~# node -v                                                   
v14.19.1                                                                  
root@OpenWrt:~# npm -v                                                    
6.14.16                                                                   
root@OpenWrt:~# node hello.js
Hello, World!
root@OpenWrt:~#

nxhack, 2022-04-05:
See the conversation that follows.

https://github.com/nxhack/openwrt-node-packages/issues/464#issuecomment-508017359

nxhack, 2022-04-05:
In any case, I do not plan to support mips.
However, OpenWrt supports mipsel with hardware FPU, so I do support that to a limited extent.

nxhack, 2022-04-05:
The v8 engine project welcomes mipseb (32, 64) maintainers and would welcome your commitments.

nxhack, 2022-04-05:
Please know that mips with software FPU is too slow and node.js does not run well enough.

predators46, 2022-04-05:
yes there is no other option for mips using fpu emulator on openwrt. this is only an alternative if anyone wants to use nodes on mips

predators46, 2022-04-05:
@nxhack 

`-- Using src='https://github.com/Softmotions/iowow/archive/master.zip'
CMake Error at extern_iowow-stamp/download-extern_iowow.cmake:170 (message):
  Each download failed!

    error: downloading 'https://github.com/Softmotions/iowow/archive/master.zip' failed
          status_code: 6
          status_string: "Couldn't resolve host name"
          log:
          --- LOG BEGIN ---
          getaddrinfo(3) failed for github.com:443

  Couldn't resolve host 'github.com'

  Closing connection 0`

predators46, 2022-04-05:
[log.txt](https://github.com/nxhack/openwrt-node-packages/files/8422151/log.txt)
Detail log error

nxhack, 2022-04-05:
You must solve this problem.
```
Couldn't resolve host 'github.com'
```

This is outside the scope of my support.
Please refer to the Ubuntu documentation and other documentation yourself for Linux network settings.

predators46, 2022-04-05:
very strange even though I'm using a vps

closed 2022-04-06 by predators46

openwrtfirmwarecostum, 2022-04-07:
@nxhack 

`ortex-a53_gcc-8.4.0_musl/include/fortify -I/home/ubuntu/gg/staging_dir/toolchain-aarch64_cortex-a53_gcc-8.4.0_musl/include -DPIC -fPIC  -c
../src/ffi.cc:4:10: fatal error: fficonfig.h: No such file or directory
 #include "fficonfig.h"
          ^~~~~~~~~~~~~
compilation terminated.`

nxhack, 2022-04-07:
Please put up another issue if the topic changes.

master:
https://github.com/nxhack/openwrt-node-packages/blob/master/node-ffi-napi/Makefile#L23

openwrt-21.02:
https://github.com/nxhack/openwrt-node-packages/blob/openwrt-21.02/node-ffi-napi/Makefile#L23

Use the branch that matches your OpenWrt version.