nxhack/openwrt-node-packages #1290: No such file or directory

p3x-robot, 2022-05-29:
```txt
  arm-openwrt-linux-muslgnueabi-g++ -Os -pipe -o /build/source/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/node-v16.15.0/out/Release/obj.target/v8_co>
  arm-openwrt-linux-muslgnueabi-g++ -Os -pipe -o /build/source/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/node-v16.15.0/out/Release/obj.target/v8_co>
  arm-openwrt-linux-muslgnueabi-g++ -Os -pipe -o /build/source/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/node-v16.15.0/out/Release/obj.target/v8_co>
  arm-openwrt-linux-muslgnueabi-g++ -Os -pipe -o /build/source/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/node-v16.15.0/out/Release/obj.target/v8_co>
  arm-openwrt-linux-muslgnueabi-g++ -Os -pipe -o /build/source/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/node-v16.15.0/out/Release/obj.target/v8_co>
  arm-openwrt-linux-muslgnueabi-g++ -Os -pipe -o /build/source/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/node-v16.15.0/out/Release/obj.target/v8_co>
  arm-openwrt-linux-muslgnueabi-g++ -Os -pipe -o /build/source/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/node-v16.15.0/out/Release/obj.target/v8_co>
  arm-openwrt-linux-muslgnueabi-g++ -Os -pipe -o /build/source/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/node-v16.15.0/out/Release/obj.target/v8_co>
  arm-openwrt-linux-muslgnueabi-g++ -Os -pipe -o /build/source/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/node-v16.15.0/out/Release/obj.target/v8_co>
  arm-openwrt-linux-muslgnueabi-g++ -Os -pipe -o /build/source/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/node-v16.15.0/out/Release/obj.target/v8_co>
  touch /build/source/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/node-v16.15.0/out/Release/obj.host/tools/v8_gypfiles/postmortem-metadata.stamp
rm -f /build/source/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/node-v16.15.0/out/Release/obj.target/tools/v8_gypfiles/libv8_init.a libv8_init.ar-fil>
printf "%s\n" >>/build/source/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/node-v16.15.0/out/Release/obj.target/tools/v8_gypfiles/libv8_init.a.libv8_i>
arm-openwrt-linux-muslgnueabi-gcc-ar crsT /build/source/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/node-v16.15.0/out/Release/obj.target/tools/v8_gyp>
  LD_LIBRARY_PATH=/build/source/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/node-v16.15.0/out/Release/lib.host:/build/source/build_dir/target-arm_cor>
  touch /build/source/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/node-v16.15.0/out/Release/obj.host/tools/v8_gypfiles/torque_generated_definitions.s>
  touch /build/source/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/node-v16.15.0/out/Release/obj.target/tools/v8_gypfiles/postmortem-metadata.stamp
  touch /build/source/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/node-v16.15.0/out/Release/obj.target/tools/v8_gypfiles/torque_generated_definitions>
rm -f /build/source/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/node-v16.15.0/out/Release/obj.host/tools/v8_gypfiles/libv8_compiler.a libv8_compiler.>
printf "%s\n" >>/build/source/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/node-v16.15.0/out/Release/obj.host/tools/v8_gypfiles/libv8_compiler.a.libv8>
printf "%s\n" >>/build/source/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/node-v16.15.0/out/Release/obj.host/tools/v8_gypfiles/libv8_compiler.a.libv8>
ar crsT /build/source/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/node-v16.15.0/out/Release/obj.host/tools/v8_gypfiles/libv8_compiler.a @/build/sourc>
rm -f /build/source/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/node-v16.15.0/out/Release/obj.target/tools/v8_gypfiles/libv8_compiler.a libv8_compile>
printf "%s\n" >>/build/source/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/node-v16.15.0/out/Release/obj.target/tools/v8_gypfiles/libv8_compiler.a.lib>
printf "%s\n" >>/build/source/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/node-v16.15.0/out/Release/obj.target/tools/v8_gypfiles/libv8_compiler.a.lib>
arm-openwrt-linux-muslgnueabi-gcc-ar crsT /build/source/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/node-v16.15.0/out/Release/obj.target/tools/v8_gyp>
  arm-openwrt-linux-muslgnueabi-g++ -Os -pipe -o /build/source/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/node-v16.15.0/out/Release/obj.target/libno>
  arm-openwrt-linux-muslgnueabi-g++ -Os -pipe -o /build/source/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/node-v16.15.0/out/Release/obj.target/libno>
  arm-openwrt-linux-muslgnueabi-g++ -Os -pipe -o /build/source/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/node-v16.15.0/out/Release/obj.target/libno>
  arm-openwrt-linux-muslgnueabi-g++ -Os -pipe -o /build/source/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/node-v16.15.0/out/Release/obj.target/libno>
  arm-openwrt-linux-muslgnueabi-g++ -Os -pipe -o /build/source/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/node-v16.15.0/out/Release/obj.target/libno>
  arm-openwrt-linux-muslgnueabi-g++ -Os -pipe -o /build/source/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/node-v16.15.0/out/Release/obj.target/libno>
  arm-openwrt-linux-muslgnueabi-g++ -Os -pipe -o /build/source/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/node-v16.15.0/out/Release/obj.target/libno>
  arm-openwrt-linux-muslgnueabi-g++ -Os -pipe -o /build/source/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/node-v16.15.0/out/Release/obj.target/libno>
  arm-openwrt-linux-muslgnueabi-g++ -Os -pipe -o /build/source/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/node-v16.15.0/out/Release/obj.target/libno>
  arm-openwrt-linux-muslgnueabi-g++ -Os -pipe -o /build/source/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/node-v16.15.0/out/Release/obj.target/libno>
In file included from ../src/cares_wrap.cc:25:
../src/cares_wrap.h:25:11: fatal error: ares_nameser.h: No such file or directory
 # include <ares_nameser.h>
           ^~~~~~~~~~~~~~~~
compilation terminated.

```

p3x-robot, 2022-05-29:
https://github.com/nodejs/node/issues/39747

nxhack, 2022-05-30:
Hi @p3x-robot 

I think you are probably talking about node v16.x with 21.02.
The c-ares package has been modified in the latest 21.02, but I'll verify with 21.02.3 (Releases).

p3x-robot, 2022-05-31:
what does itr mwan c-areas? i have to include something or download or how? thank!