nxhack/openwrt-node-packages #1312: Missing dependency: ngtcp2/version.h: No such file or directory

LGA1150, 2022-06-14:
```
../src/node_metadata.cc:17:10: fatal error: ngtcp2/version.h: No such file or directory
   17 | #include <ngtcp2/version.h>
      |          ^~~~~~~~~~~~~~~~~~
compilation terminated.
```
`ngtcp2` is missing in OpenWrt package feeds.

nxhack, 2022-06-14:
OpenWrt version?
openwrt-node-packages branch?
node.js version?

LGA1150, 2022-06-14:
OpenWrt master a7be143646
openwrt-node-packages master 857cac39978
v16.15.1

nxhack, 2022-06-14:
Thanks.
I will try to reproduce it.
It will take some time.

LGA1150, 2022-06-14:
```
 cat ./build_dir/target-aarch64_generic_musl/node-v16.15.1/config.gypi
# Do not edit. Generated by the configure script.
{ 'target_defaults': { 'cflags': [],
                       'default_configuration': 'Release',
                       'defines': [],
                       'include_dirs': [ '/home/lga/openwrt/staging_dir/target-aarch64_generic_musl/usr/include',
                                         '/home/lga/openwrt/staging_dir/target-aarch64_generic_musl/usr/include',
                                         '/home/lga/openwrt/staging_dir/target-aarch64_generic_musl/usr/include',
                                         '/home/lga/openwrt/staging_dir/target-aarch64_generic_musl/usr/include',
                                         '/home/lga/openwrt/staging_dir/target-aarch64_generic_musl/usr/include'],
                       'libraries': [ '-L/home/lga/openwrt/staging_dir/target-aarch64_generic_musl/usr/lib',
                                      '-lz',
                                      '-L/home/lga/openwrt/staging_dir/target-aarch64_generic_musl/usr/lib',
                                      '-luv',
                                      '-L/home/lga/openwrt/staging_dir/target-aarch64_generic_musl/usr/lib',
                                      '-lcares',
                                      '-L/home/lga/openwrt/staging_dir/target-aarch64_generic_musl/usr/lib',
                                      '-lnghttp2',
                                      '-L/home/lga/openwrt/staging_dir/target-aarch64_generic_musl/usr/lib',
                                      '-lcrypto',
                                      '-lssl']},
  'variables': { 'asan': 0,
                 'coverage': 'false',
                 'dcheck_always_on': 0,
                 'debug_nghttp2': 'false',
                 'debug_node': 'false',
                 'enable_lto': 'false',
                 'enable_pgo_generate': 'false',
                 'enable_pgo_use': 'false',
                 'error_on_warn': 'false',
                 'force_dynamic_crt': 0,
                 'host_arch': 'x64',
                 'icu_data_in': '../../deps/icu-tmp/icudt70l.dat',
                 'icu_default_data': '',
                 'icu_endianness': 'l',
                 'icu_gyp_path': 'tools/icu/icu-generic.gyp',
                 'icu_locales': 'en,root',
                 'icu_path': 'deps/icu-small',
                 'icu_small': 'true',
                 'icu_ver_major': '70',
                 'is_debug': 0,
                 'llvm_version': '0.0',
                 'napi_build_version': '8',
                 'node_byteorder': 'little',
                 'node_debug_lib': 'false',
                 'node_enable_d8': 'false',
                 'node_install_corepack': 'true',
                 'node_install_npm': 'true',
                 'node_library_files': [ 'lib/_http_common.js',
                                         'lib/_stream_duplex.js',
                                         'lib/_http_outgoing.js',
                                         'lib/_http_client.js',
                                         'lib/buffer.js',
                                         'lib/http.js',
                                         'lib/os.js',
                                         'lib/child_process.js',
                                         'lib/_stream_passthrough.js',
                                         'lib/path.js',
                                         'lib/sys.js',
                                         'lib/repl.js',
                                         'lib/_stream_readable.js',
                                         'lib/string_decoder.js',
                                         'lib/net.js',
                                         'lib/diagnostics_channel.js',
                                         'lib/events.js',
                                         'lib/tls.js',
                                         'lib/async_hooks.js',
                                         'lib/_http_agent.js',
                                         'lib/dns.js',
[... 273 lines cut ...]
                 'node_with_ltcg': 'false',
                 'node_without_node_options': 'false',
                 'openssl_fips': '',
                 'openssl_is_fips': 'false',
                 'openssl_quic': 'false',
                 'ossfuzz': 'false',
                 'shlib_suffix': 'so.93',
                 'target_arch': 'arm64',
                 'v8_enable_31bit_smis_on_64bit_arch': 0,
                 'v8_enable_gdbjit': 0,
                 'v8_enable_hugepage': 0,
                 'v8_enable_i18n_support': 1,
                 'v8_enable_inspector': 1,
                 'v8_enable_lite_mode': 0,
                 'v8_enable_object_print': 1,
                 'v8_enable_pointer_compression': 0,
                 'v8_enable_webassembly': 1,
                 'v8_no_strict_aliasing': 1,
                 'v8_optimized_debug': 1,
                 'v8_promise_internal_field_count': 1,
                 'v8_random_seed': 0,
                 'v8_trace_maps': 0,
                 'v8_use_siphash': 1,
                 'want_separate_host_toolset': 1}}
```

nxhack, 2022-06-14:
There seems to be no difference...
```
                 'node_module_version': 93,
                 'node_no_browser_globals': 'false',
                 'node_prefix': '/usr',
                 'node_release_urlbase': '',
                 'node_section_ordering_info': '',
                 'node_shared': 'false',
                 'node_shared_brotli': 'false',
                 'node_shared_cares': 'true',
                 'node_shared_http_parser': 'false',
                 'node_shared_libuv': 'true',
                 'node_shared_nghttp2': 'true',
                 'node_shared_nghttp3': 'false',
                 'node_shared_ngtcp2': 'false',
                 'node_shared_openssl': 'true',
                 'node_shared_zlib': 'true',

```
node_metadata.cc
```
#ifdef OPENSSL_INFO_QUIC
#include <ngtcp2/version.h>
#include <nghttp3/version.h>
#endif
```
The problem seems to be whether OPENSSL_INFO_QUIC is defined or not. 
Please give me more time.

nxhack, 2022-06-14:
I guess I could work around it by adding include_dirs, but that is out of my scope.
```
./node-v16.15.1/deps/ngtcp2/ngtcp2/lib/includes
```

closed 2023-05-09 by nxhack