nxhack/openwrt-node-packages #1378: [21.02] Request for removing node 12, which is not supported

BKPepe, 2022-10-05:
Hi @nxhack,

You are doing excellent work for node packages. Thanks!

Despite that, I have one small thing. 

Since OpenWrt 21.02, there is not required to have Python 2.7 is installed on systems. We removed it from Turris infrastructure when we are building OpenWrt, but in some rare situations, it happens that the build system tries to install Node 12.22.12 as host package, and because of that, node packages are not compiled.

The log output:
```
make[2]: Entering directory '<turris1x-build/ws/build/feeds/node/node'>
mkdir -p <turris1x-build/ws/build/dl>
SHELL= flock <turris1x-build/ws/build/tmp/.node-v12.22.12.tar.xz.flock> -c '  	<turris1x-build/ws/build/scripts/download.pl> "<turris1x-build/ws/build/dl"> "node-v12.22.12.tar.xz" "bc42b7f8495b9bfc7f7850dd180bb02a5bdf139cc232b8c6f02a6967e20714f2" "" "https://nodejs.org/dist/v12.22.12"    '
(cd <turris1x-build/ws/build/build_dir/hostpkg/node-v12.22.12/;> if [ -x configure ]; then cp -fpR <turris1x-build/ws/build/scripts/config.{guess,sub}> <turris1x-build/ws/build/build_dir/hostpkg/node-v12.22.12//> &&   bash ./configure --dest-os=linux --with-intl=small-icu --prefix=<turris1x-build/ws/build/staging_dir/hostpkg> ; fi )
Please use Python 2.7
make[2]: *** [Makefile:199: <turris1x-build/ws/build/build_dir/hostpkg/node-v12.22.12/.configured]> Error 1
make[2]: Leaving directory '<turris1x-build/ws/build/feeds/node/node'>
time: package/feeds/node/node/host-compile#0.45#0.16#0.53
    ERROR: package/feeds/node/node [host] failed to build.
make[1]: *** [package/Makefile:114: package/feeds/node/node/host/compile] Error 1
```

Would it be possible to remove node 12, which is anyway not supported anymore?

nxhack, 2022-10-05:
Yes, indeed. Removing it. (In addition, I will try to support v18.)

nxhack, 2022-10-08:
Applied.
https://github.com/nxhack/openwrt-node-packages/commit/efc3d7497846c687f18b40234aef32e6875fdcd1

BKPepe, 2022-10-09:
Thank you so much!

closed 2022-10-09 by BKPepe