nxhack/openwrt-node-packages #1459: node-bufferutil-4.0.7/build/Release/bufferutil.node: No such file or directory

mickzuev, 2023-01-30:
Hi,
I have an error:

```
time: package/feeds/node/node/compile#3.37#0.89#3.11
make[3]: Entering directory '/root/openwrt/feeds/node/node'
mkdir -p /root/openwrt/dl
SHELL= flock /root/openwrt/tmp/.node-v14.21.2.tar.xz.flock -c '         /root/openwrt/scripts/download.pl "/root/openwrt/dl" "node-v14.21.2.tar.xz" "d8f09a0f16773a77613c3817606f6d455624992d9c43443aca15e91807a1ff03" "" "https://nodejs.org/dist/v14.21.2"    '
make[3]: Leaving directory '/root/openwrt/feeds/node/node'
time: package/feeds/node/node/host-compile#3.04#0.81#2.73
make[3]: Entering directory '/root/openwrt/feeds/node/node-bufferutil'
mkdir -p /root/openwrt/dl
SHELL= flock /root/openwrt/tmp/.bufferutil-4.0.7.tgz.flock -c '         /root/openwrt/scripts/download.pl "/root/openwrt/dl" "bufferutil-4.0.7.tgz" "b618070de7ae0ae68e33a81d64f2b064431b15bbdd7eb5560b0e8629e14f43fa" "" "https://registry.npmjs.org/bufferutil/-/"    '
rm -rf /root/openwrt/build_dir/target-arm_cortex-a9+neon_musl_eabi/node-bufferutil-4.0.7/.pkgdir/node-bufferutil.installed /root/openwrt/build_dir/target-arm_cortex-a9+neon_musl_eabi/node-bufferutil-4.0.7/.pkgdir/node-bufferutil
mkdir -p /root/openwrt/build_dir/target-arm_cortex-a9+neon_musl_eabi/node-bufferutil-4.0.7/.pkgdir/node-bufferutil
install -d -m0755 /root/openwrt/build_dir/target-arm_cortex-a9+neon_musl_eabi/node-bufferutil-4.0.7/.pkgdir/node-bufferutil/usr/lib/node/bufferutil
cp -fpR /root/openwrt/build_dir/target-arm_cortex-a9+neon_musl_eabi/node-bufferutil-4.0.7/{package.json,*.md} /root/openwrt/build_dir/target-arm_cortex-a9+neon_musl_eabi/node-bufferutil-4.0.7/.pkgdir/node-bufferutil/usr/lib/node/bufferutil/
cp -fpR /root/openwrt/build_dir/target-arm_cortex-a9+neon_musl_eabi/node-bufferutil-4.0.7/{LICENSE,*.js} /root/openwrt/build_dir/target-arm_cortex-a9+neon_musl_eabi/node-bufferutil-4.0.7/.pkgdir/node-bufferutil/usr/lib/node/bufferutil/
cp -fpR /root/openwrt/build_dir/target-arm_cortex-a9+neon_musl_eabi/node-bufferutil-4.0.7/node_modules /root/openwrt/build_dir/target-arm_cortex-a9+neon_musl_eabi/node-bufferutil-4.0.7/.pkgdir/node-bufferutil/usr/lib/node/bufferutil/
install -d -m0755 /root/openwrt/build_dir/target-arm_cortex-a9+neon_musl_eabi/node-bufferutil-4.0.7/.pkgdir/node-bufferutil/usr/lib/node/bufferutil/build/Release
cp -fpR /root/openwrt/build_dir/target-arm_cortex-a9+neon_musl_eabi/node-bufferutil-4.0.7/build/Release/bufferutil.node /root/openwrt/build_dir/target-arm_cortex-a9+neon_musl_eabi/node-bufferutil-4.0.7/.pkgdir/node-bufferutil/usr/lib/node/bufferutil/build/Release/
cp: cannot stat '/root/openwrt/build_dir/target-arm_cortex-a9+neon_musl_eabi/node-bufferutil-4.0.7/build/Release/bufferutil.node': No such file or directory
make[3]: *** [Makefile:90: /root/openwrt/build_dir/target-arm_cortex-a9+neon_musl_eabi/node-bufferutil-4.0.7/.pkgdir/node-bufferutil.installed] Error 1
make[3]: Leaving directory '/root/openwrt/feeds/node/node-bufferutil'
time: package/feeds/node/node-bufferutil/compile#1.08#0.19#1.12
    ERROR: package/feeds/node/node-bufferutil failed to build.
make[2]: *** [package/Makefile:116: package/feeds/node/node-bufferutil/compile] Error 1
make[2]: Leaving directory '/root/openwrt'
make[1]: *** [package/Makefile:110: /root/openwrt/staging_dir/target-arm_cortex-a9+neon_musl_eabi/stamp/.package_compile] Error 2
make[1]: Leaving directory '/root/openwrt'
make: *** [/root/openwrt/include/toplevel.mk:230: world] Error 2
```
Ubuntu 22.04.1 LTS
```
git clone https://git.openwrt.org/openwrt/openwrt.git
cd openwrt
git checkout openwrt-21.02
git checkout v21.02.5
git pull
echo 'src-git node https://github.com/nxhack/openwrt-node-packages.git;openwrt-21.02' >> feeds.conf.default
./scripts/feeds update -a -i
./scripts/feeds install -a -f
./scripts/feeds update node
rm ./package/feeds/packages/node
rm ./package/feeds/packages/node-*
./scripts/feeds install -a -p node

make menuconfig
CONFIG_TARGET_imx6=y
CONFIG_TARGET_MULTI_PROFILE=y
languages -> node.js -> node-zigbee2mqtt
save -> exit

make -j1 V=s
```

nxhack, 2023-01-31:
Hi @mickzuev 

Tested with the same settings.

OpenWrt: v21.02.5
Node.js: v14.21.2
TARGET: imx6

bufferutil built without problems.

Please clean it once.
```
make V=s package/feeds/node/node-bufferutil/{clean,compile}
```

mickzuev, 2023-01-31:
```
npm timing action:install Completed in 1ms
npm info lifecycle node-gyp-build@4.6.0~postinstall: node-gyp-build@4.6.0
npm timing action:postinstall Completed in 0ms
npm timing stage:executeActions Completed in 84ms
npm timing stage:rollbackFailedOptional Completed in 0ms
npm info linkStuff bufferutil@4.0.7
npm info lifecycle bufferutil@4.0.7~install: bufferutil@4.0.7
npm WARN lifecycle bufferutil@4.0.7~install: cannot run in wd bufferutil@4.0.7 node-gyp-build (wd=/root/openwrt/build_dir/target-arm_cortex-a9+neon_musl_eabi/node-bufferutil-4.0.7)
npm info lifecycle bufferutil@4.0.7~postinstall: bufferutil@4.0.7
npm timing stage:runTopLevelLifecycles Completed in 6476ms
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: fsevents@~2.3.2 (node_modules/mocha/node_modules/chokidar/node_modules/fsevents):
npm WARN notsup SKIPPING OPTIONAL DEPENDENCY: Unsupported platform for fsevents@2.3.2: wanted {"os":"darwin","arch":"any"} (current: {"os":"linux","arch":"x64"})

npm timing audit submit Completed in 345ms
npm http fetch POST 200 https://registry.npmjs.org/-/npm/v1/security/audits/quick 345ms
npm timing audit body Completed in 1ms
added 1 package from 1 contributor and audited 202 packages in 6.739s
found 0 vulnerabilities

npm timing npm Completed in 7715ms
npm info ok
rm -rf /root/openwrt/tmp/npm-tmp-oiIDUHiRfJ
rm -rf /root/openwrt/tmp/npm-cache-oiIDUHiRfJ
rm -f /root/openwrt/build_dir/target-arm_cortex-a9+neon_musl_eabi/node-bufferutil-4.0.7/node_modules/.package-lock.json
find /root/openwrt/build_dir/target-arm_cortex-a9+neon_musl_eabi/node-bufferutil-4.0.7/node_modules -type d -empty -print0 | xargs -0 rmdir || true
rmdir: missing operand
Try 'rmdir --help' for more information.
touch /root/openwrt/build_dir/target-arm_cortex-a9+neon_musl_eabi/node-bufferutil-4.0.7/.built
rm -rf /root/openwrt/build_dir/target-arm_cortex-a9+neon_musl_eabi/node-bufferutil-4.0.7/.pkgdir/node-bufferutil.installed /root/openwrt/build_dir/target-arm_cortex-a9+neon_musl_eabi/node-bufferutil-4.0.7/.pkgdir/node-bufferutil
mkdir -p /root/openwrt/build_dir/target-arm_cortex-a9+neon_musl_eabi/node-bufferutil-4.0.7/.pkgdir/node-bufferutil
install -d -m0755 /root/openwrt/build_dir/target-arm_cortex-a9+neon_musl_eabi/node-bufferutil-4.0.7/.pkgdir/node-bufferutil/usr/lib/node/bufferutil
cp -fpR /root/openwrt/build_dir/target-arm_cortex-a9+neon_musl_eabi/node-bufferutil-4.0.7/{package.json,*.md} /root/openwrt/build_dir/target-arm_cortex-a9+neon_musl_eabi/node-bufferutil-4.0.7/.pkgdir/node-bufferutil/usr/lib/node/bufferutil/
cp -fpR /root/openwrt/build_dir/target-arm_cortex-a9+neon_musl_eabi/node-bufferutil-4.0.7/{LICENSE,*.js} /root/openwrt/build_dir/target-arm_cortex-a9+neon_musl_eabi/node-bufferutil-4.0.7/.pkgdir/node-bufferutil/usr/lib/node/bufferutil/
cp -fpR /root/openwrt/build_dir/target-arm_cortex-a9+neon_musl_eabi/node-bufferutil-4.0.7/node_modules /root/openwrt/build_dir/target-arm_cortex-a9+neon_musl_eabi/node-bufferutil-4.0.7/.pkgdir/node-bufferutil/usr/lib/node/bufferutil/
install -d -m0755 /root/openwrt/build_dir/target-arm_cortex-a9+neon_musl_eabi/node-bufferutil-4.0.7/.pkgdir/node-bufferutil/usr/lib/node/bufferutil/build/Release
cp -fpR /root/openwrt/build_dir/target-arm_cortex-a9+neon_musl_eabi/node-bufferutil-4.0.7/build/Release/bufferutil.node /root/openwrt/build_dir/target-arm_cortex-a9+neon_musl_eabi/node-bufferutil-4.0.7/.pkgdir/node-bufferutil/usr/lib/node/bufferutil/build/Release/
cp: cannot stat '/root/openwrt/build_dir/target-arm_cortex-a9+neon_musl_eabi/node-bufferutil-4.0.7/build/Release/bufferutil.node': No such file or directory
make[2]: *** [Makefile:90: /root/openwrt/build_dir/target-arm_cortex-a9+neon_musl_eabi/node-bufferutil-4.0.7/.pkgdir/node-bufferutil.installed] Error 1
make[2]: Leaving directory '/root/openwrt/feeds/node/node-bufferutil'
time: package/feeds/node/node-bufferutil/compile#5.69#1.84#9.41
    ERROR: package/feeds/node/node-bufferutil failed to build.
make[1]: *** [package/Makefile:116: package/feeds/node/node-bufferutil/compile] Error 1
make[1]: Leaving directory '/root/openwrt'
make: *** [/root/openwrt/include/toplevel.mk:230: package/feeds/node/node-bufferutil/compile] Error 2
```

mickzuev, 2023-01-31:
On a different path, not a root user, with 'feeds update' twice
```
cd openwrt
./scripts/feeds update -a
./scripts/feeds install -a
git branch -a
git checkout openwrt-21.02
git tag
git checkout v21.02.5
git pull
echo 'src-git node https://github.com/nxhack/openwrt-node-packages.git;openwrt-21.02' >> feeds.conf.default
./scripts/feeds update -a -i
./scripts/feeds install -a -f
./scripts/feeds update node
rm ./package/feeds/packages/node
rm ./package/feeds/packages/node-*
./scripts/feeds install -a -p node
```
I have:
```
make[3]: Entering directory '/home/mz/openwrt/package/kernel/linux'
mkdir -p /home/mz/openwrt/staging_dir/target-arm_cortex-a9+neon_musl_eabi/root-imx6/stamp

SHELL= flock /home/mz/openwrt/tmp/.root-copy.flock -c 'cp -fpR /home/mz/openwrt/build_dir/target-arm_cortex-a9+neon_musl_eabi/linux-imx6/packages/.pkgdir/kernel/. /home/mz/openwrt/staging_dir/target-arm_cortex-a9+neon_musl_eabi/root-imx6/'
touch /home/mz/openwrt/staging_dir/target-arm_cortex-a9+neon_musl_eabi/root-imx6/stamp/.kernel_installed
echo "kernel" >> /home/mz/openwrt/staging_dir/target-arm_cortex-a9+neon_musl_eabi/pkginfo/linux.default.install
make[3]: Leaving directory '/home/mz/openwrt/package/kernel/linux'
time: package/kernel/linux/compile#1.58#0.10#1.63
make[2]: Leaving directory '/home/mz/openwrt'
make[2]: Entering directory '/home/mz/openwrt'
rm -rf /home/mz/openwrt/staging_dir/packages/imx6
mkdir -p /home/mz/openwrt/staging_dir/packages/imx6
ln -s /home/mz/openwrt/bin/targets/imx6/generic/packages/*.ipk /home/mz/openwrt/staging_dir/packages/imx6; ln -s /home/mz/openwrt/bin/packages/arm_cortex-a9_neon/base/*.ipk /home/mz/openwrt/staging_dir/packages/imx6;  ln -s /home/mz/openwrt/bin/packages/arm_cortex-a9_neon/node/*.ipk /home/mz/openwrt/staging_dir/packages/imx6; ln -s /home/mz/openwrt/bin/packages/arm_cortex-a9_neon/packages/*.ipk /home/mz/openwrt/staging_dir/packages/imx6;
find /home/mz/openwrt/staging_dir/target-arm_cortex-a9+neon_musl_eabi/root-imx6 -type d | xargs -r chmod 0755
rm -rf /home/mz/openwrt/build_dir/target-arm_cortex-a9+neon_musl_eabi/root-imx6 /home/mz/openwrt/build_dir/target-arm_cortex-a9+neon_musl_eabi/root.orig-imx6
mkdir -p /home/mz/openwrt/build_dir/target-arm_cortex-a9+neon_musl_eabi/root-imx6/tmp
IPKG_NO_SCRIPT=1 IPKG_INSTROOT=/home/mz/openwrt/build_dir/target-arm_cortex-a9+neon_musl_eabi/root-imx6 TMPDIR=/home/mz/openwrt/build_dir/target-arm_cortex-a9+neon_musl_eabi/root-imx6/tmp /home/mz/openwrt/staging_dir/host/bin/opkg --offline-root /home/mz/openwrt/build_dir/target-arm_cortex-a9+neon_musl_eabi/root-imx6 --force-postinstall --add-dest root:/ --add-arch all:100 --add-arch arm_cortex-a9_neon:200 install $(cat /home/mz/openwrt/tmp/opkg_install_list)
Unknown package 'kernel'.
Collected errors:
 * pkg_hash_check_unresolved: cannot find dependency libc for kernel
 * pkg_hash_fetch_best_installation_candidate: Packages for kernel found, but incompatible with the architectures configured
 * opkg_install_cmd: Cannot install package kernel.
make[2]: *** [package/Makefile:70: package/install] Error 255
make[2]: Leaving directory '/home/mz/openwrt'
make[1]: *** [package/Makefile:111: /home/mz/openwrt/staging_dir/target-arm_cortex-a9+neon_musl_eabi/stamp/.package_install] Error 2
make[1]: Leaving directory '/home/mz/openwrt'
make: *** [/home/mz/openwrt/include/toplevel.mk:230: world] Error 2
mz@server:~/openwrt$
```
but I need latest node-zigbee2mqtt only
`opkg install node-zigbee2mqtt_1.29.2-1_arm_cortex-a9_neon.ipk` -- has been working fine for 1 hour now

But what am i doing wrong anyway (with make openwrt)?

nxhack, 2023-01-31:
```
cd openwrt
git pull
git checkout openwrt-21.02
git checkout v21.02.5
echo 'src-git node https://github.com/nxhack/openwrt-node-packages.git;openwrt-21.02' >> feeds.conf.default
./scripts/feeds update -a -i
./scripts/feeds install -a -f
./scripts/feeds update node
rm ./package/feeds/packages/node
rm ./package/feeds/packages/node-*
./scripts/feeds install -a -p node
```
**Execute this command after this**
```
make defconfig
```
After this, you can select a package.
```
make menuconfig
```
Now, let's build.
```
make
```

nxhack, 2023-01-31:
> but I need latest node-zigbee2mqtt only

In this case, it may be better to use Image Builder.
https://openwrt.org/docs/guide-user/additional-software/imagebuilder

mickzuev, 2023-01-31:
both way:
```
root@server:~/openwrt# make defconfig
#
# No change to .config
#
root@server:~/openwrt#
```
```
mz@server:~/openwrt$  make defconfig
#
# No change to .config
#
mz@server:~/openwrt$
```

nxhack, 2023-01-31:
https://openwrt.org/docs/guide-developer/toolchain/use-buildsystem
 > 
    Do everything as an unprivileged user, not root, without sudo.
    Make sure there are no spaces in the full path to the build directory.

mickzuev, 2023-01-31:
Yes, my second way :) - 'not root, without sudo, no spaces', feeds update/install before and after checkout/tag - but with 'Unknown package 'kernel''

nxhack, 2023-01-31:
Do you have a packages here?
```
cd ./bin/packages/arm_cortex-a9_neon/node
ls
```
```
node-bufferutil_4.0.7-1_arm_cortex-a9_neon.ipk
node-utf-8-validate_6.0.2-1_arm_cortex-a9_neon.ipk
node-zigbee2mqtt_1.29.2-1_arm_cortex-a9_neon.ipk
```
If the packages have been built, you can copy these three packages to the target machine and install them with okpg

nxhack, 2023-01-31:
Ah, so the package was already installed. My apologies.

nxhack, 2023-01-31:
```
$ make
make[2]: Entering directory '/mnt/openwrt-21.02/openwrt/scripts/config'
make[2]: 'conf' is up to date.
make[2]: Leaving directory '/mnt/openwrt-21.02/openwrt/scripts/config'
time: target/linux/prereq#0.18#0.05#0.37
 make[1] world
 make[2] target/compile
 make[3] -C target/linux compile
 make[2] buildinfo
 make[3] diffconfig buildversion feedsversion
 make[2] package/cleanup
 make[2] package/compile
 make[3] -C package/libs/libjson-c host-compile
 make[3] -C package/libs/libubox host-compile
 make[3] -C package/system/opkg host-compile
 make[3] -C package/libs/toolchain compile
 make[3] -C feeds/packages/libs/nghttp2 compile
 make[3] -C feeds/packages/libs/expat host-compile
 make[3] -C feeds/packages/libs/libffi host-compile
 make[3] -C package/utils/bzip2 host-compile
 make[3] -C feeds/packages/lang/python/python3 host-compile
 make[3] -C package/libs/openssl compile
 make[3] -C package/libs/zlib compile
 make[3] -C feeds/node/node compile
 make[3] -C feeds/node/node host-compile
 make[3] -C feeds/node/node-bufferutil compile
 make[3] -C feeds/node/node-utf-8-validate compile
 make[3] -C feeds/node/node-clean-modules host-compile
 make[3] -C feeds/node/node-zigbee2mqtt compile
 make[3] -C package/firmware/linux-firmware compile
 make[3] -C package/firmware/prism54-firmware compile
 make[3] -C package/kernel/linux compile
 make[2] package/install
 make[2] target/install
 make[3] -C target/linux install
 make[2] package/index
 make[2] json_overview_image_info
 make[2] checksum
```
```
$ ls -l ./bin/targets/imx6/generic/
total 17844
-rw-r--r-- 1 nxhack nxhack      776 Jan 31 19:24 config.buildinfo
-rw-r--r-- 1 nxhack nxhack      438 Jan 31 19:24 feeds.buildinfo
-rw-r--r-- 1 nxhack nxhack      296 Jan 31 19:26 openwrt-imx6.manifest
-rw-r--r-- 1 nxhack nxhack 18246342 Jan 31 19:26 openwrt-imx6-rootfs.tar.gz
drwxr-xr-x 2 nxhack nxhack     4096 Jan 31 19:16 packages
-rw-r--r-- 1 nxhack nxhack      430 Jan 31 19:26 sha256sums
-rw-r--r-- 1 nxhack nxhack       18 Jan 31 19:24 version.buildinfo
```

mickzuev, 2023-01-31:
```
mz@server:~/openwrt$ ls -l ./bin/targets/imx6/generic/
total 16
-rw-r--r-- 1 mz mz  712 янв 31 11:07 config.buildinfo
-rw-r--r-- 1 mz mz  361 янв 31 11:07 feeds.buildinfo
drwxr-xr-x 2 mz mz 4096 янв 31 04:50 packages
-rw-r--r-- 1 mz mz   18 янв 31 11:07 version.buildinfo
mz@server:~/openwrt$ ls -l ./bin/targets/imx6/generic/packages/
total 528
-rw-r--r-- 1 mz mz    821 янв 31 00:09 kernel_5.4.215-1-5e1afed82e767f67b3baf09c8fb32f8f_arm_cortex-a9_neon.ipk
-rw-r--r-- 1 mz mz   5709 янв 31 04:50 libatomic1_8.4.0-3_arm_cortex-a9_neon.ipk
-rw-r--r-- 1 mz mz 242551 янв 31 04:50 libc_1.1.24-3_arm_cortex-a9_neon.ipk
-rw-r--r-- 1 mz mz  20177 янв 31 04:50 libgcc1_8.4.0-3_arm_cortex-a9_neon.ipk
-rw-r--r-- 1 mz mz 258122 янв 31 04:50 libstdcpp6_8.4.0-3_arm_cortex-a9_neon.ipk
mz@server:~/openwrt$
mz@server:~/openwrt$ /home/mz/openwrt/staging_dir/host/bin/opkg --offline-root /home/mz/openwrt/build_dir/target-arm_cortex-a9+neon_musl_eabi/root-imx6 --force-postinstall --add-dest root:/ --add-arch all:100 --add-arch arm_cortex-a9_neon:200 install /home/mz/openwrt/bin/targets/imx6/generic/packages/kernel_5.4.215-1-5e1afed82e767f67b3baf09c8fb32f8f_arm_cortex-a9_neon.ipk --verbosity=4
opkg_install_cmd: /home/mz/openwrt/bin/targets/imx6/generic/packages/kernel_5.4.215-1-5e1afed82e767f67b3baf09c8fb32f8f_arm_cortex-a9_neon.ipk
pkg_init_from_file: applying abpkg flag to /home/mz/openwrt/bin/targets/imx6/generic/packages/kernel_5.4.215-1-5e1afed82e767f67b3baf09c8fb32f8f_arm_cortex-a9_neon.ipk
opkg_prepare_url_for_install: Package kernel provided by hand (/home/mz/openwrt/bin/targets/imx6/generic/packages/kernel_5.4.215-1-5e1afed82e767f67b3baf09c8fb32f8f_arm_cortex-a9_neon.ipk).
flag_related_packages: propagating pkg flag to dependent abpkg libc
pkg_vec_insert_merge: Adding new pkg=kernel version=5.4.215-1 arch=arm_cortex-a9_neon.
pkg_hash_load_feeds:
pkg_hash_load_package_details_helper: found yet incomplete flagged abpkg libc
pkg_hash_load_package_details: Found 1 packages requiring details, reloading feeds
pkg_hash_load_feeds:
pkg_hash_load_package_details_helper: skipping already seen flagged abpkg libc
pkg_hash_load_status_files:
pkg_info_preinstall_check: Updating file owner list.
pkg_hash_fetch_best_installation_candidate: Best installation candidate for kernel:
pkg_hash_fetch_best_installation_candidate: Adding kernel to providers.
pkg_hash_fetch_best_installation_candidate: kernel arch=arm_cortex-a9_neon arch_priority=200 version=5.4.215-1.
pkg_hash_fetch_unsatisfied_dependencies: satisfying_pkg=(nil)
pkg_hash_fetch_unsatisfied_dependencies: satisfying_pkg=(nil)
Unknown package 'kernel'.
opkg_configure_packages: Configuring unpacked packages.
opkg_configure_packages: Reordering packages before configuring them...
write_status_files_if_changed: Nothing to be done.
hash_table: pkg-hash, 24576 bytes
        n_buckets=1024, n_elements=2, n_collisions=0
        max_bucket_len=0, n_used_buckets=2, ave_bucket_len=1.00
        n_hits=3, n_misses=3
hash_table: file-hash, 24576 bytes
        n_buckets=1024, n_elements=0, n_collisions=0
        max_bucket_len=0, n_used_buckets=0, ave_bucket_len=0.00
        n_hits=0, n_misses=0
hash_table: obs-file-hash, 1536 bytes
        n_buckets=64, n_elements=0, n_collisions=0
        max_bucket_len=0, n_used_buckets=0, ave_bucket_len=0.00
        n_hits=0, n_misses=0
Collected errors:
 * pkg_hash_check_unresolved: cannot find dependency libc for kernel
 * pkg_hash_fetch_best_installation_candidate: Packages for kernel found, but incompatible with the architectures configured
 * opkg_install_cmd: Cannot install package kernel.
mz@server:~/openwrt$

```

nxhack, 2023-01-31:
Try initializing the openwrt build system once.

```
make clean
```

mickzuev, 2023-02-03:
v.22.03.03 compiled fine
rm -rf v.21 :slight_smile:

closed 2023-02-09 by mickzuev