nxhack/openwrt-node-packages #1511: A custom opkg repository for node.js is available for limited use.

nxhack, 2023-03-27:
A custom opkg repository for node.js is available for limited use.

Only the latest version of Stable Release is available. (Currently **22.03.3, 21.02.5**)
The target architecture is as follows

**22.03:** 
- aarch64_cortex-a53
- aarch64_cortex-a72
- arm_arm1176jzf-s_vfp
- arm_cortex-a15_neon-vfpv4
- arm_cortex-a5_vfpv4
- arm_cortex-a7_neon-vfpv4
- arm_cortex-a7_vfpv4
- arm_cortex-a8_vfpv3
- arm_cortex-a9_neon
- arm_cortex-a9_vfpv3-d16
- x86_64

**21.02:** 
- aarch64_cortex-a53
- aarch64_cortex-a72
- arm_arm1176jzf-s_vfp
- arm_cortex-a15_neon-vfpv4
- arm_cortex-a5_vfpv4
- arm_cortex-a7_neon-vfpv4
- arm_cortex-a8_vfpv3
- arm_cortex-a9_neon
- arm_cortex-a9_vfpv3-d16
- x86_64

**How to use:**
Add the following to /etc/opkg/distfeeds.conf or /etc/opkg/customfeeds.conf.
```
src/gz openwrt_node http://downloads.nxhack.com/releases/VERSION_NUMBER/packages/TARGET_ARCH/node
```
**VERSION_NUMBER and TARGET_ARCH should be the same as in distfeeds.conf.**

**Example:**
VERSION_NUMBER: 22.03.3
TARGET_ARCH: arm_cortex-a9_vfpv3-d16
```
src/gz openwrt_node http://downloads.nxhack.com/releases/22.03.3/packages/arm_cortex-a9_vfpv3-d16/node
```

alexbrokethis, 2025-02-26:
Is this still available?

nxhack, 2025-02-27:
@alexbrokethis 
Yes, I update the apk every time I build it for testing purposes.

Disclaimer:
I am publishing the results of my test builds, so I may not be able to respond to individual requests.