nxhack/openwrt-node-packages #1518: Linkit smart 7688

tcpipchip, 2023-03-28:
Hi, 
Back after 5 years
Node still working with linkit smart 7688 ?
And with wich openwrt version now ?
Thanks!!!

nxhack, 2023-03-28:
The latest node.js does not support big-endian MIPS.
node.js v12.22.12 will be the last.

Also, mips_24kc has no hardware FPU and uses a software FPU to run node.js, but it is super slow and useless.

If you want to use node.js on OpenWrt, I recommend you use ARM core with hardware FPU.

tcpipchip, 2023-03-29:
thanks @nxhack 
Now i am compiling
https://github.com/nxhack/openwrt-node-packages/tree/dev_18.06

tcpipchip, 2023-03-29:
got this error
/bin/sh: 1: /home/ubuntu/openwrt-18.06.4/build_dir/target-mipsel_24kc_musl/node-v10.24.1/out/Release/torque: Exec format error

nxhack, 2023-03-30:
**openwrt-18.06 is EOL. We do not support it.**
(But ;)
If you build node.js, you need to run 32bit binaries on your host. How to set that up depends on your environment.

tcpipchip, 2023-03-30:
Now I am building the lede now, the node js passed the compile at moment!
Reading your doc, looks that lede with node.js works and finally will be possible install the node-red

tcpipchip, 2023-03-30:
can you help me about this error ?
mkdir -p /home/ubuntu/lede/build_dir/target-mipsel_24kc_musl/default-settings-2/.pkgdir/default-settings
install -d -m0755 /home/ubuntu/lede/build_dir/target-mipsel_24kc_musl/default-settings-2/.pkgdir/default-settings/etc/uci-defaults
install -m0755 ./files/zzz-default-settings /home/ubuntu/lede/build_dir/target-mipsel_24kc_musl/default-settings-2/.pkgdir/default-settings/etc/uci-defaults/99-default-settings
install -d -m0755 /home/ubuntu/lede/build_dir/target-mipsel_24kc_musl/default-settings-2/.pkgdir/default-settings/usr/lib/lua/luci/i18n
po2lmo ./po/zh-cn/default.po /home/ubuntu/lede/build_dir/target-mipsel_24kc_musl/default-settings-2/.pkgdir/default-settings/usr/lib/lua/luci/i18n/default.zh-cn.lmo
bash: po2lmo: command not found
make[3]: *** [Makefile:44: /home/ubuntu/lede/build_dir/target-mipsel_24kc_musl/default-settings-2/.pkgdir/default-settings.installed] Error 127
make[3]: Leaving directory '/home/ubuntu/lede/package/lean/default-settings'
time: package/lean/default-settings/compile#0.17#0.64#0.78
    ERROR: package/lean/default-settings failed to build.
make[2]: *** [package/Makefile:116: package/lean/default-settings/compile] Error 1
make[2]: Leaving directory '/home/ubuntu/lede'
make[1]: *** [package/Makefile:110: /home/ubuntu/lede/staging_dir/target-mipsel_24kc_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/ubuntu/lede'
make: *** [/home/ubuntu/lede/include/toplevel.mk:231: world] Error 2

nxhack, 2023-03-30:
Sorry. It is not in my scope.