nxhack/openwrt-node-packages #1520: Compilation failure of node-ffi-napi for musl/arm platform

kesslerm, 2023-03-29:
Due to a hardcoded `gnu` suffix, the node-ffi-napi package fails to compile for other targets than libc.
This could be resolved by using the `$(TARGET_SUFFIX)` variable, which resolves to `gnu` on platforms that use `glibc` and are not `arm` or `armeb`, but to e.g. `muslgnueabi` on a target platform using the `musl` libc library on `arm` or `armeb`.

```
diff --git a/node-ffi-napi/Makefile b/node-ffi-napi/Makefile
index cad3be7b..b3610937 100644
--- a/node-ffi-napi/Makefile
+++ b/node-ffi-napi/Makefile
@@ -44,7 +44,7 @@ NODEJS_CPU:=$(subst powerpc,ppc,$(subst aarch64,arm64,$(subst x86_64,x64,$(subst
 TMPNPM:=$(shell mktemp -u XXXXXXXXXX)
 NPM_CACHE_DIR:=$(if $(call qstrip,$(CONFIG_NODEJS_NPM_CACHE_DIR)),$(call qstrip,$(CONFIG_NODEJS_NPM_CACHE_DIR)),$(TMP_DIR))

-TARGET_CFLAGS+=$(FPIC) -I$(BUILD_DIR)/libffi-$(PKG_LIBFFI_VERSION)/$(GNU_TARGET_NAME)-gnu
+TARGET_CFLAGS+=$(FPIC) -I$(BUILD_DIR)/libffi-$(PKG_LIBFFI_VERSION)/$(GNU_TARGET_NAME)-$(TARGET_SUFFIX)
 TARGET_CPPFLAGS+=$(FPIC)

 define Build/Compile
```

nxhack, 2023-03-29:
@kesslerm Thank you!

Merge after the test is run.

nxhack, 2023-03-29:
@kesslerm 
I am testing with the combination you mentioned in openwrt, and abs_builddir in libffi seems to be ok with -gnu fixed. If you don't mind, could you give me the detailed conditions?

kesslerm, 2023-03-29:
The issue first occured for me when building [TurrisOS (hbd branch)](https://gitlab.nic.cz/turris/os/build/-/tree/hbl) for the turris omnia. Presumably any build for the `mvebu` target in OpenWRT master would also require this fix.

nxhack, 2023-03-29:
I have merged the turris-omnia config into my existing test environment and tested it but could not reproduce it.

I will try to prepare the turris-omnia environment and do a full build. That will take some time.

nxhack, 2023-03-30:
I was able to reproduce it. The solution is simple, but I will investigate the cause of the suffix change.

nxhack, 2023-03-30:
A quick look,

> $(top_srcdir)/m4/ax_compiler_vendor.m4 \

but I haven't been able to find out why the results of automake are different.

nxhack, 2023-03-30:
The reason for the difference of suffix was due to this patch.

https://gitlab.nic.cz/turris/os/build/-/blob/hbl/patches/openwrt/to-upstream/0100-mpc85xx-hardware-SPE-float-support.patch#L239-253

Workarounds have been merged.