nxhack/openwrt-node-packages #1613: openwrt17014 build error

hzjnet, 2023-06-19:
env:
lsb_release -ra
No LSB modules are available.
Distributor ID: Debian
Description:    Debian GNU/Linux 11 (bullseye)
Release:        11
Codename:       bullseye

gcc --version
gcc (Debian 10.2.1-6) 10.2.1 20210110
Copyright (C) 2020 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

cat package/libs/openssl/Makefile
#
# Copyright (C) 2006-2016 OpenWrt.org
#
# This is free software, licensed under the GNU General Public License v2.
# See /LICENSE for more information.
#

include $(TOPDIR)/rules.mk

PKG_NAME:=openssl
PKG_BASE:=1.1.1
PKG_BUGFIX:=o
PKG_VERSION:=$(PKG_BASE)$(PKG_BUGFIX)
PKG_RELEASE:=1
PKG_USE_MIPS16:=0
ENGINES_DIR=engines-1.1

use source code：
https://github.com/HiGarfield/lede-17.01.4-Mod/commits/master
error log:
cc '-D_REENTRANT' '-DPURIFY' '-DOPENSSL_NO_COMP' '-DOPENSSL_NO_SSL3' '-DOPENSSL_NO_HEARTBEATS' '-DOPENSSL_NO_HW' '-DENGINESDIR="/dev/null"' '-DTERMIOS' '-DOPENSSLDIR="/etc/ssl"' '-DL_ENDIAN' '-DAES_ASM' '-DCPUID_ASM' '-DOPENSSL_BN_ASM_MONT' '-DOPENSSL_CPUID_OBJ' '-DSHA1_ASM' '-DSHA256_ASM' '-DSHA512_ASM' '-DGHASH_ASM' '-DVPAES_ASM' '-DBN_ASM' '-DBF_ASM' '-DBNCO_ASM' '-DDES_ASM' '-DLIB_BN_ASM' '-DMD5_ASM' '-DOPENSSL_BN_ASM' '-DRIP_ASM' '-DRMD160_ASM' '-DWHIRLPOOL_ASM' '-DWP_ASM' '-DOPENSSL_BN_ASM_MONT5' '-DOPENSSL_BN_ASM_GF2m' '-DOPENSSL_IA32_SSE2' '-DBSAES_ASM' '-DDSO_DLFCN' '-DHAVE_DLFCN_H' -I../deps/openssl -I../deps/openssl/openssl -I../deps/openssl/openssl/crypto -I../deps/openssl/openssl/crypto/asn1 -I../deps/openssl/openssl/crypto/evp -I../deps/openssl/openssl/crypto/md2 -I../deps/openssl/openssl/crypto/modes -I../deps/openssl/openssl/crypto/store -I../deps/openssl/openssl/include  -Wno-missing-field-initializers -pthread -Wall -Wextra -Wno-unused-parameter -m64 -O3 -fno-omit-frame-pointer  -MMD -MF /openwrt/lede-17.01.4-Mod/lede-17.01.4-Mod-202306140018/build_dir/target-arm_cortex-a15+neon-vfpv4_musl-1.1.24_eabi/host/node-v8.17.0/out/Release/.deps//openwrt/lede-17.01.4-Mod/lede-17.01.4-Mod-202306140018/build_dir/target-arm_cortex-a15+neon-vfpv4_musl-1.1.24_eabi/host/node-v8.17.0/out/Release/obj.target/openssl/deps/openssl/openssl/crypto/ec/ecp_nistz256.o.d.raw -I/openwrt/lede-17.01.4-Mod/lede-17.01.4-Mod-202306140018/staging_dir/host/include -I/openwrt/lede-17.01.4-Mod/lede-17.01.4-Mod-202306140018/staging_dir/host/usr/include -I/openwrt/lede-17.01.4-Mod/lede-17.01.4-Mod-202306140018/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl-1.1.24_eabi/host/include -O2 -I/openwrt/lede-17.01.4-Mod/lede-17.01.4-Mod-202306140018/staging_dir/host/include -I/openwrt/lede-17.01.4-Mod/lede-17.01.4-Mod-202306140018/staging_dir/host/usr/include -I/openwrt/lede-17.01.4-Mod/lede-17.01.4-Mod-202306140018/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl-1.1.24_eabi/host/include -c -o /openwrt/lede-17.01.4-Mod/lede-17.01.4-Mod-202306140018/build_dir/target-arm_cortex-a15+neon-vfpv4_musl-1.1.24_eabi/host/node-v8.17.0/out/Release/obj.target/openssl/deps/openssl/openssl/crypto/ec/ecp_nistz256.o ../deps/openssl/openssl/crypto/ec/ecp_nistz256.c
../deps/openssl/openssl/crypto/ec/ecp_nistz256.c:749:1: warning: 'static' is not at beginning of declaration [-Wold-style-declaration]
  749 | const static BN_ULONG def_xG[P256_LIMBS] = {
      | ^~~~~
../deps/openssl/openssl/crypto/ec/ecp_nistz256.c:754:1: warning: 'static' is not at beginning of declaration [-Wold-style-declaration]
  754 | const static BN_ULONG def_yG[P256_LIMBS] = {
      | ^~~~~
  cc '-D_REENTRANT' '-DPURIFY' '-DOPENSSL_NO_COMP' '-DOPENSSL_NO_SSL3' '-DOPENSSL_NO_HEARTBEATS' '-DOPENSSL_NO_HW' '-DENGINESDIR="/dev/null"' '-DTERMIOS' '-DOPENSSLDIR="/etc/ssl"' '-DL_ENDIAN' '-DAES_ASM' '-DCPUID_ASM' '-DOPENSSL_BN_ASM_MONT' '-DOPENSSL_CPUID_OBJ' '-DSHA1_ASM' '-DSHA256_ASM' '-DSHA512_ASM' '-DGHASH_ASM' '-DVPAES_ASM' '-DBN_ASM' '-DBF_ASM' '-DBNCO_ASM' '-DDES_ASM' '-DLIB_BN_ASM' '-DMD5_ASM' '-DOPENSSL_BN_ASM' '-DRIP_ASM' '-DRMD160_ASM' '-DWHIRLPOOL_ASM' '-DWP_ASM' '-DOPENSSL_BN_ASM_MONT5' '-DOPENSSL_BN_ASM_GF2m' '-DOPENSSL_IA32_SSE2' '-DBSAES_ASM' '-DDSO_DLFCN' '-DHAVE_DLFCN_H' -I../deps/openssl -I../deps/openssl/openssl -I../deps/openssl/openssl/crypto -I../deps/openssl/openssl/crypto/asn1 -I../deps/openssl/openssl/crypto/evp -I../deps/openssl/openssl/crypto/md2 -I../deps/openssl/openssl/crypto/modes -I../deps/openssl/openssl/crypto/store -I../deps/openssl/openssl/include  -Wno-missing-field-initializers -pthread -Wall -Wextra -Wno-unused-parameter -m64 -O3 -fno-omit-frame-pointer  -MMD -MF /openwrt/lede-17.01.4-Mod/lede-17.01.4-Mod-202306140018/build_dir/target-arm_cortex-a15+neon-vfpv4_musl-1.1.24_eabi/host/node-v8.17.0/out/Release/.deps//openwrt/lede-17.01.4-Mod/lede-17.01.4-Mod-202306140018/build_dir/target-arm_cortex-a15+neon-vfpv4_musl-1.1.24_eabi/host/node-v8.17.0/out/Release/obj.target/openssl/deps/openssl/openssl/crypto/ui/ui_compat.o.d.raw -I/openwrt/lede-17.01.4-Mod/lede-17.01.4-Mod-202306140018/staging_dir/host/include -I/openwrt/lede-17.01.4-Mod/lede-17.01.4-Mod-202306140018/staging_dir/host/usr/include -I/openwrt/lede-17.01.4-Mod/lede-17.01.4-Mod-202306140018/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl-1.1.24_eabi/host/include -O2 -I/openwrt/lede-17.01.4-Mod/lede-17.01.4-Mod-202306140018/staging_dir/host/include -I/openwrt/lede-17.01.4-Mod/lede-17.01.4-Mod-202306140018/staging_dir/host/usr/include -I/openwrt/lede-17.01.4-Mod/lede-17.01.4-Mod-202306140018/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl-1.1.24_eabi/host/include -c -o /openwrt/lede-17.01.4-Mod/lede-17.01.4-Mod-202306140018/build_dir/target-arm_cortex-a15+neon-vfpv4_musl-1.1.24_eabi/host/node-v8.17.0/out/Release/obj.target/openssl/deps/openssl/openssl/crypto/ui/ui_compat.o ../deps/openssl/openssl/crypto/ui/ui_compat.c
rm -f /openwrt/lede-17.01.4-Mod/lede-17.01.4-Mod-202306140018/build_dir/target-arm_cortex-a15+neon-vfpv4_musl-1.1.24_eabi/host/node-v8.17.0/out/Release/obj.target/deps/openssl/libopenssl.a ar-file-list; mkdir -p `dirname /openwrt/lede-17.01.4-Mod/lede-17.01.4-Mod-202306140018/build_dir/target-arm_cortex-a15+neon-vfpv4_musl-1.1.24_eabi/host/node-v8.17.0/out/Release/obj.target/deps/openssl/libopenssl.a`
make[5]: /bin/sh: Argument list too long
make[5]: *** [deps/openssl/openssl.target.mk:862: /openwrt/lede-17.01.4-Mod/lede-17.01.4-Mod-202306140018/build_dir/target-arm_cortex-a15+neon-vfpv4_musl-1.1.24_eabi/host/node-v8.17.0/out/Release/obj.target/deps/openssl/libopenssl.a] Error 127
rm aadf4e7cb57e90c45bc155d6720ace42445b33b6.intermediate
make[4]: *** [Makefile:88: node] Error 2
make[4]: Leaving directory '/openwrt/lede-17.01.4-Mod/lede-17.01.4-Mod-202306140018/build_dir/target-arm_cortex-a15+neon-vfpv4_musl-1.1.24_eabi/host/node-v8.17.0'
make[3]: *** [Makefile:213: /openwrt/lede-17.01.4-Mod/lede-17.01.4-Mod-202306140018/build_dir/target-arm_cortex-a15+neon-vfpv4_musl-1.1.24_eabi/host/node-v8.17.0/.built] Error 2
make[2]: *** [package/Makefile:108: package/feeds/packages/node/host/compile] Error 2
make[3]: Leaving directory '/openwrt/lede-17.01.4-Mod/lede-17.01.4-Mod-202306140018/feeds/packages/lang/node'
make[2]: Leaving directory '/openwrt/lede-17.01.4-Mod/lede-17.01.4-Mod-202306140018'
make[1]: *** [package/Makefile:104: /openwrt/lede-17.01.4-Mod/lede-17.01.4-Mod-202306140018/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl-1.1.24_eabi/stamp/.package_compile] Error 2
make[1]: Leaving directory '/openwrt/lede-17.01.4-Mod/lede-17.01.4-Mod-202306140018'
make: *** [/openwrt/lede-17.01.4-Mod/lede-17.01.4-Mod-202306140018/include/toplevel.mk:200: world] Error 2

thanks

nxhack, 2023-06-19:
openwrt-17.01 is EoL and not supported.

nxhack, 2023-06-19:
@hzjnet 

Tips (No warranty) ;)

The version of gcc is too new.

LEDE branches are released in 2016-2019. Hence, your build environment should also match this; if you are debian, try building with Debian 9.0 (stretch).

nxhack, 2023-06-19:
For questions not related to node.js in this repository, please ask your custom distribution.

https://github.com/HiGarfield/lede-17.01.4-Mod/issues

nxhack, 2023-06-20:
OK issue closed :)

https://github.com/HiGarfield/lede-17.01.4-Mod/issues/7

closed 2023-06-20 by nxhack